The-OpenROAD-Project/OpenROAD #1206: Core dumped when using grt/FastRoute on ispd2015

albertface, 2021-10-17:
I have tried the following two way to apply global routing on ispd2015 but keep getting the core dumped error.
1. run global_route using after_legalized.ntup.fix.def directly
2. run global_placement on floorplan.def and then global_route
I don't know how to fix this because there is no detailed error message. Could anyone give some help?

ispd2015 is downloaded from [http://www.cerc.utexas.edu/~zixuan/ispd2015dp.tar.xz](http://www.cerc.utexas.edu/~zixuan/ispd2015dp.tar.xz).

The command and the log:
`openroad> read_lef tech.lef
[INFO ODB-0222] Reading LEF file: tech.lef
[WARNING ODB-0220] WARNING (LEFPARS-2007): NAMESCASESENSITIVE statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later. See file tech.lef at line 4.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file tech.lef at line 472.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file tech.lef at line 482.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file tech.lef at line 638.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file tech.lef at line 648.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file tech.lef at line 804.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file tech.lef at line 814.

[INFO ODB-0223]     Created 9 technology layers
[INFO ODB-0224]     Created 64 technology vias
[INFO ODB-0226] Finished LEF file:  tech.lef
openroad> read_lef cells.lef
[INFO ODB-0222] Reading LEF file: cells.lef
[INFO ODB-0225]     Created 16 library cells
[INFO ODB-0226] Finished LEF file:  cells.lef
openroad> read_def after_legalized.ntup.fix.def
[INFO ODB-0127] Reading DEF file: after_legalized.ntup.fix.def
[INFO ODB-0128] Design: fft
[INFO ODB-0130]     Created 3010 pins.
[INFO ODB-0131]     Created 30631 components and 97138 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 0 connections.
[INFO ODB-0133]     Created 32088 nets and 97138 connections.
[INFO ODB-0134] Finished DEF file: after_legalized.ntup.fix.def
openroad> 
openroad> #global_placement -routability_driven -density 0.70 
openroad> 
openroad> check_antennas -report_file antennas
[INFO ANT-0001] Found 0 pin violations.
[INFO ANT-0002] Found 0 net violations in 32088 nets.
0
openroad> 
openroad> global_route -guide_file route.guide
[INFO GRT-0020] Min routing layer: metal1
[INFO GRT-0021] Max routing layer: metal5
[INFO GRT-0022] Global adjustment: 0%
[INFO GRT-0023] Grid origin: (0, 0)
[WARNING GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0224] Chose via VIA12_HV as default.
[INFO GRT-0224] Chose via VIA23_VH as default.
[INFO GRT-0224] Chose via VIA34_HV as default.
[INFO GRT-0224] Chose via VIA45_VH as default.
Segmentation fault (core dumped)`

rovinski, 2021-10-17:
It would really help if you specified which one of the designs you used.

rovinski, 2021-10-17:
Design `mgc_des_perf_1` works.

Stacktrace:
```
#0  0x00000000008ded77 in odb::dbTrackGrid::getGridPatternX(int, int&, int&, int&) ()
#1  0x0000000000c32fa3 in grt::GlobalRouter::initRoutingTracks(std::vector<grt::RoutingTracks, std::allocator<grt::RoutingTracks> >&, int) ()
#2  0x0000000000c35ae5 in grt::GlobalRouter::startFastRoute(int, int, grt::NetType) ()
#3  0x0000000000c35bdc in grt::GlobalRouter::globalRoute() ()
```

maliberty, 2021-10-17:
There are tracks defined in only one direction per layer:

TRACKS Y 0 DO 2227 STEP 200 LAYER metal1 metal3 metal5 ;
TRACKS X 100 DO 2225 STEP 200 LAYER metal2 metal4 ;

Usually we have tracks in both direction so I guess that is the problem.  I assume they are trying to restrict to unidirectional routing.

albertface, 2021-10-17:
> There are tracks defined in only one direction per layer:
>
> TRACKS Y 0 DO 2227 STEP 200 LAYER metal1 metal3 metal5 ; TRACKS X 100 DO 2225 STEP 200 LAYER metal2 metal4 ;
>
> Usually we have tracks in [...]

I add tracks both x and y in metal 1-5 and it works. Thanks!

eder-matheus, 2021-10-18:
@maliberty I've updated grt code with checks for the tracks patterns, so we don't have crashes anymore. @albertface your testcases should not fail anymore, even if there are no tracks for the non-preferred direction of a layer. I've also added unit tests for the update.

closed 2021-10-18 by eder-matheus