The-OpenROAD-Project/OpenROAD #1267: We should see if LTO improves the performance of OpenROAD

QuantamHD, 2021-11-04:
LTO stands for link time optimization, and allows clang/gcc to optimize across linking boundaries.

We can enable it by default by setting `set(CMAKE_INTERPROCEDURAL_OPTIMIZATION TRUE)` in the root CMAKE file.

https://stackoverflow.com/questions/31355692/how-do-i-enable-link-time-optimization-lto-with-cmake

maliberty, 2021-11-04:
@QuantamHD do you want to try it?

cbalint13, 2022-04-10:
@QuantamHD 
@maliberty 

OpenROAD +LTO on means:
   - 636,005,720 vs 830,627,144 **unstripped** binary.
   - 44,501,056 vs 50,634,568 **stripped** binary.
   - 3m56.108s vs 3m57.196s regression **runtime**.

LTO is more about size optimisation (speed would second by a chance)
IMO is safe to enable LTO it gains some significant amount in smaller size.

---

### Setup conditions:

* OpenROAD git hash: a1cd122fe9bf2e6cb7b2b05be43c50dc5599d8f2

* GCC version:
```
$ gcc -v
Supported LTO compression algorithms: zlib zstd
gcc version 12.0.1 20220401 (Red Hat 12.0.1-0) (GCC)
```

* OpenROAD having **external** ```OpenSTA``` & ```abc``` (booth had LTO as per Fedora default):
```
$ readelf -a /usr/bin/openroads | grep NEED
 0x0000000000000001 (NEEDED)             Shared library: [libOpenSTA.so.2]
 0x0000000000000001 (NEEDED)             Shared library: [libabc.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libtcl8.6.so]
 0x0000000000000001 (NEEDED)             Shared library: [libtclreadline-2.1.0.so]
 0x0000000000000001 (NEEDED)             Shared library: [libX11.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libjpeg.so.62]
 0x0000000000000001 (NEEDED)             Shared library: [libgomp.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libboost_serialization.so.1.76.0]
 0x0000000000000001 (NEEDED)             Shared library: [libQt5Widgets.so.5]
 0x0000000000000001 (NEEDED)             Shared library: [libQt5Gui.so.5]
 0x0000000000000001 (NEEDED)             Shared library: [libQt5Core.so.5]
 0x0000000000000001 (NEEDED)             Shared library: [libspdlog.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libfmt.so.8]
 0x0000000000000001 (NEEDED)             Shared library: [libz.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libpython3.10.so.1.0]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
```

---

### LTO on:

* Extra flags:
```
$ rpm --eval %{_lto_cflags}
-flto=auto -ffat-lto-objects
```
* Size unstripped:
```
$ ls -l build/src/openroad
-rwxr-xr-x 1 cbalint cbalint 636005720 Apr 10 17:08 build/src/openroad
```
* Size stripped:
```
$ ls -l /usr/bin/openroad
-rwxr-xr-x 1 cbalint cbalint 44501056 Apr 10 17:09 /usr/bin/openroad
```

* Regression timing:
```
Passed all tool regressions.
real	3m56.108s
user	3m38.084s
sys	0m14.937s
```

### LTO off:

* Size unstripped:
```
$ ls -l build/src/openroad
-rwxr-xr-x 1 cbalint cbalint 830627144 Apr 10 17:20 build/src/openroad
```

* Size stripped:
```
ls -l /usr/bin/openroad
-rwxr-xr-x 1 cbalint cbalint 50634568 Apr 10 17:21 /usr/bin/openroad
```
* Regression timing:
```
Passed all tool regressions.
real	3m57.196s
user	3m42.273s
sys	0m14.460s
```

QuantamHD, 2022-04-11:
LTO was broken because of the QT components. We would need some additional work to selectively disable it around the GUI.

QuantamHD, 2022-04-11:
Auto-FDO would also be worth doing.

wjrforcyber, 2025-01-23:
Hi everyone! 
The doc should be updated with the current status of project, right now the `LINK_TIME_OPTIMIZATION` is `OFF` by default due to Qt issue, but the https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/docs/user/Build.md#lto-options still shows the build is LTO by default.

titan73, 2025-01-28:
@cbalint13 What is the build time overread?