The-OpenROAD-Project/OpenROAD #1271: sta: bad index in Table2::tableValue()

antonblanchard, 2021-11-05:
While chasing a SEGV, I noticed `Table2::tableValue` is indexing an array with -1UL. With the attached patch I see:

```
index1 2 index2 ffffffffffffffff
```

The test case is too large to attach, it's at https://ozlabs.org/~anton/openroad-bad-index/fail.tar.gz

```
diff --git a/liberty/TableModel.cc b/liberty/TableModel.cc
index e06c29e..488abb5 100644
--- a/liberty/TableModel.cc
+++ b/liberty/TableModel.cc
@@ -888,7 +888,10 @@ float
 Table2::tableValue(size_t index1,
                   size_t index2) const
 {
+  printf("index1 %lx index2 %lx\n", index1, index2);
   FloatSeq *row = (*values_)[index1];
+  if (index2 == 0xffffffffffffffffUL)
+         abort();
   return (*row)[index2];
 }
```

antonblanchard, 2021-11-05:
Backtrace:

```
#1  0x00007ffff4e578a4 in abort () from /lib64/libc.so.6
#2  0x00000000004f9650 in sta::Table2::tableValue (index2=<optimized out>, index1=<optimized out>, this=<optimized out>)
    at ../src/sta/liberty/TableModel.cc:894
#3  sta::Table2::findValue (this=<optimized out>, value1=<optimized out>, value2=<optimized out>)
    at ../src/sta/liberty/TableModel.cc:912
#4  0x0000000002395723 in sta::TableModel::findValue (value3=<optimized out>, value2=<optimized out>, 
    value1=<optimized out>, pvt=0x0, cell=0x6190001eaa80, library=0x6180002cb880, this=<optimized out>)
    at ../src/sta/liberty/TableModel.cc:617
#5  sta::GateTableModel::findValue (related_out_cap=0, load_cap=-nan(0x400000), in_slew=1.1147739e-10, 
    model=<optimized out>, pvt=0x0, cell=0x6190001eaa80, library=0x6180002cb880, this=0x6060001a8e80)
    at ../src/sta/liberty/TableModel.cc:200
#6  sta::GateTableModel::gateDelay (this=0x6060001a8e80, cell=0x6190001eaa80, pvt=<optimized out>, in_slew=<optimized out>, 
    load_cap=<optimized out>, related_out_cap=<optimized out>, pocv_enabled=<optimized out>, 
    gate_delay=@0x7fffffffbea0: 1.19673977e-29, drvr_slew=<optimized out>) at ../src/sta/liberty/TableModel.cc:96
#7  0x0000000003235c29 in sta::gateModelRd (pocv_enabled=<optimized out>, pvt=<optimized out>, 
    related_out_cap=<optimized out>, c1=-nan(0x8000000000000), c2=-nan(0x8000000000000), in_slew=1.1147738981920341e-10, 
    rf=0x2cc2fa00 <sta::RiseFall::rise_>, gate_model=0x6060001a8e80, cell=0x6190001eaa80) at ../src/sta/dcalc/DmpCeff.cc:1723
#8  sta::DmpCeffDelayCalc::setCeffAlgorithm (this=0x61200003c940, drvr_library=<optimized out>, drvr_cell=0x6190001eaa80, 
    pvt=<optimized out>, gate_model=0x6060001a8e80, rf=<optimized out>, in_slew=<optimized out>, 
    related_out_cap=<optimized out>, c2=<optimized out>, rpi=<optimized out>, c1=<optimized out>)
    at ../src/sta/dcalc/DmpCeff.cc:1611
#9  0x0000000003237476 in sta::DmpCeffDelayCalc::gateDelay (this=0x61200003c940, drvr_cell=<optimized out>, 
    arc=<optimized out>, in_slew=<optimized out>, load_cap=<optimized out>, drvr_parasitic=<optimized out>, 
    related_out_cap=<optimized out>, pvt=<optimized out>, dcalc_ap=<optimized out>, gate_delay=<optimized out>, 
    drvr_slew=<optimized out>) at ../src/sta/dcalc/DmpCeff.cc:1574
#10 0x000000000150d583 in rsz::Resizer::cellWireDelay (this=this@entry=0x61900012d480, drvr_port=<optimized out>, 
    load_port=<optimized out>, wire_length=wire_length@entry=inf, delay=@0x7fffffffc4a0: -1.00000002e+30, 
    slew=@0x7fffffffc4c0: -1.00000002e+30) at ../src/rsz/src/Resizer.cc:3559
#11 0x000000000150f83a in rsz::Resizer::bufferWireDelay (slew=@0x7fffffffc4c0: -1.00000002e+30, 
    delay=@0x7fffffffc4a0: -1.00000002e+30, wire_length=inf, buffer_cell=<optimized out>, this=0x61900012d480)
    at ../src/rsz/src/Resizer.cc:3500
#12 rsz::Resizer::splitWireDelayDiff (this=this@entry=0x61900012d480, wire_length=<optimized out>, 
    buffer_cell=buffer_cell@entry=0x6190001eaa80) at ../src/rsz/src/Resizer.cc:3486
#13 0x000000000150fcb3 in rsz::Resizer::findMaxWireLength (this=this@entry=0x61900012d480, drvr_port=<optimized out>, 
    corner=corner@entry=0x60d000000790) at ../src/rsz/src/Resizer.cc:3462
#14 0x0000000001510f54 in rsz::Resizer::findMaxWireLength (corner=0x60d000000790, buffer_cell=0x6190001eaa80, 
    this=0x61900012d480) at ../src/rsz/src/Resizer.cc:3441
#15 rsz::Resizer::findMaxWireLength (this=0x61900012d480) at ../src/rsz/src/Resizer.cc:3425
#16 0x000000000160b5f8 in rsz::Resizer::resizeSlackPreamble (this=this@entry=0x61900012d480)
    at ../src/rsz/src/Resizer.cc:1784
#17 0x0000000000ab90b3 in gpl::Replace::doNesterovPlace (this=this@entry=0x611000001300, start_iter=start_iter@entry=0)
    at ../src/gpl/src/replace.cpp:326
#18 0x0000000000e88ec1 in replace_nesterov_place_cmd () at src/gpl/CMakeFiles/gpl.dir/replaceTCL_wrap.cxx:1742
#19 _wrap_replace_nesterov_place_cmd (clientData=<optimized out>, interp=0x629000064210, objc=<optimized out>, 
    objv=<optimized out>) at src/gpl/CMakeFiles/gpl.dir/replaceTCL_wrap.cxx:2178
```

jjcherry56, 2021-11-10:
your test case is calling the global placer through the back door (not the tcl command) and it is defaulting to timing driven (which makes no sense but that's a different issue) and your test case does not have an sdc file or wire rc values which is not where you want to be, but I bullet proofed it so it doesn't blow up.

antonblanchard, 2021-11-10:
Thanks @jjcherry56. This test was cut down from Openlane: [or_replace.tcl](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/openroad/or_replace.tcl). Should we fix the timing driven (and routability driven) defaults in OpenROAD, or set them to 0 where appropriate in Openlane?

In terms of the missing RC values when timing driven mode is enabled, is it enough to call `set_wire_rc` beforehand?

jjcherry56, 2021-11-11:
That whole script is using internal functions instead of the command, which doesn't make much sense.
It also fails to read liberty files, and there is no hope of timing driven working without them.

maliberty, 2021-11-11:
OL has agreed to move to the tcl wrappers rather than the swig APIs.  It will take some time for that to happen.

antonblanchard, 2021-12-13:
I've reproduced this using the latest `OpenROAD` and `OpenROAD-flow-scripts`, building `designs/sky130hd/jpeg`:

```
index1 5 index2 ffffffffffffffff

Program received signal SIGABRT, Aborted.
0x00007ffff61ae2a2 in raise () from /lib64/libc.so.6
(gdb) backtrace 
#0  0x00007ffff61ae2a2 in raise () from /lib64/libc.so.6
#1  0x00007ffff61978a4 in abort () from /lib64/libc.so.6
#2  0x000000000046ce63 in sta::Table2::findValue(float, float, float) const [clone .cold] ()
#3  0x000000000073ce9b in sta::GateTableModel::gateDelay(sta::LibertyCell const*, sta::Pvt const*, float, float, float, bool, float&, float&) const ()
#4  0x0000000000843449 in sta::DmpPi::gateDelaySlew(double&, double&) ()
#5  0x000000000084494b in sta::DmpCeffDelayCalc::gateDelay(sta::LibertyCell const*, sta::TimingArc*, float const&, float, sta::Parasitic*, float, sta::Pvt const*, sta::DcalcAnalysisPt const*, float&, float&) ()
#6  0x000000000084871d in sta::GraphDelayCalc1::findArcDelay(sta::LibertyCell*, sta::Pin const*, sta::Vertex*, sta::MultiDrvrNet*, sta::TimingArc*, sta::Parasitic*, float, sta::Vertex*, sta::Edge*, sta::Pvt const*, sta::DcalcAnalysisPt const*, sta::ArcDelayCalc*) ()
#7  0x0000000000848bc2 in sta::GraphDelayCalc1::findDriverEdgeDelays(sta::LibertyCell*, sta::Instance*, sta::Pin const*, sta::Vertex*, sta::MultiDrvrNet*, sta::Edge*, sta::ArcDelayCalc*) ()
#8  0x0000000000848db8 in sta::GraphDelayCalc1::findDriverDelays1(sta::Vertex*, bool, sta::MultiDrvrNet*, sta::ArcDelayCalc*) ()
#9  0x0000000000848e92 in sta::GraphDelayCalc1::findDriverDelays(sta::Vertex*, sta::ArcDelayCalc*) ()
#10 0x000000000084aa42 in sta::FindVertexDelays::visit(sta::Vertex*) ()
#11 0x00000000007aa359 in sta::BfsIterator::visit(int, sta::VertexVisitor*) ()
#12 0x000000000084b09f in sta::GraphDelayCalc1::findDelays(int) ()
#13 0x00000000007fde0c in sta::Sta::checkSlewLimitPreamble() ()
#14 0x000000000061d9a3 in rsz::Resizer::repairDesign(double, double, double, int&, int&, int&, int&, int&) ()
#15 0x000000000061f1df in rsz::Resizer::repairDesign(double, double, double) ()
#16 0x0000000000642953 in _wrap_repair_design_cmd ()
#17 0x00007ffff7e33042 in TclNRRunCallbacks () from /lib64/libtcl8.6.so
#18 0x00007ffff7e38da7 in ?? () from /lib64/libtcl8.6.so
#19 0x00007ffff7e395d7 in Tcl_EvalEx () from /lib64/libtcl8.6.so
#20 0x00007ffff7e395fa in Tcl_Eval () from /lib64/libtcl8.6.so
#21 0x00000000007112bc in sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) ()
#22 0x000000000055f208 in ord::tclAppInit(Tcl_Interp*) ()
#23 0x00007ffff7ee8ee6 in Tcl_MainEx () from /lib64/libtcl8.6.so
#24 0x000000000054736f in main ()
```

antonblanchard, 2021-12-13:
Test case: [sky130hd_jpeg_tablevalue.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7700147/sky130hd_jpeg_tablevalue.tar.gz)

QuantamHD, 2021-12-13:
@maliberty seems like we should reopen?

maliberty, 2021-12-13:
I just tried your test with 82d19cf9c235a103ade7728e3fa9a4beee976586 and I don't get any error when running 'openroad -exit resize.tcl".

[INFO ODB-0222] Reading LEF file: sky130_fd_sc_hd.tlef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0226] Finished LEF file:  sky130_fd_sc_hd.tlef
[INFO ODB-0222] Reading LEF file: sky130_fd_sc_hd_merged.lef
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  sky130_fd_sc_hd_merged.lef
[INFO ODB-0127] Reading DEF file: 3_2_place_iop.def
[INFO ODB-0128] Design: jpeg_encoder
[INFO ODB-0130]     Created 47 pins.
[INFO ODB-0131]     Created 76222 components and 403641 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 245056 connections.
[INFO ODB-0133]     Created 58681 nets and 158584 connections.
[INFO ODB-0134] Finished DEF file: 3_2_place_iop.def
Perform port buffering...
[INFO RSZ-0027] Inserted 19 input buffers.
[INFO RSZ-0028] Inserted 26 output buffers.
Perform buffer insertion...
[INFO RSZ-0058] Using max wire length 2431um.
[INFO RSZ-0034] Found 12387 slew violations.
[INFO RSZ-0036] Found 13 capacitance violations.
[INFO RSZ-0037] Found 2 long wires.
[INFO RSZ-0038] Inserted 225 buffers in 12387 nets.
[INFO RSZ-0039] Resized 5543 instances.
[~/w6/OpenROAD-flow-scripts/tools/OpenROAD/sky130hd_jpeg_tablevalue]
% echo $?
0

maliberty, 2021-12-13:
@QuantamHD I'll reopen when a reproducible problem exists.

antonblanchard, 2021-12-13:
Hi @maliberty did you have the patch mentioned [here](https://github.com/The-OpenROAD-Project/OpenROAD/issues/1271#issue-1045476580) ? The other option is to build with the LLVM undefined behaviour sanitizer (how I originally hit it).

maliberty, 2021-12-13:
Ok I see it with the patch.

maliberty, 2021-12-13:
@jjcherry56 a quick look shows it failing in ceff and returning a nan which leads to the downstream issue.  The gate is way outside the .lib table.

tspyrou, 2021-12-13:
It looks like the testcase has been nicely cleaned up now.

jjcherry56, 2021-12-13:
The original bug was fixed by OpenSTA commit ddd7f39 James Cherry 2021-11-09 dcalc nan proofing.
This is different similar issue fixed by OR commit 4c696dc5a or1271 sta bad index in Table2::tableValue

closed 2022-01-03 by jjcherry56