The-OpenROAD-Project/OpenROAD #1272: gpl: cannot figure out few macros

qujijie, 2021-11-05:
I have a simple test that only have few macros, i have these erroe when i use globle placement 
```
[NesterovSolve] Iter: 2760 overflow: 0.211169 HPWL: 807523
[NesterovSolve] Iter: 2770 overflow: 0.211203 HPWL: 807433
[NesterovSolve] Iter: 2780 overflow: 0.208407 HPWL: 846022
[NesterovSolve] Iter: 2790 overflow: 0.207827 HPWL: 7211705
[NesterovSolve] Iter: 2800 overflow: 0.19835 HPWL: 5721575
[NesterovSolve] Iter: 2810 overflow: 0.197172 HPWL: 7054532
[NesterovSolve] Iter: 2820 overflow: 0.197242 HPWL: 7740887
[NesterovSolve] Iter: 2830 overflow: 0.197312 HPWL: 9096105
[NesterovSolve] Iter: 2840 overflow: 0.197381 HPWL: 9259699
[ERROR GPL-0305] RePlAce diverged at newStepLength.
Error: macro_only.tcl, 5 GPL-0305
```
here is the lef/def, how can i solve these problems. 
Thank you for your kind cooperation
[macros.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/7486102/macros.zip)

rovinski, 2021-11-06:
@mgwoo if possible the error should include suggestions to correct because it's a bit esoteric.

mgwoo, 2021-11-09:
Sorry, but this testcase is not appropriate on OpenROAD testcases. If the number of instances or nets are too small, I would prefer to use a simulated annealer to place macros, etc.

rovinski, 2021-11-09:
> If the number of instances or nets are too small, I would prefer to use a simulated annealer or BiCGSTAB QP solver to place macros. The nonlinear placer cannot handle this wedged testcase.

@mgwoo is it possible to detect this behavior and add an error stating this?

maliberty, 2021-11-09:
@ravi-varadarajan hopefully rtl-mp can handle this case.  Would you verify?

ravi-varadarajan, 2021-11-10:
@mgwoo  Mingyu have you set up the test case in some location in UCSD server?

mgwoo, 2021-11-17:
Confirmed that this testcase is working well if you put the following: 

    global_placement -skip_initial_place 

or: 

    global_placement -bin 4

closed 2021-11-17 by mgwoo