The-OpenROAD-Project/OpenROAD #1277: resizer timining run crashes with segementation fault

dineshannayya, 2021-11-06:
I am using the or_resizer_timing.tcl from efabless/openlane project.

While Running the Resize timing timing optimization, tool crashes with segmentation fault.
<pre>###############################################################################
# Design Rules
###############################################################################
[INFO]: Setting RC values...
[INFO RSZ-0046] Found 51 endpoints with hold violations.
<font color="#CC0000">[ERROR]: during executing: &quot;openroad -exit input/or_resizer_timing.tcl |&amp; tee &gt;&amp;@stdout output/resizer_timing_optimization.log&quot;</font>
<font color="#CC0000">[ERROR]: Exit code: 1</font>
<font color="#CC0000">[ERROR]: Last 10 lines:</font>
<font color="#CC0000">child killed: segmentation violation</font>

<font color="#CC0000">[ERROR]: Please check openroad  log file</font>
</pre>

The issue observed with latest efabless/openlane:latest docker and Standalone Openroad version: v2.0-1424-gf864efbae 

Attached the Example:

[exp2.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7490731/exp2.tar.gz)

Direct Run option-1: tclsh input/resizer.tcl
Docker Run Option-2: docker run -ti --rm  -v $(pwd):/project efabless/openlane:latest  sh -c "cd /project && tclsh input/resizer.tcl"

jjcherry56, 2021-11-10:
I see the problem.
In the mean time, you can work around it by adding a setup constraint for spi_sdo[0] which I suspect was unintentionally
left out of the sdc.
set_output_delay 6.0000 -clock [get_clocks {spiclk}] -max -add_delay [get_ports {spi_sdo[0]}]

dineshannayya, 2021-11-10:
@jjcherry56 , Flow worked once i add max output delay to spi_sdo[0], Thanks

jjcherry56, 2021-11-11:
fixed

closed 2021-11-11 by jjcherry56