The-OpenROAD-Project/OpenROAD #1281: Obstructions partially outside die area cause malloc corruption (GlobalRouter::computeObstructionsAdjustments)

antonblanchard, 2021-11-07:
I'm hitting malloc corruption during global placement. The issue appears to be in `GlobalRouter::computeObstructionsAdjustments()`, where obstructions that are partially outside the die area (so we don't discard them), end up negatively indexing, or indexing off the end of or `v_edges_` and `h_edges_`.

The test case (too large to attach) is at https://ozlabs.org/~anton/openroad-bad-index/fail.tar.gz

antonblanchard, 2021-11-07:
A bit of a hack, but this fixes it.

```
diff --git a/src/grt/src/GlobalRouter.cpp b/src/grt/src/GlobalRouter.cpp
index 2475ce34c..e01c69c06 100644
--- a/src/grt/src/GlobalRouter.cpp
+++ b/src/grt/src/GlobalRouter.cpp
@@ -1201,10 +1201,34 @@ void GlobalRouter::computeObstructionsAdjustments()
         int last_tile_reduce = grid_->computeTileReduce(
             obs, last_tile_box, track_space, false, tech_layer->getDirection());
 
+        int x_start = first_tile._x;
+        if (x_start < 0) {
+	    printf("adjusting x_start from %d to %d\n", x_start, 0);
+            x_start = 0;
+	}
+
+        int y_start = first_tile._y;
+        if (y_start < 0) {
+	    printf("adjusting y_start from %d to %d\n", y_start, 0);
+            y_start = 0;
+	}
+
+        int x_end = last_tile._x;
+        if (x_end >= fastroute_->x_grid_) {
+	    printf("adjusting x_end from %d to %d\n", x_end, fastroute_->x_grid_-1);
+            x_end = fastroute_->x_grid_-1;
+	}
+
+        int y_end = last_tile._y;
+        if (y_end >= fastroute_->y_grid_) {
+	    printf("adjusting y_end from %d to %d\n", y_end, fastroute_->y_grid_-1);
+            y_end = fastroute_->y_grid_-1;
+	}
+
         if (!vertical) {
-          for (int x = first_tile._x; x < last_tile._x; x++) {
-            for (int y = first_tile._y; y <= last_tile._y; y++) {
-              if (y == first_tile._y) {
+          for (int x = x_start; x < x_end; x++) {
+            for (int y = y_start; y <= y_end; y++) {
+              if (y == y_start) {
                 int edge_cap
                     = fastroute_->getEdgeCapacity(x, y, level, x + 1, y, level);
                 edge_cap -= first_tile_reduce;
@@ -1212,7 +1236,7 @@ void GlobalRouter::computeObstructionsAdjustments()
                   edge_cap = 0;
                 fastroute_->addAdjustment(
                     x, y, level, x + 1, y, level, edge_cap, true);
-              } else if (y == last_tile._y) {
+              } else if (y == y_end) {
                 int edge_cap
                     = fastroute_->getEdgeCapacity(x, y, level, x + 1, y, level);
                 edge_cap -= last_tile_reduce;
@@ -1227,9 +1251,9 @@ void GlobalRouter::computeObstructionsAdjustments()
             }
           }
         } else {
-          for (int x = first_tile._x; x <= last_tile._x; x++) {
-            for (int y = first_tile._y; y < last_tile._y; y++) {
-              if (x == first_tile._x) {
+          for (int x = x_start; x <= x_end; x++) {
+            for (int y = y_start; y < y_end; y++) {
+              if (x == x_start) {
                 int edge_cap
                     = fastroute_->getEdgeCapacity(x, y, level, x, y + 1, level);
                 edge_cap -= first_tile_reduce;
@@ -1237,7 +1261,7 @@ void GlobalRouter::computeObstructionsAdjustments()
                   edge_cap = 0;
                 fastroute_->addAdjustment(
                     x, y, level, x, y + 1, level, edge_cap, true);
-              } else if (x == last_tile._x) {
+              } else if (x == x_end) {
                 int edge_cap
                     = fastroute_->getEdgeCapacity(x, y, level, x, y + 1, level);
                 edge_cap -= last_tile_reduce;
diff --git a/src/grt/src/fastroute/include/FastRoute.h b/src/grt/src/fastroute/include/FastRoute.h
index 1b0fb45ff..d8a570c18 100644
--- a/src/grt/src/fastroute/include/FastRoute.h
+++ b/src/grt/src/fastroute/include/FastRoute.h
@@ -185,6 +185,9 @@ class FastRouteCore
   void setDebugTree2D(bool tree2D);
   void setDebugTree3D(bool tree3D);
 
+  int x_grid_;
+  int y_grid_;
+
  private:
   NetRouteMap getRoutes();
   void init_usage();
@@ -470,8 +473,6 @@ class FastRouteCore
   int num_adjust_;
   int v_capacity_;
   int h_capacity_;
-  int x_grid_;
-  int y_grid_;
   int x_corner_;
   int y_corner_;
   int w_tile_;
```

rovinski, 2021-11-07:
I'm not sure if it's reasonable to have obstructions outside of the die area. We may want to be checking for that instead of trying to work around it in the global router. What's the use case?

antonblanchard, 2021-11-07:
I'm trying to harden a caravel user_project_wrapper in one step: https://github.com/efabless/caravel_user_project/blob/main/docs/source/index.rst#hardening-options

[image]

I'm guessing the I/O pins are causing the issue, and they extend out of the die area so efabless can guarantee the GDS will insert into the rest of the chip without modification. I could do it in two steps (avoiding any standard cells on the second step), but that seems like a waste of die space.

antonblanchard, 2021-11-07:
We already warn and throw away obstructions completely outside the die space, it's just ones that partially extend outside it.

rovinski, 2021-11-07:
So if I understand correctly, the user design is being created as a block and then that block is being instantiated in the top-level design? And then the problem arises because there are pins that extend outside of the die area at the block level?

If so, then overall that's a user error. There should never be any shapes extending outside of the die area. The die area should instead be extended to cover the shapes. OpenROAD should report an error about shapes outside of the die area.

The standard methodology is for macro pins to be exposed on the edges (for signal pins) or top layer (for power pins) with a wide enough pitch between pins to ensure routability. Ideally the pins should be wide enough to allow on-grid routing regardless of placement, but the detailed router should be able to handle off-grid pins for sky130.

antonblanchard, 2021-11-07:
I believe that is how it works, but I'll leave it to @donn or @RTimothyEdwards to confirm.

maliberty, 2021-11-08:
@antonblanchard I don't understand "they extend out of the die area so efabless can guarantee the GDS will insert into the rest of the chip without modification".  How does making them extend outside help?

antonblanchard, 2021-11-08:
@maliberty I'm not sure, and I'm fast getting out of my depth here. These Openlane config options appear to do it:

```
FP_IO_VEXTEND  |  Extends the vertical io pins outside of the die by the specified units (Default: -1 Disabled) |
FP_IO_HEXTEND  |  Extends the horizontal io pins outside of the die by the specified units (Default: -1 Disabled) |
```

maliberty, 2021-11-08:
If we handle obstructions completely outside then we should handle ones partially outside.  I do think it is a strange methodology but the tool should never crash.  It doesn't seem dire enough to be an error.

maliberty, 2021-11-08:
@eder-matheus I would fix this by having Grid::addObstruction clip the obstruction to the Grid and discard it if nothing remains.  It's safer than sanitizing one use of the obstructions.

closed 2021-11-09 by eder-matheus