The-OpenROAD-Project/OpenROAD #1296: Replace diverges at initial iteration on trivial design

donn, 2021-11-11:
OpenROAD Version: v2.0-1750-gce49eb6f2

## Description
A simple inverter def layout fails to place with RePlace.

## Logs
```
OpenROAD v2.0-1750-gce49eb6f2 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./tmp/merged_unpadded.lef
[WARNING ODB-0220] WARNING (LEFPARS-2036): SOURCE statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later. See file ./tmp/merged_unpadded.lef at line 794.

[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  ./tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: ./in.def
[INFO ODB-0128] Design: inverter
[INFO ODB-0130]     Created 4 pins.
[INFO ODB-0131]     Created 32 components and 116 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 114 connections.
[INFO ODB-0133]     Created 2 nets and 2 connections.
[INFO ODB-0134] Finished DEF file: ./in.def
-verbose_level 1 -density 0.75 -overflow 0.9 -init_density_penalty 0.0001 -initial_place_max_iter 20 -bin_grid_count 64
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 460 2720
[INFO GPL-0004] CoreAreaLxLy: 5520 10880
[INFO GPL-0005] CoreAreaUxUy: 28520 43520
[INFO GPL-0006] NumInstances: 32
[INFO GPL-0007] NumPlaceInstances: 1
[INFO GPL-0008] NumFixedInstances: 31
[INFO GPL-0009] NumDummyInstances: 0
[INFO GPL-0010] NumNets: 2
[INFO GPL-0011] NumPins: 4
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 34165 54885
[INFO GPL-0014] CoreAreaLxLy: 5520 10880
[INFO GPL-0015] CoreAreaUxUy: 28520 43520
[INFO GPL-0016] CoreArea: 750720000
[INFO GPL-0017] NonPlaceInstsArea: 98844800
[INFO GPL-0018] PlaceInstsArea: 3753600
[INFO GPL-0019] Util(%): 0.58
[INFO GPL-0020] StdInstsArea: 3753600
[INFO GPL-0021] MacroInstsArea: 0
[InitialPlace]  Iter: 1 CG Error: 0.00000008 HPWL: 93755
[InitialPlace]  Iter: 2 CG Error: 0.00000008 HPWL: 92385
[InitialPlace]  Iter: 3 CG Error: 0.00000008 HPWL: 92385
[InitialPlace]  Iter: 4 CG Error: 0.00000008 HPWL: 92385
[InitialPlace]  Iter: 5 CG Error: 0.00000008 HPWL: 92385
[INFO GPL-0031] FillerInit: NumGCells: 130
[INFO GPL-0032] FillerInit: NumGNets: 2
[INFO GPL-0033] FillerInit: NumGPins: 4
[INFO GPL-0023] TargetDensity: 0.75
[INFO GPL-0024] AveragePlaceInstArea: 3753600
[INFO GPL-0025] IdealBinArea: 5004800
[INFO GPL-0026] IdealBinCnt: 150
[INFO GPL-0027] TotalBinArea: 750720000
[INFO GPL-0028] BinCnt: 64 64
[INFO GPL-0029] BinSize: 360 510
[INFO GPL-0030] NumBins: 4096
[ERROR GPL-0304] RePlAce diverged at initial iteration. Re-run with a smaller init_density_penalty value.
Error: replace.tcl, 83 GPL-0304
```

donn, 2021-11-16:
For the record I'm going to state we've started utilizing a random OpenDB-based placer for tiny designs, so you don't need to rush this. Take your time

mgwoo, 2021-11-17:
This issue has been fixed and the testcase is added to the simple07-08 regression unit testcases (link)[https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/gpl/test/simple07.def]

closed 2021-11-17 by mgwoo