The-OpenROAD-Project/OpenROAD #1303: read_def crashes when reading a 5.7 DEF file TECHNOLOGY statement

mbalestrini, 2021-11-12:
OR version v2.0-1685-g50f0e3241

When I load a .DEF file (version 5.7) with the **TECHNOLOGY** statement (that I understand is part of the 5.7 file format) I get this error and OR exits:
```
[WARNING ODB-0126] error: TECHNOLOGY is unsupported
ERROR (DEFPARS-6010): An error has been reported in callback. Error in file def_with_techonolgy_5.7.def at line 6, on token ;.
Update the def file before parsing the file again.
[WARNING ODB-0149] DEF parser returns an error!
```

To reproduce the error:
```
read_lef sky130_fd_sc_hd.tlef
read_def def_with_technolgy_5.7
```
[def_with_technolgy_5.7.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/7530545/def_with_technolgy_5.7.zip)

rovinski, 2021-11-12:
The cause is right there in the error message. LEF/DEF 5.7 is unsupported. Use 5.3, 5.4, 5.5, 5.6, or 5.8.

maliberty, 2021-11-13:
@rovinski where does it say 5.7 is unsupported?  I just see "error: TECHNOLOGY is unsupported"

We could just ignore that statement.

@mbalestrini I don't see any TECHNOLOGY statement in your in.def.

rovinski, 2021-11-13:
We can't write DEF5.7, so I assumed that we can't write it either. Is this wrong? And if so then why don't we write DEF5.7?

maliberty, 2021-11-13:
@rovinski  You are write but it looks like a mistake to me - the original opendb support 5.3-5.6 and Cherry added 5.8 but skipped over 5.7 (https://github.com/The-OpenROAD-Project/OpenDB/commit/73270b82cb67deab67b9923553dc3946a353c218).  I don't see any reason so I'll fix that.  However this seems unrelated to the issue here.

mbalestrini, 2021-11-13:
> 
> @mbalestrini I don't see any TECHNOLOGY statement in your in.def.

It's supposed to be there. The .def file only has this inside:
```
VERSION 5.7 ;
   NAMESCASESENSITIVE ON ;
   DIVIDERCHAR "/" ;
   BUSBITCHARS "()" ;
   DESIGN test_design ;
   TECHNOLOGY sky130A ;
   UNITS DISTANCE MICRONS 1000 ;
   DIEAREA ( 0 0 ) ( 1000 1000 ) ; 
```

maliberty, 2021-11-13:
Btw, what tool generated this DEF?

closed 2021-11-13 by maliberty

mbalestrini, 2021-11-13:
It was generated in magic. I was trying the `def write` command.
(I took away all the content and left just the header in this case)

mbalestrini, 2021-11-13:
You didn't get the error when you tried it? Because I updated to the last commit and I'm still getting the error, just with the new message:
```
[WARNING ODB-0293] TECHNOLOGY is ignored
ERROR (DEFPARS-6002): Incomplete def file. Error in file def_with_technolgy_5.7.def at line 9, on token ;.
Update the def file before parsing the file again.
[WARNING ODB-0149] DEF parser returns an error!
```

The problem is that after the warning, as it happen before, the app throws that error (and exits). Seems like it wasn't really ignoring that line.
I'm going to try a fresh download of the repo and build the app again just in case.

maliberty, 2021-11-13:
The def file is incomplete - all DEF must end with "END DESIGN".  The def itself is empty and useless - I was just fixing the TECHNOLOGY issue.

mbalestrini, 2021-11-13:
Yeah, sorry. I was trying to simplify the example but I deleted too much.