The-OpenROAD-Project/OpenROAD #1305: STA Segfault

donn, 2021-11-13:
```
OpenROAD v2.0-1750-gce49eb6f2

This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  ./tmp/merged_unpadded.lef
./run: line 14: 462423 Segmentation fault      (core dumped) $OPENROAD_BIN -exit $OR_SCRIPT_0
```

[packaged.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7531419/packaged.tar.gz)

donn, 2021-11-13:
Also verified on OpenROAD v2.0-1764-g9d9403c75, which only builds with clang and not gcc for some reason.

donn, 2021-11-13:
lldb backtrace for your convenience:

```
Process 476509 stopped
* thread #1, name = 'openroad', stop reason = signal SIGSEGV: invalid address (fault address: 0x0)
    frame #0: 0x000055555586a705 openroad`void utl::Logger::log<char const*, char const*>(utl::ToolId, spdlog::level::level_enum, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const* const&, char const* const&) + 53
openroad`utl::Logger::log<char const*, char const*>:
->  0x55555586a705 <+53>: movswl 0x38(%rcx,%rax,2), %edx
    0x55555586a70a <+58>: leal   0x1(%rdx), %esi
    0x55555586a70d <+61>: movw   %si, 0x38(%rcx,%rax,2)
    0x55555586a712 <+66>: leaq   0x221fe8f(%rip), %rsi     ; utl::Logger::max_message_print
(lldb) bt
* thread #1, name = 'openroad', stop reason = signal SIGSEGV: invalid address (fault address: 0x0)
  * frame #0: 0x000055555586a705 openroad`void utl::Logger::log<char const*, char const*>(utl::ToolId, spdlog::level::level_enum, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const* const&, char const* const&) + 53
    frame #1: 0x000055555586e1b9 openroad`sta::dbNetwork::readLibertyAfter(sta::LibertyLibrary*) + 553
    frame #2: 0x00005555559461ee openroad`sta::Sta::readLibertyFile(char const*, sta::Corner*, sta::MinMaxAll const*, bool, sta::Network*) + 190
    frame #3: 0x00005555559460c3 openroad`sta::Sta::readLiberty(char const*, sta::Corner*, sta::MinMaxAll const*, bool) + 83
    frame #4: 0x000055555582b467 openroad`_wrap_read_liberty_cmd(void*, Tcl_Interp*, int, Tcl_Obj* const*) + 535
    frame #5: 0x00007ffff7e20c62 libtcl.so`TclNRRunCallbacks + 114
    frame #6: 0x00007ffff7e22b3a libtcl.so`TclEvalEx + 1386
    frame #7: 0x00007ffff7e23363 libtcl.so`Tcl_EvalEx + 19
    frame #8: 0x00007ffff7e23387 libtcl.so`Tcl_Eval + 23
    frame #9: 0x0000555555874eb8 openroad`sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) + 120
    frame #10: 0x00005555556a78de openroad`ord::tclAppInit(Tcl_Interp*) + 1214
    frame #11: 0x00007ffff7ee5135 libtcl.so`Tcl_MainEx + 453
    frame #12: 0x00005555556a7f54 openroad`main + 1268
    frame #13: 0x00007ffff6030b25 libc.so.6`__libc_start_main + 213
    frame #14: 0x00005555556976ce openroad`_start + 46
```

RTimothyEdwards, 2021-11-13:
@donn :   Any ideas what's going on?  There should have been no changes to liberty files in the last open_pdks update, so nothing should have changed for STA. . . So I don't even know where to begin looking.

jjcherry56, 2021-11-13:
What is going on is there are errors in the LEF/liberty port names and the logger is not properly initialized. The problem does not show up with well formed libraries.

RTimothyEdwards, 2021-11-13:
@jjcherry56 :  Can anybody tell me what those errors are?  The liberty files did not change, but I did make a change to the LEF files to fix the erroneous "soft" ports in the original SkyWater LEF macros for the standard cells.  Chances are that the modification introduced some syntax error into the LEF files.  But I don't know where to go looking for that, since OpenROAD crashed rather than generate an error message.

RTimothyEdwards, 2021-11-13:
In my book, a segfault is something that needs immediate attention and a fix, even if it is the input data that's at fault.

jjcherry56, 2021-11-13:
FIne, next time I won't bother to tell you why it seg faulted, I'll just make you wait until it is fixed. I don't need a lesson on the severity of a seg fault.

jjcherry56, 2021-11-14:
fixed.
Note that the sta.tcl script will discard any DEF placement information if it is ever used with CURRENT_DEF because the unconditional  read_verilog following read_def will clobber the netlist in the database.

closed 2021-11-14 by jjcherry56

RTimothyEdwards, 2021-11-14:
@donn :  With the segfault fixed, is there any additional information on what input caused the issue in the first place?  Does it remain a problem or not?