The-OpenROAD-Project/OpenROAD #1310: report_power segfault

donn, 2021-11-15:
```
OpenROAD v2.0-1771-g2d043b9f4 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  ./tmp/merged_unpadded.lef
if {[info exists ::env(CLOCK_PORT)] && $::env(CLOCK_PORT) != ""} { 
    create_clock [get_ports $::env(CLOCK_PORT)]  -name $::env(CLOCK_PORT)  -period $::env(CLOCK_PERIOD)
} else {
    create_clock -name __VIRTUAL_CLK__ -period $::env(CLOCK_PERIOD)
    set ::env(CLOCK_PORT) __VIRTUAL_CLK__
}
set input_delay_value [expr $::env(CLOCK_PERIOD) * $::env(IO_PCT)]
set output_delay_value [expr $::env(CLOCK_PERIOD) * $::env(IO_PCT)]
puts "\[INFO\]: Setting output delay to: $output_delay_value"
[INFO]: Setting output delay to: 2.0
puts "\[INFO\]: Setting input delay to: $input_delay_value"
[INFO]: Setting input delay to: 2.0
set_max_fanout $::env(SYNTH_MAX_FANOUT) [current_design]
set clk_indx [lsearch [all_inputs] [get_port $::env(CLOCK_PORT)]]
#set rst_indx [lsearch [all_inputs] [get_port resetn]]
set all_inputs_wo_clk [lreplace [all_inputs] $clk_indx $clk_indx]
#set all_inputs_wo_clk_rst [lreplace $all_inputs_wo_clk $rst_indx $rst_indx]
set all_inputs_wo_clk_rst $all_inputs_wo_clk
# correct resetn
set_input_delay $input_delay_value  -clock [get_clocks $::env(CLOCK_PORT)] $all_inputs_wo_clk_rst
#set_input_delay 0.0 -clock [get_clocks $::env(CLOCK_PORT)] {resetn}
set_output_delay $output_delay_value  -clock [get_clocks $::env(CLOCK_PORT)] [all_outputs]
# TODO set this as parameter
set_driving_cell -lib_cell $::env(SYNTH_DRIVING_CELL) -pin $::env(SYNTH_DRIVING_CELL_PIN) [all_inputs]
set cap_load [expr $::env(SYNTH_CAP_LOAD) / 1000.0]
puts "\[INFO\]: Setting load to: $cap_load"
[INFO]: Setting load to: 0.033442
set_load  $cap_load [all_outputs]
puts "\[INFO\]: Setting clock uncertainity to: $::env(SYNTH_CLOCK_UNCERTAINITY)"
[INFO]: Setting clock uncertainity to: 0.25
set_clock_uncertainty $::env(SYNTH_CLOCK_UNCERTAINITY) [get_clocks $::env(CLOCK_PORT)]
puts "\[INFO\]: Setting clock transition to: $::env(SYNTH_CLOCK_TRANSITION)"
[INFO]: Setting clock transition to: 0.15
set_clock_transition $::env(SYNTH_CLOCK_TRANSITION) [get_clocks $::env(CLOCK_PORT)]
puts "\[INFO\]: Setting timing derate to: [expr {$::env(SYNTH_TIMING_DERATE) * 10}] %"
[INFO]: Setting timing derate to: 0.5 %
set_timing_derate -early [expr {1-$::env(SYNTH_TIMING_DERATE)}]
set_timing_derate -late [expr {1+$::env(SYNTH_TIMING_DERATE)}]
min_report

===========================================================================
report_checks -path_delay min (Hold)
============================================================================
Startpoint: rst (input port clocked by clk)
Endpoint: _569_ (removal check against rising-edge clock clk)
Path Group: **async_default**
Path Type: min

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.15    0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (ideal)
                          2.00    2.00 v input external delay
                  0.03    0.02    2.02 v rst (in)
     1    0.00                           rst (net)
                  0.03    0.00    2.02 v _238_/A (sky130_fd_sc_hd__inv_2)
                  0.05    0.05    2.07 ^ _238_/Y (sky130_fd_sc_hd__inv_2)
     4    0.01                           _097_ (net)
                  0.05    0.00    2.08 ^ _239_/A (sky130_fd_sc_hd__buf_1)
                  0.17    0.16    2.24 ^ _239_/X (sky130_fd_sc_hd__buf_1)
     5    0.01                           _095_ (net)
[... 849 lines cut ...]
worst slack 0.24
worst_slack_end
clock_skew

===========================================================================
 report_clock_skew
============================================================================
Clock clk
Latency      CRPR       Skew
_507_/CLK ^
   0.76
_507_/CLK ^
   0.69      0.00       0.07

clock_skew_end
power_report

===========================================================================
 report_power
============================================================================

./run: line 24:   757 Segmentation fault      (core dumped) $OPENROAD_BIN -exit $OR_SCRIPT_0
```

[packaged.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7537471/packaged.tar.gz)

jjcherry56, 2021-11-15:
please rewrite the testcase to not reference shell variables; convert them to tcl variables in the testcase tcl script. I have no way to pass shell variables into my IDE

maliberty, 2021-11-15:
@donn in ORFS we generate a .gdb file which has the necessary setup for gdb (https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/6002fa81a6c646bcc51410ff91c2995d7a5fdbbb/flow/util/utils.mk#L80).  Would you do the same for or_issue?

donn, 2021-11-15:
@jjcherry56 
[RUN_2021.11.15_09.28.40_sta_packaged.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/7540587/RUN_2021.11.15_09.28.40_sta_packaged.zip)

---

@maliberty Is that just a list of `set env`s?

maliberty, 2021-11-15:
@donn yes - they can be sourced in gdb to setup all the envars needed

donn, 2021-11-15:
Sure, I'll throw it in or_issue.

jjcherry56, 2021-11-15:
I use lldb so gdb is not relevant.
They should be tcl variables (which ORFS also generates for issues, albeit broken in many ways).
That way the can be part of the command script and require no external environment.

donn, 2021-11-15:
@jjcherry56 The download I linked is just that, manually converted for expedience. I'm also working on making that an auto-generated file as part of `or_issue`.

jjcherry56, 2021-11-15:
Thanks; that's what I had to do last time. I figured it was time to automate the process so I didn't have to do it everytime.

jjcherry56, 2021-11-16:
fixed

just a heads up that the reason you are seeing failures is because the LEF has pins for liberty pg_pins and in the past it did not. that exposed a bunch of bugs. our version of the sky130 LEF files do not have the well pins.

closed 2021-11-16 by jjcherry56