The-OpenROAD-Project/OpenROAD #1311: GUI feature: Add custom script to menu

mbalestrini, 2021-11-15:
I think it could be useful to have a way to add custom scripts to a menu aside from the toolbar.
It could be directly on `Tools` or maybe some new one specially made for this (ej: `Custom scripts`)

The idea is to be able to build your script library with many useful functions, and there's not much space on the toolbar for that, without using short more criptic names

arlpetergadfort, 2021-11-15:
> We could have a way to add custom menus so that you aren't limited to one user menu.

You mean something line:
```
add_menu -script "Blah" -name "Custom Scripts/Item1"
add_menu -script "Blah" -name "Custom Scripts/Item2"
add_menu -script "Blah" -name "SomethingElse/ItemA"
```

would add two menus to the menu bar `Custom Scripts` and `SomethingElse` with the three items in the respective menus?

maliberty, 2021-11-15:
@arlpetergadfort sure though to be consistent with create_toolbar_button I would use create_menu_item.

mbalestrini, 2021-11-17:
Thanks! I think is really useful

I started using it:
[image]

maliberty, 2021-11-17:
That's cool - would you share your .openroad?  I think others might like that too.

mbalestrini, 2021-11-18:
yes, now that we can load lots of scripts I was thinking of creating a repository of useful .openroad scripts, where people colaborate adding theirs
I'll create that with some of my scripts and I'll share it!

mbalestrini, 2021-12-13:
I was busy but I finally created the repository with the scripts I'm using right now: https://github.com/mbalestrini/openroad_useful_scripts

maliberty, 2021-12-13:
Thanks I'll take a look.