The-OpenROAD-Project/OpenROAD #1324: SEGV during estimate_parasitics -global_routing

antonblanchard, 2021-11-17:
I'm hitting a NULL pointer when running a design through the OpenLane flow:

```
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior ../src/grt/src/GlobalRouter.cpp:288:46 in
AddressSanitizer:DEADLYSIGNAL
=================================================================
==1362927==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000008 (pc 0x0000039f5159 bp 0x7ffe554df0d0 sp 0x7ffe554def90 T0)
==1362927==The signal is caused by a READ memory access.
==1362927==Hint: address points to the zero page.
    #0 0x39f5159 in __gnu_cxx::__normal_iterator<grt::Pin*, std::vector<grt::Pin, std::allocator<grt::Pin> > >::__normal_iterator(grt::Pin* const&) /usr/lib/gcc/x86_64-redhat-linux/11/../../../../include/c++/11/bits/stl_iterator.h:1011:20
    #1 0x39f5159 in std::vector<grt::Pin, std::allocator<grt::Pin> >::begin() /usr/lib/gcc/x86_64-redhat-linux/11/../../../../include/c++/11/bits/stl_vector.h:812:16
    #2 0x3a39b33 in grt::MakeWireParasitics::makePinRoutePts(std::vector<grt::Pin, std::allocator<grt::Pin> >&) /home/anton/OpenROAD/build-debug/../src/grt/src/MakeWireParasitics.cpp:93:17
    #3 0x3a3985b in grt::MakeWireParasitics::estimateParasitcs(odb::dbNet*, std::vector<grt::Pin, std::allocator<grt::Pin> >&, std::vector<grt::GSegment, std::allocator<grt::GSegment> >&) /home/anton/OpenROAD/build-debug/../src/grt/src/MakeWireParasitics.cpp:85:3
    #4 0x39c78d3 in grt::GlobalRouter::estimateRC() /home/anton/OpenROAD/build-debug/../src/grt/src/GlobalRouter.cpp:288:15
    #5 0x1e4d622 in rsz::Resizer::estimateParasitics(rsz::ParasiticsSrc) /home/anton/OpenROAD/build-debug/../src/rsz/src/EstimateWireParasitics.cc:181:21
    #6 0x1e63d54 in rsz::estimate_parasitics_cmd(rsz::ParasiticsSrc) /home/anton/OpenROAD/build-debug/src/rsz/src/CMakeFiles/rsz.dir/ResizerTCL_wrap.cxx:1861:12
    #7 0x1e63d54 in _wrap_estimate_parasitics_cmd(void*, Tcl_Interp*, int, Tcl_Obj* const*) /home/anton/OpenROAD/build-debug/src/rsz/src/CMakeFiles/rsz.dir/ResizerTCL_wrap.cxx:2760:7
    #8 0x7fdfcd84e041 in TclNRRunCallbacks (/lib64/libtcl8.6.so+0x3c041)
    #9 0x7fdfcd853da6  (/lib64/libtcl8.6.so+0x41da6)
    #10 0x7fdfcd8545d6 in Tcl_EvalEx (/lib64/libtcl8.6.so+0x425d6)
    #11 0x7fdfcd8545f9 in Tcl_Eval (/lib64/libtcl8.6.so+0x425f9)
    #12 0x215eb9c in sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) /home/anton/OpenROAD/build-debug/../src/sta/app/StaMain.cc:95:14
    #13 0x19e2ea3 in tclAppInit(int&, char**, char const*, Tcl_Interp*) /home/anton/OpenROAD/build-debug/../src/Main.cc:335:26
    #14 0x7fdfcd903ee5 in Tcl_MainEx (/lib64/libtcl8.6.so+0xf1ee5)
    #15 0x19e3a3d in main /home/anton/OpenROAD/build-debug/../src/Main.cc:222:3
    #16 0x7fdfcbb0eb74 in __libc_start_main (/lib64/libc.so.6+0x27b74)
    #17 0x18f379d in _start (/home/anton/OpenROAD/build-debug/src/openroad+0x18f379d)
```

The test case is too large to attach, but it can be found at: [openroad-estimate_parasitics.tar.gz](https://ozlabs.org/~anton/openroad-estimate_parasitics/openroad-estimate_parasitics.tar.gz)

antonblanchard, 2021-11-17:
Just in case this is another issue where OpenLane is driving things via the low level API, this is what it is doing:

```
cat resizer_routing_timing.tcl
read_liberty "resizer_sky130_fd_sc_hd__tt_025C_1v80.lib"
read_lef "merged_unpadded.lef"
read_def "20-resizer_timing.def"
read_sdc "20-resizer_timing.sdc"

set_dont_use "sky130_fd_sc_hd__a2111oi_0 sky130_fd_sc_hd__a21boi_0 sky130_fd_sc_hd__and2_0 sky130_fd_sc_hd__buf_16 sky130_fd_sc_hd__clkdlybuf4s15_1 sky130_fd_sc_hd__clkdlybuf4s18_1 sky130_fd_sc_hd__fa_4 sky130_fd_sc_hd__lpflow_bleeder_1 sky130_fd_sc_hd__lpflow_clkbufkapwr_1 sky130_fd_sc_hd__lpflow_clkbufkapwr_16 sky130_fd_sc_hd__lpflow_clkbufkapwr_2 sky130_fd_sc_hd__lpflow_clkbufkapwr_4 sky130_fd_sc_hd__lpflow_clkbufkapwr_8 sky130_fd_sc_hd__lpflow_clkinvkapwr_1 sky130_fd_sc_hd__lpflow_clkinvkapwr_16 sky130_fd_sc_hd__lpflow_clkinvkapwr_2 sky130_fd_sc_hd__lpflow_clkinvkapwr_4 sky130_fd_sc_hd__lpflow_clkinvkapwr_8 sky130_fd_sc_hd__lpflow_decapkapwr_12 sky130_fd_sc_hd__lpflow_decapkapwr_3 sky130_fd_sc_hd__lpflow_decapkapwr_4 sky130_fd_sc_hd__lpflow_decapkapwr_6 sky130_fd_sc_hd__lpflow_decapkapwr_8 sky130_fd_sc_hd__lpflow_inputiso0n_1 sky130_fd_sc_hd__lpflow_inputiso0p_1 sky130_fd_sc_hd__lpflow_inputiso1n_1 sky130_fd_sc_hd__lpflow_inputiso1p_1 sky130_fd_sc_hd__lpflow_inputisolatch_1 sky130_fd_sc_hd__lpflow_isobufsrc_1 sky130_fd_sc_hd__lpflow_isobufsrc_16 sky130_fd_sc_hd__lpflow_isobufsrc_2 sky130_fd_sc_hd__lpflow_isobufsrc_4 sky130_fd_sc_hd__lpflow_isobufsrc_8 sky130_fd_sc_hd__lpflow_isobufsrckapwr_16 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_2 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_4 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_4 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_1 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_2 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_4 sky130_fd_sc_hd__mux4_4 sky130_fd_sc_hd__o21ai_0 sky130_fd_sc_hd__o311ai_0 sky130_fd_sc_hd__or2_0 sky130_fd_sc_hd__probe_p_8 sky130_fd_sc_hd__probec_p_8 sky130_fd_sc_hd__xor3_1 sky130_fd_sc_hd__xor3_2 sky130_fd_sc_hd__xor3_4 sky130_fd_sc_hd__xnor3_1 sky130_fd_sc_hd__xnor3_2 sky130_fd_sc_hd__xnor3_4 "

set signal_min_layer "li1"
set signal_max_layer "met5"

set clock_min_layer $signal_min_layer
set clock_max_layer $signal_max_layer

puts "\[INFO]: Setting singal min routig layer to: $signal_min_layer and clock min routing layer to $clock_min_layer. "
puts "\[INFO]: Setting signal max routig layer to: $signal_max_layer and clock min routing layer to $clock_max_layer.  "

set_routing_layers -signal [subst $signal_min_layer]-[subst $signal_max_layer] -clock [subst $clock_min_layer]-[subst $clock_max_layer]

grt::set_capacity_adjustment "0.0"

grt::add_layer_adjustment 1 "0.99"
grt::add_layer_adjustment 2 "0"
grt::add_layer_adjustment 3 "0"
grt::add_layer_adjustment 4 "0"
grt::add_layer_adjustment 5 "0"
grt::add_layer_adjustment 6 "0"

global_route -verbose 3 -congestion_iterations "50"

# set rc values
set ::env(VIAS_RC) " mcon 9.249146E-3, via 4.5E-3, via2 3.368786E-3, via3 0.376635E-3, via4 0.00580E-3"
set ::env(LAYERS_RC) " li1 1.499e-04 7.176e-02, met1 1.449e-04 8.929e-04, met2 1.331e-04 8.929e-04, met3 1.464e-04 1.567e-04, met4 1.297e-04 1.567e-04, met5 1.501e-04 1.781e-05"
set ::env(WIRE_RC_LAYER) "met1"
source set_rc.tcl 

set_propagated_clock [all_clocks]

# estimate wire rc parasitics
estimate_parasitics -global_routing

if { [catch {repair_timing -hold \
    -slack_margin "0.1" \
    -max_buffer_percent "50"} ]} {
    puts "Hold utilization limit is reached. Continuing the flow... "
}

if { [catch {repair_timing -setup \
    -slack_margin "0.05" \
    -max_buffer_percent "50"} ]} {
    puts "Setup utilization limit is reached. Continuing the flow... "
}

# set_placement_padding -global -right $::env(CELL_PAD)
# set_placement_padding -masters $::env(CELL_PAD_EXCLUDE) -right 0 -left 0

detailed_placement

optimize_mirroring

check_placement -verbose

write_def save.def
write_sdc save.sdc

# Run post timing optimizations STA
estimate_parasitics -global_routing

#set ::env(RUN_STANDALONE) 0
#source sta.tcl 
```

maliberty, 2021-11-17:
@donn would you fix " OpenLane is driving things via the low level API"

donn, 2021-11-17:
Uuuh sure

donn, 2021-11-17:
https://github.com/The-OpenROAD-Project/OpenLane/pull/719

maliberty, 2021-11-17:
@donn thanks.  I don't think that will change the underlying problem here for @eder-matheus

jjcherry56, 2021-11-24:
The failure has nothing to do with estimate_parasitics; it fails in the global route command.
I suspect it is fallout from this:
[WARNING GRT-0040] Net vccd1 has wires outside die area.
but it should either error (throw out of the command) or skip the net instead of calling assert and exiting.

closed 2021-12-06 by eder-matheus