The-OpenROAD-Project/OpenROAD #1329: GUI Feature: load a new DEF without having to close the app

mbalestrini, 2021-11-17:
It would be good to be able to load a new DEF or reset the internal database to load new content without having to close the app.

maliberty, 2021-11-17:
I agree.   Do you want to just load a new DEF or a whole new PDK (e.g. lef, lib)?

mbalestrini, 2021-11-18:
I found myself trying to load a new DEF more than loading the whole PDK. But if I had to choose just one option maybe doing a whole reset is more flexible. And can load everything again.

arlpetergadfort, 2021-11-22:
Current ODB just generates an error when attempting to load new DEFs (https://github.com/The-OpenROAD-Project/OpenROAD/blob/f8c243bd63829123a58a9c22112c8d76b681522a/src/odb/src/defin/definReader.cpp#L1630 this is in the form of a printf instead of going to the logger). So that would need to be updated to support replacing the whole design over again (somehow).

The callbacks from ODB probably need to be expended to include information about the impending removal of the design (so all pointers can be cleared before ODB removes them), otherwise, I think the changes to the GUI would be fairly small to support this.