The-OpenROAD-Project/OpenROAD #1332: CTS Segfault

donn, 2021-11-18:
```
OpenROAD 6d56d36324b9ed2e59e8609fff5ddd3089c1c0bf 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  ./tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: ./in.def
[INFO ODB-0128] Design: spm
[INFO ODB-0130]     Created 38 pins.
[INFO ODB-0131]     Created 499 components and 2901 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 1792 connections.
[INFO ODB-0133]     Created 368 nets and 1109 connections.
[INFO ODB-0134] Finished DEF file: ./in.def
###############################################################################
# Created by write_sdc
# Thu Nov 18 09:59:50 2021
###############################################################################
current_design spm
###############################################################################
# Timing Constraints
###############################################################################
create_clock -name clk -period 10.0000 [get_ports {clk}]
set_clock_transition 0.1500 [get_clocks {clk}]
set_clock_uncertainty 0.2500 clk
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {rst}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[0]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[10]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[11]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[12]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[13]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[14]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[15]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[16]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[17]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[18]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[19]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[1]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[20]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[21]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[22]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[23]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[24]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[25]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[26]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[27]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[28]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[29]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[2]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[30]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[31]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[3]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[4]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[5]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[6]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[7]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[8]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {x[9]}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {y}]
set_output_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {p}]
###############################################################################
# Environment
###############################################################################
set_load -pin_load 0.0334 [get_ports {p}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {clk}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {rst}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {y}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {x[31]}]
[... 20 lines cut ...]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {x[10]}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {x[9]}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {x[8]}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {x[7]}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {x[6]}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {x[5]}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {x[4]}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {x[3]}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {x[2]}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {x[1]}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_1 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {x[0]}]
set_timing_derate -early 0.9500
set_timing_derate -late 1.0500
###############################################################################
# Design Rules
###############################################################################
set_max_fanout 5.0000 [current_design]
[INFO]: Setting RC values...
[INFO]: Configuring cts characterization...
[INFO]: Performing clock tree synthesis...
[INFO]: Looking for the following net(s): clk
[INFO]: Running Clock Tree Synthesis...
Segmentation fault (core dumped)
```
[packaged.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7561910/packaged.tar.gz)

jjcherry56, 2021-11-18:
It would be possible for cts to sort this out by looking at the liberty instead of the LEF, but it seems to me that the LEF VNB/VPB pins should have a USE POWER because they definitely are not signal pins, and that is how the are defaulting.

tspyrou, 2021-11-18:
If its impossible to get the lef changes quickly a tcl script using opendb commands could be created to add this property after LEF is read.

jjcherry56, 2021-11-20:
Just to be clear, the LEF for the clock buffers (and any other buffer for that matter) has 3 signal input pins that are only distinguishable by their name. cts is picking the first input because it expects a buffer to only have one input. It needs to find the first input with use type SIGNAL to avoid the well pins, but even that will fail with the LEF in this example.

smunaut, 2021-12-21:
Can someone comment if this is the same issue : ?

(I also get a crash at running CTS, but the comments on this issue make me think there was some out-of-band discussion because I can't see a link between the first comment and the second one so I have no clue if it's the same problem or a distinct one, or how it works at all for some people apparently since they did use the flow ...)

```
(gdb) bt full
#0  0x000055ad8be6b622 in sta::RiseFallMinMax::maxValue (this=0x68, max_value=@0x7ffe9a9efc14: -1.00000002e+30, 
    exists=@0x7ffe9a9efc13: false) at /mnt/sky130/mpw4/openlane_workspace/_build/openroad_app/src/sta/sdc/RiseFallMinMax.cc:187
        mm_index = 0
        rf_index = 0
#1  0x000055ad8bdd4876 in sta::LibertyPort::capacitance (this=0x0)
    at /mnt/sky130/mpw4/openlane_workspace/_build/openroad_app/src/sta/liberty/Liberty.cc:1944
        cap = -1.00000002e+30
        exists = false
#2  0x000055ad8c66b7b3 in cts::TechChar::computeTopologyResults (this=0x55ad90263a30, solution=..., outPinVert=0x55ad922be640, 
    load=4.99999991e-15, setupWirelength=27200) at /mnt/sky130/mpw4/openlane_workspace/_build/openroad_app/src/cts/src/TechChar.cpp:931
        length = 27200
        firstInstLiberty = 0x55ad914defb0
        firstPinLiberty = 0x0
        firstPinCap = -5.46796163e-28
        results = {load = 1.65000039e-13, inSlew = 5.9999998e-11, wirelength = 27200, pinSlew = 5.9999998e-11, 
          pinArrival = 4.10305774e-12, totalcap = 1.7e-13, totalPower = 4.15583366e-12, isPureWire = true, 
          topology = std::vector of length 0, capacity 0}
        totalPower = 4.15583366e-12
        incap = 0
        totalcap = 3.07346792e-41
        pinArrival = -6.5755709e-23
        pinRise = 4.59149455e-41
        pinFall = -1.55407183e-31
        pinSlew = 3.07346792e-41
#3  0x000055ad8c66ce8a in cts::TechChar::create (this=0x55ad90263a30)
    at /mnt/sky130/mpw4/openlane_workspace/_build/openroad_app/src/cts/src/TechChar.cpp:1200
        results = {load = 1.65000039e-13, inSlew = 5.9999998e-11, wirelength = 27200, pinSlew = 5.9999998e-11, 
          pinArrival = 4.10305774e-12, totalcap = 1.7e-13, totalPower = 4.15583366e-12, isPureWire = true, 
          topology = std::vector of length 0, capacity 0}
        solutionKey = {load = 1.65000039e-13, wirelength = 27200, pinSlew = 5.9999998e-11, totalcap = 1.7e-13}
        inputslew = 4.99999998e-12
        __for_range = std::vector of length 12, capacity 16 = {4.99999998e-12, 9.99999996e-12, 1.49999995e-11, 1.99999999e-11, 
          2.50000003e-11, 2.9999999e-11, 3.49999994e-11, 3.99999998e-11, 4.50000003e-11, 5.00000007e-11, 5.50000011e-11, 5.9999998e-11}
--Type <RET> for more, q to quit, c to continue without paging--
        __for_begin = 4.99999998e-12
        __for_end = -0
        load = 4.99999991e-15
        __for_range = std::vector of length 33, capacity 64 = {4.99999991e-15, 9.99999982e-15, 1.49999997e-14, 1.99999996e-14, 
          2.49999996e-14, 2.99999995e-14, 3.49999977e-14, 3.99999993e-14, 4.50000009e-14, 5.00000025e-14, 5.50000041e-14, 
          6.00000057e-14, 6.50000073e-14, 7.00000089e-14, 7.50000105e-14, 8.00000121e-14, 8.50000138e-14, 9.00000154e-14, 
          9.5000017e-14, 1.00000019e-13, 1.0500002e-13, 1.10000022e-13, 1.15000023e-13, 1.20000025e-13, 1.25000027e-13, 
          1.30000028e-13, 1.3500003e-13, 1.40000031e-13, 1.45000033e-13, 1.50000035e-13, 1.55000036e-13, 1.60000038e-13, 
          1.65000039e-13}
        __for_begin = 4.99999991e-15
        __for_end = 3.07346792e-41
        inBTerm = 0x55ad92297f40
        outPin = 0x55ad92297f80
        r1 = 0
        piExists = true
        outBTerm = 0x55ad92297f80
        inPinVert = 0x55ad922be618
        c1 = 0
        lastNet = 0x55ad922947b0
        inPin = 0x55ad92297f40
        outPinVert = 0x55ad922be640
        firstPinLastNet = 0x55ad9229efa0
        c2 = 0
        buffersUpdate = 3
        solution = {netVector = std::vector of length 2, capacity 2 = {0x55ad92294740, 0x55ad922947b0}, 
          nodesWithoutBufVector = std::vector of length 2, capacity 2 = {1, 0}, inPort = 0x55ad922559c4, outPort = 0x55ad922559e0, 
          instVector = std::vector of length 1, capacity 1 = {0x55ad9229bb28}, 
          topologyDescriptor = std::vector of length 2, capacity 2 = {"27200", "sky130_fd_sc_hd__clkbuf_2"}, isPureWire = false}
        __for_range = std::vector of length 2, capacity 2 = {{netVector = std::vector of length 1, capacity 1 = {0x55ad922946d0}, 
            nodesWithoutBufVector = std::vector of length 1, capacity 1 = {1}, inPort = 0x55ad9225598c, outPort = 0x55ad922559a8, 
[... 38 lines cut ...]
No locals.
#8  0x00007f81a8c705f2 in TclNRRunCallbacks () from /lib/x86_64-linux-gnu/libtcl8.6.so
No symbol table info available.
#9  0x00007f81a8c71924 in ?? () from /lib/x86_64-linux-gnu/libtcl8.6.so
No symbol table info available.
#10 0x00007f81a8c71367 in Tcl_EvalEx () from /lib/x86_64-linux-gnu/libtcl8.6.so
No symbol table info available.
#11 0x00007f81a8c724fa in Tcl_Eval () from /lib/x86_64-linux-gnu/libtcl8.6.so
No symbol table info available.
#12 0x000055ad8bdad3bf in sta::sourceTclFile (
    filename=0x7ffe9a9f2b6c "/mnt/sky130/mpw4/openlane_workspace/openlane/scripts/openroad/cts.tcl", echo=false, verbose=false, 
    interp=0x55ad902bd450) at /mnt/sky130/mpw4/openlane_workspace/_build/openroad_app/src/sta/app/StaMain.cc:95
        cmd = "source /mnt/sky130/mpw4/openlane_workspace/openlane/scripts/openroad/cts.tcl"
        code = 32766
[... cut ...]
#13 0x000055ad8baa2e39 in tclAppInit (argc=@0x55ad8fdd6c20: 2, argv=0x7ffe9a9f0788, init_filename=0x55ad8d95f3a8 ".openroad", 
    interp=0x55ad902bd450) at /mnt/sky130/mpw4/openlane_workspace/_build/openroad_app/src/Main.cc:335
        result = 1
        cmd_file = 0x7ffe9a9f2b6c "/mnt/sky130/mpw4/openlane_workspace/openlane/scripts/openroad/cts.tcl"
        threads = 0x0
        exit_after_cmd_file = true
        gui_enabled = false
#14 0x000055ad8baa303e in ord::tclAppInit (interp=0x55ad902bd450)
    at /mnt/sky130/mpw4/openlane_workspace/_build/openroad_app/src/Main.cc:363
```

vijayank88, 2021-12-21:
@smunaut 
share tar file by running
```
make cts_issue
```

smunaut, 2021-12-21:
I'm not sure what you expect this to do ...

This is a self-built bare metal install of openlane, no docker or anything like that, so I have no clue where that 'cts_issue' make target would even be coming from.

maliberty, 2021-12-21:
If you are using openroad-flow-script then that is a make target.  If you are using openlane see https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md.

jjcherry56, 2021-12-21:
It is most certainly not the same issue.

maliberty, 2023-06-14:
The LEF has been fixed so this is moot.

closed 2023-06-14 by maliberty