The-OpenROAD-Project/OpenROAD #1349: Segmentation Violation during Resizer Timing Optimizations

kishan-shanthakumar, 2021-11-23:
[error.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/7585419/error.log)
[bit8alu.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/7585424/bit8alu.txt)
[config.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/7585428/config.txt)

These are the error file, verilog file and config tcl file respectively.

The error message:
[ERROR]: during executing: "openroad -exit /openlane/scripts/openroad/resizer_routing_timing.tcl |& tee >&@stdout /openlane/designs/bit8alubehav/runs/RUN_2021.11.22_19.15.10/logs/routing/12-resizer_timing_optimization.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: segmentation violation

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openlane/designs/bit8alubehav/runs/RUN_2021.11.22_19.15.10/error.log

maliberty, 2021-11-23:
You need to provide a full test case for a crash.  Please see https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

vijayank88, 2021-11-26:
@kishan-shanthakumar 
You need to file this issue wrt openlane flow in https://github.com/The-OpenROAD-Project/OpenLane/issues
Try below solution, if still facing issue raise this issue in openlane page.

design config.tcl you defined 
```
set ::env(CLOCK_PORT) "clk"
```
But your .v file not have clk signal. so set above variable as empty and run the flow.
```
set ::env(CLOCK_PORT) ""
```
Also set only required variables in config.tcl and appropriate values here.

closed 2021-11-26 by vijayank88

vijayank88, 2021-11-26:
@kishan-shanthakumar  use this config.tcl working fine in Openlane flow
```
# User config
set ::env(DESIGN_NAME) bit8alubehav

# Change if needed
set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]

# Fill this
#set ::env(CLOCK_PERIOD) "20.0"
set ::env(CLOCK_PORT) ""
set ::env(PL_TARGET_DENSITY) "0.7"
#set ::env(PL_RANDOM_GLB_PLACEMENT) 1
#set ::env(PL_RANDOM_INITIAL_PLACEMENT) 1
set ::env(FP_CORE_UTIL) 40
set ::env(CLOCK_TREE_SYNTH) 0
#set ::env(PL_RESIZER_TIMING_OPTIMIZATIONS) 0
set ::env(GLB_RESIZER_TIMING_OPTIMIZATIONS) 0
set filename $::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}

```