The-OpenROAD-Project/OpenROAD #1360: [ERROR DRT-0218] Guide is not connected to design

Manarabdelaty, 2021-11-28:
We are getting this warning on some of the sram pins: 
```
[WARNING DRT-1001] No guide in the pin neighborhood
[WARNING DRT-0215] Pin genblk1\[1\].RAM/dout0[22] not covered by guide.
[WARNING DRT-0215] Pin genblk1\[1\].RAM/dout0[22] not covered by guide.
[WARNING DRT-0225] sram_rdata\[1\]\[22\] 2 pin not visited, fall back to feedthrough mode.
[WARNING DRT-0224] sram_rdata\[1\]\[22\] 2 pin not visited, number of guides = 6.
```

Then, the detailed router fails with this error 

`[ERROR DRT-0218] Guide is not connected to design.
`

Tested with openroad commit: `934e0cc13d8d24ca477f5dfd26b64b232065951a`
[sram_droute_packaged.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/7613904/sram_droute_packaged.zip)

maliberty, 2021-11-28:
@eder-matheus I assume the issue is in grt not drt; if you find otherwise please assign it back.

eder-matheus, 2021-11-29:
@Manarabdelaty could you attach the groute_package for this run?

Manarabdelaty, 2021-11-29:
@eder-matheus Here it is
[sram_groute_packaged.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/7620488/sram_groute_packaged.zip)

eder-matheus, 2021-11-29:
@maliberty The pin with the error is not being covered by the guide because the pin geometry intersects with a single track, only at its right boundary. Check the figure below:
[image]

grt is computing the position on the routing grid for this pin based on this single track, and this position generates a guide that is not covering the pin. Check this other figure:
[image]

This is a perfect example of the mismatch between drt pin access and grt "pin to grid". I can workaround this issue by ignoring the track position for IO and macro pins, but a better correlation between grt and drt will still be needed.

eder-matheus, 2021-11-29:
@maliberty When I ignore the position of the tracks for IO and macro pins, the generated guides are valid for TR, which ends with 1 DRCs.

maliberty, 2021-11-29:
@eder-matheus I agree about grt/drt PA (topic for tomorrow's mtg).  
@Manarabdelaty is this urgent or can it wait a while for a better solution?

Manarabdelaty, 2021-11-30:
@maliberty We have a workaround for now to get around this issue (flipping some of the macros helped avoid hitting this error).

jameyhicks, 2022-01-11:
@Manarabdelaty what is the workaround? I've run into this problem also.

tspyrou, 2022-01-12:
@eder-matheus it looks like the workaround is not working, @donn mentioned this today.
Please take a look at this.
A possible workaround might be to run the pin_access command before GR.

eder-matheus, 2022-01-25:
I've fixed a small bug in the `grt` code that uses the access points from `drt`. I've tested the falling design running the `pin_access` command and it is working properly now.

closed 2022-01-25 by eder-matheus