The-OpenROAD-Project/OpenROAD #1361: `estimate_parasitics -global_routing` keeps iterating indefinitely

Manarabdelaty, 2021-11-28:
GRT keeps printing these messages

```
[WARNING GRT-0211] dbGcellGrid already exists in db. Clearing existing dbGCellGrid.
[INFO GRT-0020] Min routing layer: li1
[INFO GRT-0021] Max routing layer: met5
[INFO GRT-0022] Global adjustment: 0%
[INFO GRT-0023] Grid origin: (0, 0)
[WARNING GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0003] Macros: 10
[INFO GRT-0004] Blockages: 1589272
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 2
[INFO GRT-0009] Nets to reroute: 2.
[INFO GRT-0111] Final number of vias: 4
[INFO GRT-0112] Final usage 3D: 14
[WARNING GRT-0211] dbGcellGrid already exists in db. Clearing existing dbGCellGrid.
[INFO GRT-0020] Min routing layer: li1
[INFO GRT-0021] Max routing layer: met5
[INFO GRT-0022] Global adjustment: 0%
[INFO GRT-0023] Grid origin: (0, 0)
[WARNING GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0003] Macros: 10
[INFO GRT-0004] Blockages: 1589276
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 2
[INFO GRT-0009] Nets to reroute: 2.
[INFO GRT-0111] Final number of vias: 4
[INFO GRT-0112] Final usage 3D: 14
[WARNING GRT-0211] dbGcellGrid already exists in db. Clearing existing dbGCellGrid.
```

Then, it prints a message that the message limit reached, and gets stuck at this point for hours

`[INFO GRT-0009] message limit reached, this message will no longer print
`

[wrapper_resizer_routing_timing_packaged.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/7614154/wrapper_resizer_routing_timing_packaged.zip)

shalan, 2021-11-28:
If it runs on a system with low memory (8GB), it crashes after some time with the message: `[ERROR GRT-0183] heap underflow during 3D maze routing.`

shalan, 2021-12-01:
Any update here. It is kinda a blocking issue for designs larger than 20k cells.

maliberty, 2021-12-01:
I see many:
[WARNING GRT-0040] Net vccd1 has wires outside die area.
..
[WARNING GRT-0036] Pin analog_io[0] is outside die area.

warnings.  I don't know if this is the cause but it is unexpected.  Is there a reason for this?

maliberty, 2021-12-01:
@donn the packaging of this test case shows a problem:  LIB_RESIZER_OPT is set in run.sh but not in env.gdb

donn, 2021-12-01:
I'll take a look

maliberty, 2021-12-01:
You are running repair_timing after global_route.  In ORFS we run that after CTS but before global_routing.  Repair_timing after global routing is in development and is not yet recommended.

You have:
[INFO RSZ-0046] Found 6768 endpoints with hold violations.

after each endpoint repair, it calls grt to reroute the buffered net(s), updates estimate_rc and then sta before proceeding to the next endpoint.  You can see this in
[INFO GRT-0009] Nets to reroute: 4.

so it is making progress but very slowly.  This is a known performance problem in this under development work.

maliberty, 2021-12-01:
@donn I worked around it but noticed a bunch more were missing as well.

closed 2021-12-01 by maliberty

donn, 2021-12-01:
@maliberty Yeah I'm going to have to take a look. Most likely there's a file that's getting sourced but just not getting loaded properly by or_issue.py.