The-OpenROAD-Project/OpenROAD #1381: Hardcoding the router for GF14.

QuantamHD, 2021-12-02:
While reviewing some other code I saw this block of code.

https://github.com/The-OpenROAD-Project/OpenROAD/blob/54e3b7dff0eb5570953565b7060514d4838f3b99/src/drt/src/TritonRoute.cpp#L143-L153

It looks like its hardcoding router parameters in the source code which seems very brittle. Can these be moved into the flow scripts instead of in the router?

QuantamHD, 2021-12-02:
@tspyrou @maliberty

maliberty, 2021-12-02:
This is a known problem and @Stephanommg has working on it as a background task.  There are a bunch of other similar cases elsewhere.  Some have already been fixed.

maliberty, 2021-12-02:
The fix in many cases is to identify what is the underlying condition and use that instead (e.g. unidirectional routing constraint, not GF14).

QuantamHD, 2021-12-03:
Even so can't these parameters be set outside the context of the application as flags?

maliberty, 2021-12-05:
These ones can, there are others that can't (like default via).

closed 2022-02-16 by Stephanommg