The-OpenROAD-Project/OpenROAD #1385: estimate_parasitics command crashes with segmentation fault after repair_timing -setup command

dineshannayya, 2021-12-03:
I am using Openlane resizer_routing_timing.tcl to optimize the design.

After  repair_timing -setup execution followed by estimate_parasitics -global_routing command crashes with segmentation fault.

Issue not observed with bypassing setup fix stage.

Run command: run_cmd   OR openroad input/resizer_routing_timing.tcl

</pre>
Error Message:
<pre>[WARNING GRT-0211] dbGcellGrid already exists in db. Clearing existing dbGCellGrid.
[INFO RSZ-0040] Inserted 6 buffers.
[INFO RSZ-0041] Resized 2 instances.
[WARNING RSZ-0062] Unable to repair all setup violations.
Placement Analysis
---------------------------------
total displacement         19.2 u
average displacement        0.0 u
max displacement            4.5 u
original HPWL          181209.5 u
legalized HPWL         186261.6 u
delta HPWL                    3 %

[INFO DPL-0020] Mirrored 2030 instances
[INFO DPL-0021] HPWL before          186261.6 u
[INFO DPL-0022] HPWL after           181221.4 u
[INFO DPL-0023] HPWL delta               -2.7 %
Segmentation fault (core dumped)
</pre>

OpenRoad Version : v2.0-2064-g2f7d6a270 

OpenRoad Git Log

<pre><font color="#C4A000">commit 2f7d6a270eb801b6d2eec0d45d39d0012343cf6f (</font><font color="#06989A"><b>HEAD -&gt; </b></font><font color="#4E9A06"><b>master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/HEAD</b></font><font color="#C4A000">)</font>
Merge: 4be5b9657 ec466d968
Author: Stephanommg &lt;stephanommg@yahoo.com.br&gt;
Date:   Thu Dec 2 19:50:18 2021 -0300

    Merge pull request #1374 from The-OpenROAD-Project-staging/secure-TR_connFix
    
    Connectivity fix

<font color="#C4A000">commit 4be5b96578cdbc297c70681fa0981246bfc4f7f2</font>
Merge: f6c16f5b7 641050ed7
Author: arlpetergadfort &lt;46405338+arlpetergadfort@users.noreply.github.com&gt;
Date:   Thu Dec 2 14:00:39 2021 -0500

    Merge pull request #1375 from The-OpenROAD-Project-staging/gui-select-dbu-um
    
    gui: toggle DBU vs. microns
</pre>

[exp2.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7646505/exp2.tar.gz)

vijayank88, 2021-12-04:
@dineshannayya 
Do you running this project directly in openroad and got this error or using openlane flow you got this error?

dineshannayya, 2021-12-04:
I observed this issue  openlane flow (With latest efabless/openlane docker) and I have created standalone testcase to cross-check and see same issue observed with latest openroad tools also ..

vijayank88, 2021-12-04:
@dineshannayya  possible to share your config.tcl and source RTL

dineshannayya, 2021-12-04:
[config.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/7654042/config.txt)

The project database is at : https://github.com/dineshannayya/logic_bist,  

Openlane run use to fails for me **wb_host** block

1. You need to change the docker name and Tag in openlane.Makefile .. As i have done some hack on docker
OPENLANE_TAG = mpw4
OPENLANE_IMAGE_NAME = dineshannayya/openlane:$(OPENLANE_TAG)

2. Run wb_host at openlane folder.

eder-matheus, 2021-12-06:
@dineshannayya I was able to workaround the segfault by calling the global_route command before the last "estimate_parasitics -global_routing" command. Apparently, the internal structures of grt are not being updated after the repair_timing commands.

I'm working to fix this issue, but adding the global_route command as I mentioned should solve the issue temporarily.

maliberty, 2021-12-06:
In general we don't recommend using repair_timing after global_routing.  This is an area that is under development and isn't ready for production usage.  At best it will be very slow to run.  (FYI @donn )

eder-matheus, 2021-12-07:
@dineshannayya Please, check the latest version of the master branch. I've just merged a fix for this issue.

closed 2021-12-07 by eder-matheus