The-OpenROAD-Project/OpenROAD #1394: error when trying to build pdn component

proppy, 2021-12-07:
When attempting to build the `pdn` openroad component from `master` or `v2.0`, I get the following error:
```
Scanning dependencies of target pdn
make[2]: Leaving directory '/content/OpenROAD'
make  -f src/pdn/src/CMakeFiles/pdn.dir/build.make src/pdn/src/CMakeFiles/pdn.dir/build
make[2]: Entering directory '/content/OpenROAD'
...
[ 32%] Linking CXX static library pdn.a
cd /content/OpenROAD/src/pdn/src && /content/miniconda-openroad/bin/cmake -P CMakeFiles/pdn.dir/cmake_clean_target.cmake
cd /content/OpenROAD/src/pdn/src && /content/miniconda-openroad/bin/cmake -E cmake_link_script CMakeFiles/pdn.dir/link.txt --verbose=1
/content/miniconda-openroad/bin/gcc-ar qc pdn.a CMakeFiles/pdn.dir/CMakeFiles/pdn.dir/PdnGenTCL_wrap.cxx.o CMakeFiles/pdn.dir/pdn-tclInitVar.cc.o CMakeFiles/pdn.dir/PdnGen.cc.o CMakeFiles/pdn.dir/MakePdnGen.cc.o
/content/miniconda-openroad/bin/gcc-ranlib pdn.a
cd /content/OpenROAD/src/pdn && /content/OpenROAD/etc/find_messages.py > /content/OpenROAD/src/pdn/src/../messages.txt
Error: PDN 0053 used 2 times
Error: PDN 0054 used 2 times
Error: PDN 0053 used 2 times
Error: PDN 0054 used 2 times
src/pdn/src/CMakeFiles/pdn.dir/build.make:154: recipe for target 'src/pdn/src/pdn.a' failed
```
is this error meant to be fatal?

maliberty, 2021-12-07:
Yes the error is meant to be fatal but it doesn't occur in the CI build.  I am guessing you have some extraneous file with messages in it (like a copy of PdnGen.tcl).

maliberty, 2021-12-07:
try "grep -r 54 src/pdn" and see what files come up.

vijayank88, 2023-06-14:
@proppy 
Is that required any further action?