The-OpenROAD-Project/OpenROAD #1395: Build issue

ratul619, 2021-12-07:
Hi , 

Trying to build openroad , but build stops with below erors : 

Any idea ? Anybody faced them previously ?

Thanks !

Steps : 

1. sudo  ./etc/DependencyInstaller.sh -run
2. sudo ./etc/DependencyInstaller.sh -dev
3. sudo ./etc/Build.sh

[ 97%] Linking CXX executable openroad
cd /home/ratul619/Desktop/OpenROAD/build/src && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/openroad.dir/link.txt --verbose=1
/usr/bin/g++  -O3 -DNDEBUG   CMakeFiles/openroad.dir/OpenRoad.cc.o CMakeFiles/openroad.dir/Main.cc.o  -o openroad ifp/src/ifp.a openroad_swig.a gpl/gpl.a dpl/dpl.a fin/fin.a rsz/src/rsz.a ppl/ppl.a stt/stt.a dbSta/src/dbSta.a ../../src/sta/app/libOpenSTA.a odb/src/db/libodb.a odb/src/swig/tcl/odbtcl.a rcx/src/rcx.a rmp/src/rmp.a cts/src/cts.a grt/grt.a tap/src/tap.a gui/gui.a drt/drt.a mpl/mpl.a mpl2/mpl2.a psm/src/psm.a ant/src/ant.a par/par.a utl/utl.a pdn/src/pdn.a /usr/lib/x86_64-linux-gnu/libtcl.so -pthread /usr/lib/x86_64-linux-gnu/libtclreadline.so /usr/lib/x86_64-linux-gnu/libz.so _openroad_swig_py.a odb/src/swig/python/_odbpy.a /usr/lib/x86_64-linux-gnu/librt.so ppl/src/munkres/libMunkres.a rsz/src/rsz.a grt/grt.a dpl/dpl.a ant/src/ant.a grt/src/fastroute/libFastRoute4.1.a stt/stt.a /usr/lib/gcc/x86_64-linux-gnu/9/libgomp.so /usr/lib/x86_64-linux-gnu/libpthread.so mpl/src/ParquetFP/libParquetFP.a par/src/MLPart/libMLPart.a dbSta/src/dbSta.a gui/gui.a /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8 /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8 /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8 ../../src/sta/app/libOpenSTA.a odb/src/db/libodb.a odb/src/defout/libdefout.a odb/src/lefout/liblefout.a odb/src/tm/libtm.a odb/src/zlib/libzlib.a /usr/lib/x86_64-linux-gnu/libtcl.so odb/src/cdl/libcdl.a odb/src/swig/common/libodb_swig_common.a odb/src/defin/libdefin.a odb/src/def/libdef.a odb/src/def/libdefzlib.a odb/src/lefin/liblefin.a utl/utl.a /usr/lib/x86_64-linux-gnu/libspdlog.so.1.5.0 -pthread odb/src/zutil/libzutil.a odb/src/lef/liblef.a odb/src/lef/liblefzlib.a /usr/lib/x86_64-linux-gnu/libz.so /usr/lib/x86_64-linux-gnu/libpython3.8.so 
/usr/bin/ld: CMakeFiles/openroad.dir/OpenRoad.cc.o: in function `ord::OpenRoad::setThreadCount(int, bool)':
OpenRoad.cc:(.text+0xe03): undefined reference to `fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<char> >, char>::iterator fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: OpenRoad.cc:(.text+0xe3c): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: OpenRoad.cc:(.text+0x108a): undefined reference to `fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<char> >, char>::iterator fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: OpenRoad.cc:(.text+0x10ca): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'

maliberty, 2021-12-07:
On what OS?  Which compiler/version?

ratul619, 2021-12-07:
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0

Thanks

maliberty, 2021-12-07:
@vvbandeira any guesses?

vvbandeira, 2021-12-07:
We build on Ubuntu 20.04 with GCC in our CI successfully.

Make sure your workspace is clean: `git clean -d -x -i && git submodule foreach --recursive git clean -d -x -i`

Are you using conda environment or anything similar? If so, deactivate them.

ratul619, 2021-12-07:
Using virtualbox with ubuntu installation. I have refired the installation with the git clean commands as provided. 
Lets see. 
Thanks

ratul619, 2021-12-07:
Thanks @maliberty  & @vvbandeira  -- built cleanly now

ib.a ../odb/src/zutil/libzutil.a /usr/lib/x86_64-linux-gnu/libtcl.so /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8 /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8 /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8 ../../../src/sta/app/libOpenSTA.a ../utl/utl.a /usr/local/lib/libspdlog.a -lpthread /usr/lib/gcc/x86_64-linux-gnu/9/libgomp.so /usr/lib/x86_64-linux-gnu/libpthread.so /usr/lib/x86_64-linux-gnu/libz.so 
make[2]: Leaving directory '/home/ratul619/Desktop/OpenROAD/build'

[100%] Built target trTest
make[1]: Leaving directory '/home/ratul619/Desktop/OpenROAD/build'
/usr/local/bin/cmake -E cmake_progress_start /home/ratul619/Desktop/OpenROAD/build/CMakeFiles 0

real	45m50.624s
user	42m37.661s
sys	1m46.062s

closed 2021-12-07 by ratul619