The-OpenROAD-Project/OpenROAD #1419: Diode insertion appears to insert diodes to illegal positions

donn, 2021-12-13:
OpenROAD e9bd11509b1d907cbc3205e17f699ef05ae2181c via OpenLane

```
[etc]
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 2 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
[INFO GRT-0012] Antenna violations: 13
[INFO GRT-0015] 13 diodes inserted.
[WARNING DPL-0005] Overlap check failed (4).
[ERROR DPL-0033] detailed placement checks failed.
Error: groute.tcl, 74 DPL-0033
```

Reproducible:
[groute.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7705111/groute.tar.gz)

tspyrou, 2021-12-13:
@eder-matheus if this is a detailed placement problem please let me know and I will pass it along to cherry.

maliberty, 2021-12-13:
I have a fix for this that I'm testing now.  The problem is antenna repair only looks at global padding not padding by master type.

maliberty, 2021-12-13:
Fixed with https://github.com/The-OpenROAD-Project/OpenROAD/pull/1421

closed 2021-12-13 by maliberty