The-OpenROAD-Project/OpenROAD #1424: openroad should be distributed as a regular python package

proppy, 2021-12-14:
Currently the documented way to interact with the openroad python package is thru `openroad -python` (thru https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/odb/src/swig/python/main.cpp?), it would be great if it was also distributed as a regular python package on [PyPI](https://pypi.org/) and/or [conda-eda](https://github.com/hdl/conda-eda) so that developers can import it from a regular python interpreter without relying on the `openroad` binary, ex:
```
$ python
>>> import openroad
```

maliberty, 2021-12-14:
We used to support odb in python as you describe but have never tried to support all of openroad.   I suspect it is possible but a fair amount of work to do and support (e.g. we don't build shared libraries currently).  Would you describe how this would be advantageous over "openroad -python"?

proppy, 2021-12-15:
>  Would you describe how this would be advantageous over "openroad -python"?

It does make it easier to use openroad from other python projects or from a jupyter notebook.

> but have never tried to support all of openroad

I wonder if the remaining part could be bridged with https://docs.python.org/3/library/tkinter.html, it does seems to allow TCL from python using: https://docs.python.org/3/library/tkinter.html#tkinter.Tcl

maliberty, 2021-12-15:
I don't think that will help as we need to have the tcl interpreter object in c++ in order to register our swig wrapper on it.  We could make a command to forward to tcl though that would be the ugliest interface possible.

proppy, 2021-12-17:
> I don't think that will help as we need to have the tcl interpreter object in c++ in order to register our swig wrapper on it.

Ah yes, you're right those must be two different Tcl context. 

> that would be the ugliest interface possible.

Ack, out of the command listed under https://openroad.readthedocs.io/en/latest/main/README.html#run, what are the ones that are heavily dependant on a non-trivial amount of TCL to be usable, and therefore not trivially expose-able to python thru the existing SWIG interface?

maliberty, 2021-12-17:
> Ack, out of the command listed under https://openroad.readthedocs.io/en/latest/main/README.html#run, are there ones that are heavily dependent on a non-trivial amount of TCL to be usable? and therefore not trivially expose-able to python thru the existing SWIG interface?

pdngen & icewall are totally in tcl so they are the worst.  Most others to argument handling in tcl and then drop to c++.  Reading sdc implicitly requires tcl (SDC is a set of commands in tcl per its definition) - I think we will have to have a behinds the scenes interpreter for read_sdc.

proppy, 2021-12-20:
another thing to consider could be to  have something similar to http://yowasp.org/ (OpenROAD compiled as a WASM bundle using https://github.com/WebAssembly/wasi-sdk) to enable cross platform distribute on PyPI.

maliberty, 2022-02-21:
In the future UPF support will also require TCL for the same reason as SDC.
I looked at the wasi docs and it seem rather immature with lots of caveats.  I don't want to add extra headaches.

I'm still unclear if you are proposing an identical interface in python as tcl or a more programmatic one.

proppy, 2022-02-21:
To go back to my original issue https://github.com/The-OpenROAD-Project/OpenROAD/issues/1424#issue-1079742993, I'm just suggesting that we support importing the current python binding outside of the python interpreter embedded in the openroad process (`from openroad import odb` from a regular `python` process), we can discuss the scope of those bindings in a separate issue #1659.

mithro, 2022-05-18:
FYI  - You will need to build OpenROAD against the "manylinux" platforms - https://github.com/pypa/manylinux#docker-images

mithro, 2022-05-18:
@umarcor has been helping us get GitHub Actions running on a bunch of other repositories.

proppy, 2022-05-19:
Looks like https://cibuildwheel.readthedocs.io/en/stable/ can also help achieve this.

proppy, 2022-05-19:
Both are being maintained under @pypa so that should be future safe to depends on it.

umarcor, 2022-05-19:
@proppy pypi-publish is actually a shell wrapper calling twine: https://github.com/pypa/gh-action-pypi-publish/blob/master/twine-upload.sh. Nowadays, it's not worth using a Container Action for such task. A Reusable Workflow or a Composite Action is enough. See https://github.com/pyTooling/Actions/#context.

umarcor, 2022-05-19:
> The @pypa action does seems to have additional knobs to target different PyPI repository (ex: test PyPI) and run twine check to verify the package metadata before uploading.

That's a feature of twine. See:

- https://github.com/pypa/gh-action-pypi-publish/blob/master/twine-upload.sh#L51-L54
- https://github.com/VUnit/vunit/blob/master/.github/workflows/push.yml#L168-L174
- https://github.com/pyTooling/Actions/blob/main/.github/workflows/PublishOnPyPI.yml#L70-L80

> Maybe it would worth filling an issue on https://github.com/pypa/gh-action-pypi-publish to suggest that they use composite action over containerized one? And then @pyTooling could simply reuse/extend it.

Might be interesting. So far pyTooling has been focused on building, testing, and generating docs/reports, rather than covering the wides range of platforms. Since building is out of scope of pypi-publish, pyTooling might need to be enhanced with e.g. manylinux before using pypi-publish (or specific options of twine, see ).

/cc @Paebbels

Paebbels, 2022-05-19:
Adding a `twine check` call is a welcome addition to pyTooling. Is signing the package with PGP also required? @umarcor, I think we should support this too, right?

Is there also a need for [twine verify-upload](https://pypi.org/project/twine-verify-upload/)?

But for supporting the PyPI test repository, I don't see a use case. What additional knowledge would be gained from a pipeline step uploading to that repository?

proppy, 2022-05-19:
> Since building is out of scope of pypi-publish, pyTooling might need to be enhanced with e.g. manylinux before using pypi-publish (or specific options of twine).

It seems that @pypa maintains https://github.com/pypa/cibuildwheel for this, so maybe pyTooling could be refactored as a composite action that uses both?

proppy, 2022-05-19:
Let's move the meta discussion to https://github.com/pyTooling/Actions/issues/47 and keep this issue about the specifics of the OpenROAD implementation.

maliberty, 2022-05-23:
Is this moving towards a conclusion?

maliberty, 2022-05-23:
I have no experience with pip or its configuration files, etc.  We don't package at all today.  Can you give a more specific set of steps?

umarcor, 2022-05-23:
Do not care about "pip" for now. Let's focus on "Python setup files". Some examples:

- https://github.com/chipsalliance/f4pga/blob/main/f4pga/setup.py#L72-L97
- https://github.com/cocotb/cocotb/blob/master/setup.py#L88-L136
- https://github.com/edaa-org/pyEDAA.ProjectModel/blob/main/setup.py#L41-L48

So, conceptually it's very similar to any other "recipe" (say deb, rpm, conda...). You declare the sources that belong to the "package", the "entrypoints" you want to generate, and other metadata (author, license, compatible versions, etc.).

I believe that the "shell wrappers" of f4pga can be inspiring to you:

- We picked some shell/bash scripts: https://github.com/chipsalliance/f4pga/tree/main/f4pga/wrappers/sh/xc7.
- We created a "wrapper" in Python which has one function for each scripts (https://github.com/chipsalliance/f4pga/blob/main/f4pga/wrappers/sh/__init__.py).
- We created "entrypoints" in the Python package, each pointing to one wrapper function, which calls the corresponding shell script:
  - https://github.com/chipsalliance/f4pga/blob/main/f4pga/setup.py#L95
  - https://github.com/chipsalliance/f4pga/blob/main/f4pga/setup.py#L55-L70
  
Now, when users install f4pga through pip, they get "binaries" such as `symbiflow_pack` created "somewhere in the system available in the PATH". At the same time, the python sources and shell scripts are installed "somewhere in the system where python libs are installed". The `symbiflow_pack` "binary" calls the `?/wrappers/sh/__init__.py`, which executed `?/wrappers/sh/xc7/pack.f4pga.sh`.

In the context of openroad, you might do something similar. That would provide a not-ideal but working solution to install openroad through pip. From there on, the Python wrapper functions can be reworked to use tkinter or some other "more elegant" solution than blinding calling an "external" TCL script as a system tool.

proppy, 2022-05-24:
A bit more background:
The goal of https://github.com/The-OpenROAD-Project/OpenROAD/pull/1849, is to expose a subset of the OpenROAD C++ API as a regular [python extension](https://docs.python.org/3/extending/extending.html), and not really to replace the main OpenROAD executable with a Python entrypoint.

In addition to the example that @umarcor pointed to for the "Python setup files", https://packaging.python.org/en/latest/tutorials/packaging-projects/  is also a good resources for getting started with Python packaging.

So the steps are likely to be:
a. write the required `setup.py/cfg`, `pyproject.toml` metadata
b. bridge the python build process with cmake using something like https://github.com/diegoferigo/cmake-build-extension
c. integrate with a cross-platform build GH action like https://github.com/pypa/cibuildwheel and https://github.com/pyTooling/Actions to build the wheel for all supported platform on each commits
d. integrate with a publishing GH action like https://github.com/pypa/gh-action-pypi-publish or https://github.com/pyTooling/Actions to publish to PyPI prod and/or test at the desired release cadence.

More than happy to contribute/help navigate those steps, maybe we could already start the work with the existing `odb` binding and update it include the newer binding when #1849 lands, wdyt?

maliberty, 2022-05-25:
@proppy I'm happy to have you start with odb