The-OpenROAD-Project/OpenROAD #1428: Segfault in global routing stage

donn, 2021-12-15:
OpenROAD 174a460d88e87f538ac36e26ef82d0bba0f4182c via OpenLane

For context, this issue only shows up when we update Yosys. I'm not sure what feature yosys may have added to expose this.

```
[etc]
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 3 T shapes.
Notice 0: Split top of 1 T shapes.
./run.sh: line 37:  2772 Segmentation fault      (core dumped) $OPENROAD_BIN -exit $OR_SCRIPT_0
```

[groute.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7719479/groute.tar.gz)

eder-matheus, 2021-12-22:
@donn I've notice you are using `li1` as the min routing layer for this technology. It is more likely to have loops between wires and pin shapes when we have segments in `li1`. Changing the min layer to `met1` "fixed" the segfault.

maliberty, 2021-12-22:
@donn I thought the min routing layer was already fixed?

donn, 2021-12-22:
@maliberty This issue was filed a week ago- the minimum layer was fixed since yes. I'm going to re-test yosys now.

QuantamHD, 2021-12-22:
@eder-matheus We definitely shouldn't segfault at minimum we should tolerate loops and error out.

eder-matheus, 2021-12-22:
> @eder-matheus We definitely shouldn't segfault; at minimum we should tolerate loops and error out.

We already discussed this segfault. This happens during order wires, and this piece of code is pretty hard to understand. I can try to update rrepair_antennas to detect loops and error out

donn, 2021-12-22:
Yeah, according to Matt order_wires is a nightmarish piece of code.

As this has a workaround, we can just leave this issue open but deprioritize it.

QuantamHD, 2021-12-22:
This is bad for users we need to at least trap the segfault and notify the user that if they're on a low metal layer this can cause an issue.

eder-matheus, 2021-12-22:
@donn maybe we can close https://github.com/The-OpenROAD-Project/OpenROAD/issues/1111, since they appear to be duplicated.

donn, 2021-12-22:
sure, why not.

eder-matheus, 2022-02-24:
@donn My last update in grt (https://github.com/The-OpenROAD-Project/OpenROAD/pull/1668) fixes the crash on this testcase, even using `li1` as min routing layer. I will close the issue for now, but please let me know if you face any other problems.

closed 2022-02-24 by eder-matheus