The-OpenROAD-Project/OpenROAD #1430: Segmentation violation during global routing

kauser-rl, 2021-12-15:
This is my first issue I'm reporting so bare with me. I'm using Openlane tag `2021.12.15_04.14.23` to trial our design and hit a segmentation violation during the global routing phase. The message dumped by Openlane is-

```
during executing: "openroad -exit /openlane/scripts/openroad/groute.tcl |& tee >&@stdout /openlane/designs/<my_design>/runs/f_15dec2021_1224/logs/routing/18-global.log"
Last 10 lines:
child killed: segmentation violation
```

The log from `18-global.log` is attached.
Note: I have removed some confidential information about our design.

Let me know what information I can provide. I won't be able to unfortunately send across the design database.
[18-global.log.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/7719917/18-global.log.txt)

kauser-rl, 2021-12-15:
I looked at the logs from #1428. While the point at which the fail occurs appears to be the same, it is impossible for me to say it is the same. I'm using the current version of Openlane (which uses an older version of Yosys), whereas #1428 integrates a newer version of the synthesis tool. The two designs aren't the same either. The one I'm attempting to run is a much larger design.

kauser-rl, 2021-12-15:
I don't mind waiting for #1428 to be fixed first so I can re-run on the fixed version to see if I had hit the same issue.

eder-matheus, 2021-12-22:
@kauser-rl As I mentioned in #1428, using the min layer as `li1` can be the cause of the segfault. Could you try your design with min layer as `met1`?

eder-matheus, 2022-02-24:
@kauser-rl My latest update on grt has fixed #1428 segfault, that is probably the same issue you were facing. Could you try the latest commit on OpenROAD? If you still see the segfault, feel free to open this issue again. Thanks!

closed 2022-02-24 by eder-matheus