The-OpenROAD-Project/OpenROAD #1437: padding on DECAPs creates overlap error with tapcells

msaligane, 2021-12-17:
`Summary`
While running the temperature sensor design, I got a new error related to how padding is used. Padding seems to be applied to `sky130_fd_sc_hd__decap_4`  cell and the tool thinks there is an overlap with the tapcells.

**Observed behavior:** 
Design fails at detail placement:

```
[INFO DPL-0020] Mirrored 63 instances
[INFO DPL-0021] HPWL before            5213.7 u
[INFO DPL-0022] HPWL after             5133.0 u
[INFO DPL-0023] HPWL delta               -1.5 %
[WARNING DPL-0005] Overlap check failed (7).
 TAP_119 overlaps PHY_13
 TAP_123 overlaps PHY_17
 TAP_128 overlaps PHY_25
 TAP_133 overlaps PHY_33
 TAP_138 overlaps PHY_41
 TAP_143 overlaps PHY_49
 TAP_146 overlaps PHY_53
[ERROR DPL-0033] detailed placement checks failed.
Error: detail_place.tcl, 121 DPL-0033
0:00.48elapsed 99%CPU 84476memKB
make[1]: *** [results/sky130hd/tempsense/3_4_place_dp.def] Error 1
make[1]: Leaving directory `/shared/OpenFASOC/generators/temp-sense-gen/flow’
```

[image]

**Expected Behavior**

Design should not fail at detail placement. My work around is to remove padding for the decap cells as follows:
`set_placement_padding -left 0 -right 0 -masters “sky130_fd_sc_hd__decap_4”` in tapcell.tcl
`set_placement_padding -left 0 -right 0 -masters “sky130_fd_sc_hd__decap_4”` in detail_place.tcl

Test case is provided here
[floorplan_tempsense_sky130hd__2021-12-17_03-43.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7731939/floorplan_tempsense_sky130hd__2021-12-17_03-43.tar.gz)

rovinski, 2021-12-17:
DPL checks based on the LEF MACRO class. For `sky130_fd_sc_hd__decap_4`, the class is `CORE` which is probably not correct. It should be `CORE SPACER`.

QuantamHD, 2021-12-18:
I'm happy to update the PDK rather than trying to work around it.

maliberty, 2021-12-18:
@rovinski why should decap be a SPACER?  It isn't a fill cell.  sky130_fd_sc_hd__fill_* are marked as SPACER.
@msaligane is there a reason to use decap rather than fill?

rovinski, 2021-12-18:
@maliberty It would be nice if there was a LEF class for decap but there isn't. Decap cells don't have any pins except power/ground, so they fit the same profile as filler cells. Most (all?) commercial PDKs I surveyed mark their decaps cells as such (`CORE SPACER`).

maliberty, 2021-12-18:
@rovinski I see you are correct.  @QuantamHD can you update the PDK?