The-OpenROAD-Project/OpenROAD #1470: error: no template named 'any_io_executor' in namespace 'boost::asio'; did you mean 'is_executor'?

yurivict, 2021-12-25:
Build fails with clang-13:
```
In file included from /disk-samsung/freebsd-ports/cad/openroad/work/.build/src/dst/CMakeFiles/dst.dir/DistributedTCL_wrap.cxx:1701:
/disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0-2346-g95a1df82f/src/dst/include/dst/Distributed.h:48:46: error: no template named 'any_io_executor' in namespace 'boost::asio'; did you mean 'is_executor'?
typedef asio::basic_stream_socket<tcp, asio::any_io_executor> socket;
```

Rev. 95a1df82f

maliberty, 2021-12-26:
What version of boost are you using?

yurivict, 2021-12-26:
boost-libs-1.72.0_6

iruhm, 2021-12-26:
For me the build also failed on Ubuntu 20.04.3 LTS

$ cat /usr/include/boost/version.hpp
...
#define BOOST_VERSION 107100
...

maliberty, 2021-12-27:
Could you try changing in src/dst/include/dst/Distributed.h:

```
#if BOOST_VERSION >= 107000
typedef asio::basic_stream_socket<tcp, asio::any_io_executor> socket;
#else
typedef asio::basic_stream_socket<tcp> socket;
#endif

``` 
to

```
#if BOOST_VERSION >= 107400
typedef asio::basic_stream_socket<tcp, asio::any_io_executor> socket;
#if BOOST_VERSION >= 107000
typedef asio::basic_stream_socket<tcp, asio::executor> socket;
#else
typedef asio::basic_stream_socket<tcp> socket;
#endif

```
any let me know if that works for you?

maliberty, 2021-12-27:
Actually I think the whole thing could just be reduced to
`typedef asio::basic_stream_socket<tcp> socket;
`
as I think the second arg is properly defaulted in all versions.  Would you try that?

yurivict, 2021-12-27:
Replacing with ```typedef asio::basic_stream_socket<tcp> socket; ``` helped.

maliberty, 2021-12-27:
fixed with https://github.com/The-OpenROAD-Project/OpenROAD/pull/1475

closed 2021-12-27 by maliberty