The-OpenROAD-Project/OpenROAD #1476: Build fails: error: duplicate symbol: memUsedPeak()

yurivict, 2021-12-27:
OpenRoad bundles ```abc``` and fails with the same error ```abc``` fails with:
```
ld: error: duplicate symbol: memUsedPeak()
>>> defined at System.cpp
>>>            CMakeFiles/libabc.dir/src/sat/glucose/System.cpp.o:(memUsedPeak())
>>> defined at System2.cpp
>>>            CMakeFiles/libabc.dir/src/sat/glucose2/System2.cpp.o:(.text+0x50)
```

clang-13

```abc``` bug [report](https://github.com/berkeley-abc/abc/issues/146).

QuantamHD, 2021-12-27:
Strong disagree on relying on the system abc installation. Different system abcs can have widely circuit results from commit to commit, and that make reproducibility hard when engineers might be using slightly different versions of ABC when building the software. 

IMO static linking for this kind of compiler has distinct advantages in terms of supportability and reproducibility.

maliberty, 2021-12-27:
@QuantamHD  abc is statically linked afaik.

QuantamHD, 2021-12-27:
I was trying to argue in favor of a statically linked and pinned version of abc for OpenROAD. I think potentially modifying the users system installation/relying on what's installed would be a mistake. Which I believe is what @yurivict is suggesting, but I could be totally misunderstanding what @yurivict is trying to suggest.

yurivict, 2021-12-27:
> I was trying to argue in favor of a statically linked and pinned version of abc for OpenROAD.

It is always beneficial to some degree to pin dependency versions, but bundling causes exponential compilation and maintenance costs. Bugs in each package need to be fixed in all bundled versions. Same software is recompiled a potentially unlimited number of times.

QuantamHD, 2021-12-28:
I think it makes sense to rely on the system packages for something like spdlog or other very stable dependencies, but heuristic based software like ABC has the potential to completely change its results from commit to commit due to small changes in its cost functions without changing its underlying API. That isn't very desirable for this kind of compiler/CAD software.

The additional compilation time is generally a one time cost for most users, and is likely much less time in aggregate than solving dependency issues that only show up because they have a particular version of ABC installed. There are also many issues that are difficult to solve when relying on system packages.

1. Build machines that are managed by another organization in a large company may make it impossible to install a compatible version of ABC.
2. People may rely on a particular version of ABC for other use cases that are incompatible with OpenROADs needs, and the users might not have the right relationships in their organization to implement a solution. This software is targeted at both software engineers as well as hardware engineers after all. 
3. ABC has the potential to heavily impact the final performance of the output, and is updated frequently with new algorithms that increase the performance even further.  Meaning that users on particularly old OSes will see very degraded results that reflect poorly on the project as a whole.

I think for those reasons it makes sense to bundle and pin. Though I understand that you might weigh different cost matrices than I do; especially as a package maintainer.

maliberty, 2021-12-28:
I don't get the debate here - OR has abc as a submodule so we aren't relying on any abc from the distro.  That wasn't reported as an issue.  I still don't know how to reproduce the issue reported.

yurivict, 2021-12-28:
> I still don't know how to reproduce the issue reported.

The issue is likely with incorrect ifdef __FreeBSD__ around the function in question because the issue is on FreeBSD.

maliberty, 2021-12-28:
As I've said before - OR doesn't support freebsd and doesn't plan to.  You are welcome to submit PRs that improve things on FreeBSD but we won't spend time investigating issues related to FreeBSD.  If abc fixes this up stream we are glad to pick it up.

closed 2021-12-28 by maliberty

yurivict, 2021-12-28:
I'm just trying to provide packages for open source EDA software using which users would be able to easily install this software.
As it stands today FreeBSD ports is the only packaging system that can keep up with many version updates in many projects.
The alternative is for all users to rebuild all projects by hand, which is a lot more difficult and error prone.

maliberty, 2021-12-28:
Its a laudable goal but we lack sufficient developers to devote to it.  Another alternative is to use docker which is what openlane does.