The-OpenROAD-Project/OpenROAD #1480: `global_route` does not support its documented arguments

donn, 2021-12-28:
OpenROAD 719c6dc617d4b8d6896500bb41cdf057864f9e09

* Issue:

`global_route` does not support its documented arguments, opting to show this error:
```
$ echo "global_route -verbose 3" | openroad -exit
OpenROAD v2.0-2364-g719c6dc61 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[ERROR STA-0406] global_route positional arguments not supported.
STA-0406
```

* Probable Cause: https://github.com/The-OpenROAD-Project/OpenROAD/commit/110c23c2565b07e460c4f4ab7d3cd4e68ddfd5f7

* Reproducible:
- `echo "global_route -verbose 3" | openroad -exit`

donn, 2021-12-28:
Critical for OpenLane, btw

eder-matheus, 2021-12-28:
@donn I've updated the `-verbose` option to be a flag. We don't have the levels now, that's why you have the failure. I've created a PR updating the `-verbose` documentation.

donn, 2021-12-28:
Thanks. I've updated the invocation

closed 2021-12-28 by vvbandeira