The-OpenROAD-Project/OpenROAD #1505: PDN Question: How to handle multiple power domain connections ?

Manarabdelaty, 2022-01-04:
In this design, we have a macro `user_proj_example` which has four power domains:  `vccd1` `vccd2` `vdda1` `vdda2`. The issue we are having is that pdngen shorts all the power domains together when it connects the `user_proj_example` macro to the top level pdn. 

For example, here the four pdn stripes inside `user_proj_exmaple` are shorted together. 
[image]

[user_project_wrapper_pdn_packaged.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/7807497/user_project_wrapper_pdn_packaged.zip)

maliberty, 2022-01-04:
@msaligane do you have any comments?  I guess you have similar designs in fasoc.

msaligane, 2022-01-05:
I think it is more like having multi vdd rings than voltage domain here?

Manarabdelaty, 2022-01-05:
This wasn't an issue with a very old versions of openroad (for example: `1415572a73d43dfc91ba4501170fe6a245729e87`) because we were able to tell pdngen which pins to connect on the macro by specifying the `power_pins` / `ground_pins` keys like: 
```
set macro {
    orient {R0 R180 MX MY R90 R270 MXR90 MYR90}
    power_pins $::env(VDD_NET)
    ground_pins $::env(GND_NET)
    blockages $::env(MACRO_BLOCKAGES_LAYER)
    straps {
    }
    connect {{$::env(FP_PDN_LOWER_LAYER)_PIN_ver $::env(FP_PDN_UPPER_LAYER)}}
}
```

But, it looks like pdngen is now ignoring the `power_pins` / `ground_pins` keys.

maliberty, 2022-01-05:
We use power_pins & ground_pins  so it is unlikely they are ignored.  Perhaps there is some change in behaviour.

tspyrou, 2022-01-12:
@abk-openroad fyi for Colin.

Colin-Holehouse, 2022-01-13:
The pdngen code was updated to not require the specification of power_pins and ground_pins, insterad the code would look at all the pins on the macro for signal types POWER and GROUND. This works in the case where all POWER pins are to be connected to the same POWER net and all GROUND pins are to be connected to the same GROUND net.

Clearly that is not the requirement here. With the older code base, specifying a subset of the power/ground pins resulted in just the subset beinug connected to the POWER/GROUND nets.

The correct behavior should be for pdngen to check for pins on the macro connected to the logical net before making the physical connection.

The changes I am currently working on will cause pdngen to operate in this way, but will rely on the add_global_connection to be done correctly to create the required logical connection. [Taking a look at the script it seems like this is done correctly already].

I will add a link to the PR once it is submitted.

kareefardi, 2022-02-21:
Hello. Any update on this?

mithro, 2022-02-21:
I believe that @msaligane has done something like this in his OpenFASoC designs?

msaligane, 2022-02-21:
@kareefardi we are currently working on this with @minghungumich and @Colin-Holehouse 

However, we have enabled 2 voltage domains in our sensors, ldo using custom scripts to workaround issues. 

The issue you filed doesn't provide me with the whole picture of what you are trying to do. Can you please help me understand or provide a testcase of what you're trying to do?