The-OpenROAD-Project/OpenROAD #1509: grt: algorithm changes based on execution time

antonblanchard, 2022-01-04:
Global routing has some strange code which measures the time taken to execute a chunk of code, and then makes decisions based on it:

```
const clock_t t1 = clock();
```
Do a bunch of stuff...
```
  const clock_t t2 = clock();
  const float gen_brk_Time = (float) (t2 - t1) / CLOCKS_PER_SEC;
```
...
```
  if (gen_brk_Time < 60) {
    ripupTH3D = 15;
  } else if (gen_brk_Time < 120) {
    ripupTH3D = 18;
  } else {
    ripupTH3D = 20;
  }
```

Very strange, especially since the execution time is not normalized. What if my CPU is 10x faster, or compiler optimisations speed the code up 10x?

maliberty, 2022-01-05:
Yuck - let's just drop that and use 20 (checking CI of course).  Even that is pretty arbitrary but at least is isn't crazy.

QuantamHD, 2022-01-05:
This might be the source ofthr non determinism we've been seeing in Google blocks.

antonblanchard, 2022-01-05:
PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/1513 created to fix this.

eder-matheus, 2022-01-06:
PR with the fiz merged.

closed 2022-01-06 by eder-matheus