The-OpenROAD-Project/OpenROAD #1512: drt: macro pin access creates DRC issue within macro

antonblanchard, 2022-01-05:
I'm having issues routing out of a macro. There is lots of room around the macro:

[image]

However some of the macro pins are accessed from within the macro (I assume planar access to the pin, if I understand the terminology). Looking at part of the post routed def below, the middle pin is not being accessed at the bottom edge of the macro, but we drop a via all the way from met5 to met2 within the macro:

[image]

There are obstructions in the macro LEF at every level, apart from the pins. I do notice some of the pins extend well within the macro, perhaps contributing to the issue (in green, pin 9 and 10 from the right, on the bottom of the macro):

[image]

The LEF backs this up. The bad pins extend a long way into the macro:

```
  PIN Di0[53]
    DIRECTION INPUT ;
    USE SIGNAL ;
    PORT
      LAYER met2 ;
        RECT 939.480 0.000 939.620 110.470 ;
    END
  END Di0[53]
  PIN Di0[54]
    DIRECTION INPUT ;
    USE SIGNAL ;
    PORT
      LAYER met2 ;
        RECT 956.960 0.000 957.100 62.170 ;
    END
  END Di0[54]
```

And the next pin along, just a little:

```
  PIN Di0[55]
    DIRECTION INPUT ;
    USE SIGNAL ;
    PORT
      LAYER met2 ;
        RECT 974.440 0.000 974.580 2.450 ;
    END
  END Di0[55]
```

Even so, it seems bad that detailed routing is dropping a via through many obstructed layers.

Working on packaging up a test case.

antonblanchard, 2022-01-05:
The macro was created by magic's `lef write -hide` which creates an abstract LEF, with only pins and obstructions.

antonblanchard, 2022-01-05:
I hand modified the macro LEF so that the pins are all the same short length and the issues went away. I had to rerun global routing because detailed routing complained about pin access (I guess the original pin was so long it must cross grid cells, and global routing chose the end inside the macro).

What to make of this? It seems like magic isn't helping us, but should global and detailed routing be a bit more intelligent about choosing where to connect to a pin?

antonblanchard, 2022-01-05:
Test case: [macro-pin-access.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7814186/macro-pin-access.tar.gz)

maliberty, 2022-01-05:
I appreciate the pictures but it is also helpful to have instance/pin names to know exactly what we are talking about.

maliberty, 2022-01-06:
What were your global route settings to reproduce your guide file?

antonblanchard, 2022-01-06:
This adds the global route step to the above tarball: [macro-pin-access-grt.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7818642/macro-pin-access-grt.tar.gz)

I'll find some of the offending pin names.

maliberty, 2022-01-06:
I think this is the general problem in grt that it doesn't know about pin access.  It is just picking the middle of the shape as the access point and since it is so large a shape drt doesn't look far enough out to find a detour.

We have some work to move pin access earlier in the flow so that it's results are available to grt.  I think that will solve this issue once it lands (hopefully soon but it needs a bunch of testing).
@eder-matheus FYI

maliberty, 2022-01-06:
I can test my theory once I have your grt setup.

antonblanchard, 2022-01-06:
Thanks @maliberty that makes sense. In parallel I will look at why magic is creating such large pins (it seems to merge the rest of the metal on the layer into the pin).

If I use the ORFS scripts to convert the def to lef I get small pins. This also gives me a workaround for now.

maliberty, 2022-01-06:
another possible workaround is macro pin extension though I haven't tried it.

maliberty, 2022-01-06:
@eder-matheus I ran the grt provided and with "draw_route_guides {soc0.processor.icache_0._014_\[53\]}" I see the guide does go down through a fully blocked m3 layer.  That is surprising to me.  Would you look at that?  I know we don't do via congestion but even so we have to consider that a via requires a shape and there has to be less than 100% blockage.

maliberty, 2022-01-06:
@eder-matheus interestingly adding pin_access doesn't help

"set_macro_extension 2" does work as a workaround to get a better guide.

eder-matheus, 2022-01-26:
> @eder-matheus interestingly adding pin_access doesn't help
> 
> "set_macro_extension 2" does work as a workaround to get a better guide.

It is possible that the pin_access didn't help due to the bug that I've fixed this week. I've tried the design with the latest master branch and the guide going through the m3 layer is not there anymore. Here is the comparison:

Previous global route:
[image]

Global route + pin access:
[image]

Even so, the final detailed route ends with 32 DRCs. I didn't look at them yet, but at least the net you've mentioned before seems to be working with the pin_access cmd.

maliberty, 2022-01-26:
Great - I'm glad PA fixes the known issue.  Are you looking at the 32 drcs?

eder-matheus, 2022-02-02:
@maliberty @antonblanchard Sorry for the delay on it. I've run the design again with the latest OpenROAD (where we have the update for the macro obstructions) and with the pin_access cmd before the global route. Below is a brief summary of the state of this design:

- GRT ends with congestion. Here is the final congestion report:
[image]
And here is the congestion map:
[image]
Note that even when I remove the layer adjustments, GRT ends with congestion.

- DRT ends with 25 violations, all of them near macros. Here is a figure of DRC viewer:
[image]

As you can see, all the metal spacing violations happen between a net and the VDD, and we also have 9 short violations with the VSS. The remaining 12 violations are shorts with the macro obstructions.

eder-matheus, 2022-02-02:
Here is the tarball with the final routing and the DRC report. You can load it with the script `load_final_dr.tcl`.
[1512_final_dr.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7989557/1512_final_dr.tar.gz)

maliberty, 2022-02-02:
@eder-matheus is your tar with layer adjustments on or off?

eder-matheus, 2022-02-02:
> @eder-matheus is your tar with layer adjustments on or off?

Adjustments on.

antonblanchard, 2022-02-02:
I was trying to reproduce the DRC issues, using a recent checkout (0192cb7a083e) and I hit this:

```
[INFO DRT-0195] Start 5th optimization iteration.
    Completing 10% with 1774 violations.
    elapsed time = 00:00:00, memory = 24062.58 (MB).
    Completing 20% with 1774 violations.
    elapsed time = 00:00:00, memory = 24062.58 (MB).
    Completing 30% with 1432 violations.
    elapsed time = 00:00:25, memory = 24062.58 (MB).
    Completing 40% with 1432 violations.
    elapsed time = 00:00:26, memory = 24062.58 (MB).
    Completing 50% with 1193 violations.
    elapsed time = 00:00:39, memory = 24062.58 (MB).
    Completing 60% with 1193 violations.
    elapsed time = 00:00:39, memory = 24062.58 (MB).
    Completing 70% with 1193 violations.
    elapsed time = 00:00:40, memory = 24062.58 (MB).
    Completing 80% with 962 violations.
    elapsed time = 00:00:48, memory = 24062.58 (MB).
    Completing 90% with 962 violations.
    elapsed time = 00:00:49, memory = 24062.58 (MB).
    Completing 100% with 612 violations.
    elapsed time = 00:01:08, memory = 24062.58 (MB).
Error: soc0.processor.icache_0._014_\[28\] 1 pin not visited #guides = 37
Error: checkConnectivity break, net soc0.processor.icache_0._014_\[28\]
Objs not visited:
frPathSeg: begin (1569290 274210 ) end ( 1569290 277950 ) layerNum 6
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (1597350 192610 ) end ( 1601490 192610 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (1601490 183940 ) end ( 1601490 192610 ) layerNum 6
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (1569290 274210 ) end ( 1597350 274210 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (1597350 192610 ) end ( 1597350 274210 ) layerNum 6
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2094380 205700 ) end ( 2094380 219300 ) layerNum 10
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2094380 202300 ) end ( 2094380 205700 ) layerNum 12
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2094380 202300 ) end ( 2099900 202300 ) layerNum 12
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2099900 200260 ) end ( 2099900 202300 ) layerNum 10
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2166140 183940 ) end ( 2166140 200260 ) layerNum 10
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2116690 200260 ) end ( 2116690 222700 ) layerNum 6
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2116690 219300 ) end ( 2116690 222700 ) layerNum 12
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2116690 200940 ) end ( 2117150 200940 ) layerNum 6
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2117150 200260 ) end ( 2117150 200940 ) layerNum 6
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2117150 200260 ) end ( 2117610 200260 ) layerNum 6
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2099900 200260 ) end ( 2116690 200260 ) layerNum 8
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2094380 219300 ) end ( 2116690 219300 ) layerNum 12
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (1601490 183940 ) end ( 2166140 183940 ) layerNum 8
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (2117610 200260 ) end ( 2166140 200260 ) layerNum 8
	beginStyle: 1
	endStyle: 1
frVia: at ( 1569290 277950 )
VIA DEF:
VIA L1M1_PR DEFAULT
  RECT -85 -85 85 85
  RECT -85 -85 85 85
  RECT -145 -115 145 115
frVia: at ( 1569290 277950 )
VIA DEF:
VIA M1M2_PR DEFAULT
  RECT -160 -130 160 130
  RECT -75 -75 75 75
  RECT -130 -160 130 160
frVia: at ( 1569290 274210 )
VIA DEF:
VIA M1M2_PR DEFAULT
  RECT -160 -130 160 130
  RECT -75 -75 75 75
  RECT -130 -160 130 160
frVia: at ( 1597350 192610 )
VIA DEF:
VIA M1M2_PR DEFAULT
  RECT -160 -130 160 130
  RECT -75 -75 75 75
  RECT -130 -160 130 160
frVia: at ( 1601490 192610 )
VIA DEF:
VIA M1M2_PR DEFAULT
  RECT -160 -130 160 130
  RECT -75 -75 75 75
  RECT -130 -160 130 160
frVia: at ( 1601490 183940 )
VIA DEF:
VIA M2M3_PR DEFAULT
  RECT -140 -185 140 185
  RECT -100 -100 100 100
  RECT -165 -165 165 165
frVia: at ( 1597350 274210 )
VIA DEF:
VIA M1M2_PR DEFAULT
  RECT -160 -130 160 130
  RECT -75 -75 75 75
  RECT -130 -160 130 160
frVia: at ( 2166140 183940 )
VIA DEF:
VIA M3M4_PR DEFAULT
  RECT -190 -160 190 160
  RECT -100 -100 100 100
  RECT -165 -165 165 165
frVia: at ( 2094380 219300 )
VIA DEF:
VIA M4M5_PR DEFAULT
  RECT -590 -590 590 590
  RECT -400 -400 400 400
  RECT -710 -710 710 710
frVia: at ( 2094380 205700 )
VIA DEF:
VIA M4M5_PR DEFAULT
  RECT -590 -590 590 590
  RECT -400 -400 400 400
  RECT -710 -710 710 710
frVia: at ( 2099900 202300 )
VIA DEF:
VIA M4M5_PR DEFAULT
  RECT -590 -590 590 590
  RECT -400 -400 400 400
  RECT -710 -710 710 710
frVia: at ( 2099900 200260 )
VIA DEF:
VIA M3M4_PR DEFAULT
  RECT -190 -160 190 160
  RECT -100 -100 100 100
  RECT -165 -165 165 165
frVia: at ( 2166140 200260 )
VIA DEF:
VIA M3M4_PR DEFAULT
  RECT -190 -160 190 160
  RECT -100 -100 100 100
  RECT -165 -165 165 165
frVia: at ( 2116690 200260 )
VIA DEF:
VIA M2M3_PR DEFAULT
  RECT -140 -185 140 185
  RECT -100 -100 100 100
  RECT -165 -165 165 165
frVia: at ( 2116690 222700 )
VIA DEF:
VIA M2M3_PR DEFAULT
  RECT -140 -185 140 185
  RECT -100 -100 100 100
  RECT -165 -165 165 165
frVia: at ( 2116690 222700 )
VIA DEF:
VIA M3M4_PR DEFAULT
  RECT -190 -160 190 160
  RECT -100 -100 100 100
  RECT -165 -165 165 165
frVia: at ( 2116690 222700 )
VIA DEF:
VIA M4M5_PR DEFAULT
  RECT -590 -590 590 590
  RECT -400 -400 400 400
  RECT -710 -710 710 710
frVia: at ( 2117610 200260 )
VIA DEF:
VIA M2M3_PR DEFAULT
  RECT -140 -185 140 185
  RECT -100 -100 100 100
  RECT -165 -165 165 165
INSTTERM: (INST/CELL/TERM/NET) _44171_ sky130_fd_sc_hd__mux2i_4 Y soc0.processor.icache_0._014_\[28\]
Segmentation fault (core dumped)
Stream error: Input/Output error
```

maliberty, 2022-02-02:
@eder-matheus would you provide the router inputs so I can see what's happening in pa.

@antonblanchard would you provide a testcase tgz for your crash

eder-matheus, 2022-02-02:
> @eder-matheus would you provide the router inputs so I can see what's happening in pa.
> 
> @antonblanchard would you provide a testcase tgz for your crash

Here they are. Just untar them into the directory.
[router_inputs.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7990200/router_inputs.tar.gz)

antonblanchard, 2022-02-02:
@maliberty I just reran the original test case above to hit that fail:

```
wget https://github.com/The-OpenROAD-Project/OpenROAD/files/7814186/macro-pin-access.tar.gz
wget https://github.com/The-OpenROAD-Project/OpenROAD/files/7818642/macro-pin-access-grt.tar.gz
tar xzf macro-pin-access.tar.gz
tar xzf macro-pin-access-grt.tar.gz
cd macro-pin-access
openroad -exit global_route.tcl
openroad -exit detail_route.tcl
```

antonblanchard, 2022-02-14:
Thanks @eder-matheus and @maliberty. I'm not seeing any issues now.

closed 2022-02-14 by antonblanchard