The-OpenROAD-Project/OpenROAD #1514: sta: report_checks clock reconvergence pessimism bug

mbalestrini, 2022-01-05:
Running the same report_checks gives different results after calling another report_checks between them:

```report_checks .... -through mprj/wrapped_quad_pwm_fet_drivers_5```   >>> ```-0.69    clock reconvergence pessimism```
```report_checks .... -through mprj/wrapped_quad_pwm_fet_drivers_5/_3655_```   >>> ```-0.69    clock reconvergence pessimism```
```report_checks .... -through mprj/wrapped_quad_pwm_fet_drivers_5```   >>> ```0.00    clock reconvergence pessimism```

Test case: 
https://videogamo.com/tmp/asic/sta_crpr_issue.zip

running `sta sta_script.tcl` you can see the different cprp results

jjcherry56, 2022-01-05:
The variable $::env(PDK_ROOT) is not set in the script and the pdk data is not in the test case.

maliberty, 2022-01-05:
@mbalestrini it is best to use https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md to package the test case.

mbalestrini, 2022-01-05:
Didn't know about or_issue.py
I'll use it and upload the result test case

mbalestrini, 2022-01-05:
I think this or_issue.py script is oriented to an openlane flow error. This error came up running a custom sta script with the results from different flows (caravel, user_project_wrapper), and doesn't use .def files
I made a new zip the includes the pdk files needed: https://videogamo.com/tmp/asic/sta_crpr_issue_with_pdk.zip

jjcherry56, 2022-01-08:
This is really an opensta issue and has nothing to do with openroad.
Fixed by
e6d5e2d7d Merge pull request #1527 from jjcherry56/sta_report_thru

closed 2022-01-08 by jjcherry56