The-OpenROAD-Project/OpenROAD #1518: grt creates guides under obstructed layers when it doesn't need to

antonblanchard, 2022-01-06:
I'm getting a short near a macro where there is loads of free routing space nearby:

[image]

I think the problem is that the macro has a slightly smaller obstruction on met1 vs all other layers. The route guide for `_012772_` looks like this (I added the macro obstructions to this picture too):

[image]

and `net153` looks like:

[image]

So the guides for both nets are in a region that has only `met1` unobstructed, hence the short.

It seems we should penalise global routing cells that have obstructed layers (and very heavily penalise those that only have one layer).

antonblanchard, 2022-01-06:
Test case: [grt-track-under-obstruction.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7821047/grt-track-under-obstruction.tar.gz)

maliberty, 2022-01-06:
We have seen something similar with chameleon.  This area under the shelf of m2 allows for no errors in track assignment as there is no way to uncross wires.  In that case we worked around it with macro extension.

We already remove capacity for blockages so I don't think we need to penalize extra for that.  We do need to penalize for cases where you are restricted to planar routing due to a blockage above.  Note that initial routing is done in a 2d plane so there is no notion of blockage above.

@eder-matheus I think we should look at reducing the capacity on the 1st routing layer if the 2nd routing layer is obstructed.  We would have to carry that through 3d routing as well since there are available m5 tracks here.

maliberty, 2022-01-06:
@antonblanchard if you want to tackle this yourself and give a PR you are welcome to.  It will probably be a bit before we can get it fixed (Eder is out next week).

antonblanchard, 2022-01-12:
@maliberty Thanks, I'm still trying to dig myself out of some other bugs right now. Will update the issue if I do get some time.

eder-matheus, 2022-01-26:
> We have seen something similar with chameleon. This area under the shelf of m2 allows for no errors in track assignment as there is no way to uncross wires. In that case we worked around it with [...]

@maliberty Should this be applied only in the case of stacked routing obstructions (like in macros)? Just to make sure, the idea is to "extend" the 1st routing layer blockage so it is aligned with the 2nd layer blockage? In this way, we would avoid these areas where the detailed router cannot detour.

maliberty, 2022-01-26:
@eder-matheus yes extend to blockage either literally or by removing all capacity from the m1 gcells that are blocked above completely .

eder-matheus, 2022-01-28:
@maliberty @antonblanchard I've implemented an initial version of the fix for this issue, and the guides for the net mentioned above (net `_012772`) looks like this after my fix:
[image]

antonblanchard, 2022-01-28:
Much better. Thank you @eder-matheus!

closed 2022-02-01 by eder-matheus