The-OpenROAD-Project/OpenROAD #1528: OpenROAD GUI: is it possible to generate by script a jpeg or png from a selection?

iruhm, 2022-01-09:
I have this small TCL script and I would like to directly script the jpeg or png generation to get the figure with the highlighted nets
 
for {set i 0} {$i < 10} {incr i} {
    select -type Net -name clknet_$i* -highlight $i
}

maliberty, 2022-01-09:
See https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/scripts/save_images.tcl for examples

vijayank88, 2023-06-14:
@iruhm 
Please refer latest OpenROAD gui features. Still you need any help on this?

titan73, 2024-09-03:
I've tested saving an image with highlighted nets using the code example:
```
set nets {cd d cp q}
for {set i 0} {$i < [llength $nets]} {incr i} {
    select -type Net -name [lindex $nets $i] -highlight $i
}

set height [ord::dbu_to_microns [[[ord::get_db_block] getBBox] getDY]]
set resolution [expr $height / 1000]
save_image -resolution $resolution design.png
```

I got a image nets highlighted:

[image]

I think we can close this issue.

closed 2024-09-03 by maliberty