The-OpenROAD-Project/OpenROAD #1530: Compile error with newer boost version

piegamesde, 2022-01-10:
```
/build/source/src/TritonRoute/src/frBaseTypes.h:331:20: error: 'segment' in namespace 'boost::geometry::model' does not name a template type
  331 | typedef bg::model::segment<point_t> segment_t;
      |                    ^~~~~~~
```

See also https://github.com/The-OpenROAD-Project/OpenROAD/issues/866

The latest boost version known to work is 1.69, the earliest version reported not to work is 1.75.

Please fix compilation with newer boost versions. If you do not plan on doing this anytime soon, then please pin down the dependency so that it already fails during configuration.

vijayank88, 2022-01-10:
@piegamesde 
new boost version required is 1.76
```
./etc/DependencyInstaller.sh -help

Usage: ./etc/DependencyInstaller.sh -run[time]      # installs dependencies to run a pre-compiled binary
       ./etc/DependencyInstaller.sh -dev[elopment]  # installs dependencies to compile the openroad binary
```

maliberty, 2022-01-10:
Platform integrators complain about pinned versions

QuantamHD, 2022-01-10:
We may want to throw an error on known bad versions of boost or bundling a known good version.

piegamesde, 2022-01-11:
> Platform integrators complain about pinned versions

Bad wording on my side. I meant specifying the supported version range in the build system. It is better to have an error message early on that says "you are using a version we don't support" than getting a random compile error and trying to figure out its cause. But I see that this can be rather difficult, especially if the dependency does not follow SemVer.

-----

I tried building latest master, with little success. I had to configure boost for static compilation and set `-Wno-error` to get somewhere at all. Now it fails after a few minutes without apparent error message. These are the last log lines:

```
[ 29%] Linking CXX static library liblefin.a
cd /build/source/build/src/odb/src/lefin && /nix/store/10cjb32rn1rsjhbqxbxa77s4s1p62gbc-cmake-3.22.1/bin/cmake -P CMakeFiles/lefin.dir/cmake_clean_target.cmake
cd /build/source/build/src/odb/src/lefin && /nix/store/10cjb32rn1rsjhbqxbxa77s4s1p62gbc-cmake-3.22.1/bin/cmake -E cmake_link_script CMakeFiles/lefin.dir/link.txt --verbose=1
/nix/store/cdm6zywd51mbabxhklsixwcskv4n70s3-binutils-2.35.2/bin/ar qc liblefin.a CMakeFiles/lefin.dir/lefin.cpp.o CMakeFiles/lefin.dir/reader.cpp.o CMakeFiles/lefin.dir/lefTechLayerSpacingEolParser.cpp.o CMakeFiles/lefin.dir/lefTechLayerMinStepParser.cpp.o CMakeFiles/lefin.dir/lefTechLayerCornerSpacingParser.cpp.o CMakeFiles/lefin.dir/lefTechLayerSpacingTablePrlParser.cpp.o CMakeFiles/lefin.dir/lefTechLayerRightWayOnGridOnlyParser.cpp.o CMakeFiles/lefin.dir/lefTechLayerRectOnlyParser.cpp.o CMakeFiles/lefin.dir/lefTechLayerCutClassParser.cpp.o CMakeFiles/lefin.dir/lefTechLayerCutSpacingParser.cpp.o CMakeFiles/lefin.dir/lefTechLayerCutSpacingTableParser.cpp.o CMakeFiles/lefin.dir/lefTechLayerCutEnclosureParser.cpp.o CMakeFiles/lefin.dir/lefTechLayerTypeParser.cpp.o CMakeFiles/lefin.dir/lefTechLayerEolExtensionParser.cpp.o CMakeFiles/lefin.dir/lefTechLayerEolKeepOutRuleParser.cpp.o
/nix/store/cdm6zywd51mbabxhklsixwcskv4n70s3-binutils-2.35.2/bin/ranlib liblefin.a
make[2]: Leaving directory '/build/source/build'
[ 29%] Built target lefin
make[1]: Leaving directory '/build/source/build'
make: *** [Makefile:139: all] Error 2
builder for '/nix/store/kfy59bg9v293v7kbcznyq6isa292vskx-OpenROAD.drv' failed with exit code 2
```

maliberty, 2022-01-11:
We don't test with each and every boost version so its hard to know what is supported beyond the one we do test with.   Usually some #ifdef can solve it if you dig a bit.  Unfortunately boost changes their APIs from time to time.

vvbandeira, 2022-11-28:
Closing due to lack of activity. Please open a new issue in case the problem is still relevant and occurring with the latest commit.

Note that we test with boost 1.76, and is known to work with boost up to 1.80.

closed 2022-11-28 by vvbandeira