The-OpenROAD-Project/OpenROAD #1535: odb: SITE statement is allowed in both tech and library LEF

antonblanchard, 2022-01-12:
I tried importing the sky130 PDK with separate `openroad -tech` and `openroad -library` commands, but it failed because the tech LEF has a `SITE` statement, and odb assumes it can only appear in a library LEF. The spec suggests it is allowed in both tech and library LEF.

Not a high priority, since ORFS and Openlane create a merged LEF and load it in one go.

maliberty, 2022-01-12:
@osamahammad21 not urgent as Anton notes but good for a background task.

osamahammad21, 2022-02-01:
@maliberty In the current model, the site ownership belongs to dbLib. Should it be moved to dbTech?

maliberty, 2022-02-01:
@antonblanchard what did you mean by "openroad -tech" and "openroad -library"?  The openroad executable does not take those command line options?  In general ORFS does load in two steps:

  read_lef $::env(TECH_LEF)
  read_lef $::env(SC_LEF)

osamahammad21, 2022-02-01:
@maliberty I think he means read_lef -tech/-lib

maliberty, 2022-02-01:
@antonblanchard if you do
```
read_lef -tech -library sky130_fd_sc_hd.tlef
read_lef -library sky130_fd_sc_hd_merged.lef
```
it works as does:
```
read_lef sky130_fd_sc_hd.tlef
read_lef sky130_fd_sc_hd_merged.lef
```
only 
```
read_lef -tech sky130_fd_sc_hd.tlef
read_lef -library sky130_fd_sc_hd_merged.lef
```
fails.   In ORFS we don't use either -tech or -library and I don't really think they are very useful.  I'd rather just deprecate them unless you see a use case where it matters.

antonblanchard, 2022-10-03:
> In ORFS we don't use either -tech or -library and I don't really think they are very useful. I'd rather just deprecate them unless you see a use case where it matters.

(Going through some of my older bugs). I agree with deprecating the options, I don't see much point in them.

vijayank88, 2023-06-14:
@maliberty 
With `read_lef` I'm not getting any switch for `-tech` or `-library`. Can we close this as resolved?

maliberty, 2023-06-14:
I see the switches still exist in the code so I'm not sure what "I'm not getting any switch" means.

closed 2026-06-14 by github-actions[bot]