The-OpenROAD-Project/OpenROAD #1549: write_sdf writing single (IOPATH A Z (0.00:0.00)) for some instance pins, and correct (IOPATH B Z (1.23::1.23) (1.23::1.23)) for others of same cell type

Mark-at-Apex, 2022-01-13:
We are starting to use OpenSTA fairly successfully, reading about 210K lines of verilog with about 514 standard cell types, with delay models using rise and fall resistances and pin capacitances for each.

We are trying to run gate-level simulations, with sdf annotation, written using OpenSTA's write_sdf command, and it is seemingly working successfully, except that about 6% of the IOPATH lines in the SDF have the (IOPATH A Z (0.00::0.00)) format, and then simulate as zero delay.  The real issue is that none of our cells have resistance and intrinsic delays which can result in 0.00, the intrinsic delays are all 0.1, and all the resistances are positive > 0 and all the capacitances are positive > 0.  Further proof of a programming issue is that
a frequently used CELLTYPE, such as SC_AND2_100_60  will sometimes have valid SDF delays like this:

(CELL
  (CELLTYPE "SC_AND2_100_60")
  (INSTANCE MI1373/MI1493/X43758)
  (DELAY
     (ABSOLUTE
       (IOPATH A1 Z (3.51::3.51) (3.09::3.09))
       (IOPATH A2 Z (3.51::3.51) (3.09::3.09))
       )
    )
)

but then many other instances will have SDF delays like this:

(CELL
  (CELLTYPE "SC_AND2_100_60")
  (INSTANCE MI1373/MI1490/X46078)
  (DELAY
     (ABSOLUTE
       (IOPATH A1 Z (0.00::0.00))
       (IOPATH A2 Z (0.00::0.00))
       )
    )
)

When the liberty file entry for this cell type looks like this:
cell ( SC_AND2_100_60 ) {
        area                         : 1              ;
        auxillary_pad_cell    : false        ;
        bus_name_style      : "%s[%d]" ;
        dont_touch             : true         ;
        dont_use                : true         ;
        interface_timing     : true         ;
        pad_cell                  : false        ;
        pin ( A1 ) {
                      direction  : input       ;
                      capacitance: 2.032    ;
        }
        pin ( A2 ) {
                      direction  : input       ;
                      capacitance: 2.032    ;
        }
        pin ( Z ) {
                      direction : output     ;
                      function  : A1     & A2 ;
                      timing () {
                            intrinsic_rise   : 0.1 ;
                            intrinsic_fall    : 0.1 ;
                            rise_resistance: 1.68 ;
                            fall_resistance:  1.86666666667 ;
                            related_pin    :  A1 ;
                     }
                      timing () {
                            intrinsic_rise   : 0.1 ;
                            intrinsic_fall    : 0.1 ;
                            rise_resistance: 1.68 ;
                            fall_resistance:  1.86666666667 ;
                            related_pin    :  A2 ;
                     }
         }
}

This is admittedly a detailed issue, but having 6% of the SDF delay values clearly wrong at 0.00 is hard to cope with, and so I am wondering if this is a known bug, or if there is a version with this fixed?  If not, I am happy to provide more data to help debug the issue.

Mark Vancura

Mark-at-Apex, 2022-01-14:
Not sure how this got closed by me, I am new to all the buttons here, I must have pushed one I didnt realize.

Mark-at-Apex, 2022-01-14:
Thanks Tom, there is virtually no SDC in force at least as we start the write_sdf command, we are "learning" the circuti, and don't know yet how to apply the SDC, plus it is a very asynchronous circuit.  But I can look at report_calc for sure.  I wonder too if there are "automatic" ways that timing arcs are getting disabled

Mark-at-Apex, 2022-01-14:
The results from report_dcalc look like this:
% help report_dcalc
report_dcalc [-from from_pin] [-to to_pin] [-corner corner_name] [-min] [-max]
   [-digits digits]
% report_dcalc -from MI1373/MI1370/X105613/A -to MI1373/MI1370/X105613/ZN -digits 2 -max
Library: NXP7105_first_liberty
Cell: SC_INV_100_60
Arc sense: negative_unate
Arc type: combinational
A ^ -> ZN v
Delay = 0.10 + 1.87 * 3.29 = 6.24
.............................................

A v -> ZN ^
Delay = 0.10 + 1.68 * 3.29 = 5.62
.............................................

While the sdf looks like this:
 (CELL
  (CELLTYPE "SC_INV_100_60")
  (INSTANCE MI1373/MI1370/X105613)
  (DELAY
   (ABSOLUTE
    (IOPATH A ZN (0.00::0.00))
   )
  )
 )

However, the report_disabled_edges shows 14386 lines, and one of them is:
MI1373/MI1370/X105613 A ZN constant A=1

And thus the 0.00 delay above makes sense, thanks, I believe I can move forward with this. I will have to look at
the documentation to understand the constant, loop, and positive_unate and negative_unate reports.  But this all happened just from read_verilog, current_design, link_design, so I didn't expect all that analysis already.  Thanks!

tspyrou, 2022-01-14:
@Mark-at-Apex did you mean to close this?

tspyrou, 2022-01-14:
@Mark-at-Apex sorry I see you meant to close it. Re-open if you have more questions.

Mark-at-Apex, 2022-01-14:
Yes, I think you explained it enough for me to move forward, as I said, I need to understand the loop and constant reports more but that is just documenation I need to read.  Sorry, I didn't realize this analysis happened as part of write_sdf

Mark-at-Apex, 2022-01-14:
Tom,
I am trying to actually identify the loops.  I do see that the timing arc with the 0.00 delay is the arc which has been disabled to break the loop.  But OpenSTA.pdf on pages 31-32 has the report_disable_edges command, and it then references a report_disable_timing command, which unfortunately when I try it, only gives me:

% report_disable_timing
invalid command name "report_disable_timing"

And if I look at the help output, I don't find a report_disable_timing command.

I have created some tcl procs, that allow me to probe nets and probe instances as follows:

Consider one loop arc from report_disabled_edges:
MI102/X108826 A1 ZN loop

It is consistent with the SDF 0.00 delay feature:

 (CELL
  (CELLTYPE "SC_NOR2B2_90_25")
  (INSTANCE MI102/X108826)
  (DELAY
   (ABSOLUTE
    (IOPATH A1 ZN (0.00::0.00))
    (IOPATH A2N ZN (16.93::16.93) (20.30::20.30))
   )
  )
 )

and I can see how it is connected with my probe inst command:
% probe inst MI102/X108953/ZN
cmd=<probeInst MI102/X108953>
SC_NOR2_90_30 MI102/X108953 (
	.A1	( X106072_ZN ),
	.ZN	( MI102/X108953_ZN ),
	.A2	( MI102/X108828_ZN ),
	.VS	( MI102/VS ),
	.GS	( GND ),
	);

and I can trace the nets, with my probe net (tcl proc) as follows:
% probe net MI102/X108953_ZN
found MI102/X108953_ZN
netName = MI102/X108953_ZN
cmd=<probeNet MI102/X108953_ZN>
wire MI102/X108953_ZN:
	> MI102/X108953/ZN	MI102/X108953_ZN
	> MI102/X108828/A2	MI102/X108953_ZN

But I have not been able to identify a "LOOP" with the A1->ZN arc of the X108826 instance in it.

So at this point I am wondering if there are some loop reporting capabilities I have not yet recognized?

Thanks for any help you can offer here.

Mark-at-Apex, 2022-01-14:
sorry, I cut and pasted the wrong outputs above, it should have been:

% probe inst MI102/X108826
cmd=<probeInst MI102/X108826>
SC_NOR2B2_90_25 MI102/X108826 (
	.A1	( X106072_ZN ),
	.ZN	( MI102/X108826_ZN ),
	.A2N	( MI102/X108828_ZN ),
	.VS	( MI102/VS ),
	.GS	( GND ),
	);

and
% probe net MI102/X108826_ZN
found MI102/X108826_ZN
netName = MI102/X108826_ZN
cmd=<probeNet MI102/X108826_ZN>
wire MI102/X108826_ZN:
	> MI102/X109441/A	MI102/X108826_ZN
	> MI102/X108956/A2	MI102/X108826_ZN
	> MI102/X108826/ZN	MI102/X108826_ZN

maliberty, 2022-01-15:
Its report_disabled_edges

Mark-at-Apex, 2022-01-17:
yah, that report only shows one arc in the loop, which matches the one in the SDF with 0.00 delay, but it doesn't help me find the rest of the arcs in the loop.  I am trying to identify combinational latches which I may want to replace with cells that have better sequential timing arcs.

maliberty, 2022-01-19:
I don't know of a tool for that but there are some developer debugs in sta that might help.  "sta::set_debug loop 2" will report on loop breaking in the timer (set before running sta).

Mark-at-Apex, 2022-01-20:
That seemed to be quite helpful, but I would alert you to the following issue I cannot explain:

If I start in this way:
% sta::set_debug loop 2
% read_liberty NXP7105_liberty.lib

 % source read_veri.tcl

and then do 
% source write_sdf.tcl > write_sdf.set_debug.log

Then I do get a big log:
      1 loop: Loop
      2 loop:  MI1373/MI1370/X30703A/Q -> MI1373/MI1370/X54375/A1
      3 loop:  MI1373/MI1370/X54375/A1 -> MI1373/MI1370/X54375/ZN
      4 loop:  MI1373/MI1370/X54375/ZN -> MI1373/MI1370/X30703A/CP
      5 loop:  MI1373/MI1370/X30703A/CP -> MI1373/MI1370/X30703A/Q
      6 loop: Loop
      7 loop:  MI1373/MI1370/X18339/ZN -> MI1373/MI1370/X18339/B1N
      8 loop:  MI1373/MI1370/X18339/B1N -> MI1373/MI1370/X18339/ZN
      9 loop: Loop
     10 loop:  MI1373/MI1370/X38281/ZN -> MI1373/MI1370/X71960/A1
     11 loop:  MI1373/MI1370/X71960/A1 -> MI1373/MI1370/X71960/ZN
     12 loop:  MI1373/MI1370/X71960/ZN -> MI1373/MI1370/X78723/A1
     13 loop:  MI1373/MI1370/X78723/A1 -> MI1373/MI1370/X78723/ZN
     14 loop:  MI1373/MI1370/X78723/ZN -> MI1373/MI1370/X38281/A1
     15 loop:  MI1373/MI1370/X38281/A1 -> MI1373/MI1370/X38281/ZN
     16 loop: Loop
     17 loop:  MI1373/MI1370/X32955/Z -> MI1373/MI1370/X32955/A2
     18 loop:  MI1373/MI1370/X32955/A2 -> MI1373/MI1370/X32955/Z
     19 loop: Loop
     20 loop:  MI1373/MI1370/X17482/ZN -> MI1373/MI1370/X17482/B1N
     21 loop:  MI1373/MI1370/X17482/B1N -> MI1373/MI1370/X17482/ZN
...
2933633 loop: Loop
2933634 loop:  MI1373/MI1370/X36996/ZN -> MI1373/MI1370/X68299/A2
2933635 loop:  MI1373/MI1370/X68299/A2 -> MI1373/MI1370/X68299/ZN
2933636 loop:  MI1373/MI1370/X68299/ZN -> MI1373/MI1370/X36996/B
2933637 loop:  MI1373/MI1370/X36996/B -> MI1373/MI1370/X36996/ZN
2933638 loop: Loop
2933639 loop:  MI1373/MI1370/X35924/Z -> MI1373/MI1370/X35924/A2
2933640 loop:  MI1373/MI1370/X35924/A2 -> MI1373/MI1370/X35924/Z
2933641 loop: Loop
2933642 loop:  MI1373/MI1370/X35924/Z -> MI1373/MI1370/X36115/A1
2933643 loop:  MI1373/MI1370/X36115/A1 -> MI1373/MI1370/X36115/Z
2933644 loop:  MI1373/MI1370/X36115/Z -> MI1373/MI1370/X35924/B
2933645 loop:  MI1373/MI1370/X35924/B -> MI1373/MI1370/X35924/Z

But if I compare the (IOPATH A1 ZN (0.00::0.00)) lines in the SDF with the loops above,
then I find that  12737 of the SDF 0.00 delay pins can be matched with pins found in the loop report above.

However, I also find that 1009 of the SDF 0.00 delay pins do not match any of the pins in the loop report above.matches 
look like this for my analysis:
###################################
shortInst = X108826)
(INSTANCE MI102/X108826) (IOPATH A1 ZN (0.00::0.00))
inpin = A1
inst2 = X108826.A1
loop:  MI1373/MI1370/X106072/ZN -> MI102/X108826/A1
loop:  MI102/X108826/A1 -> MI102/X108826/ZN
match = 1
to 
###################################
shortInst = X3635)
(INSTANCE MI95/X3380_X3635) (IOPATH B2 ZN (0.00::0.00)) (INSTANCE MI95/X3635) (IOPATH B2 ZN (0.00::0.00))
inpin = B2
inst2 = X3635.B2
loop:  MI95/X3380_X3653/Q -> MI95/X3380_X3635/B2
loop:  MI95/X3380_X3635/B2 -> MI95/X3380_X3635/ZN
loop:  MI95/X3653/Q -> MI95/X3635/B2
loop:  MI95/X3635/B2 -> MI95/X3635/ZN
match = 12736
inpin = B2
inst2 = X3635.B2
loop:  MI95/X3380_X3653/Q -> MI95/X3380_X3635/B2
loop:  MI95/X3380_X3635/B2 -> MI95/X3380_X3635/ZN
loop:  MI95/X3653/Q -> MI95/X3635/B2
loop:  MI95/X3635/B2 -> MI95/X3635/ZN
match = 12737

while the non-matching SDF 0.00 lines look like this:
###################################
shortInst = X108824)
(INSTANCE MI102/X108824) (IOPATH A1 ZN (0.00::0.00))
inpin = A1
inst2 = X108824.A1
nomatch = 1

to
shortInst = X3650)
(INSTANCE MI95/X3380_X3650) (IOPATH A ZN (0.00::0.00)) (INSTANCE MI95/X3650) (IOPATH A ZN (0.00::0.00))
inpin = A
inst2 = X3650.A
nomatch = 1008
inpin = A
inst2 = X3650.A
nomatch = 1009

SO to me it seems that about 1/12 of the Zero-delay SDF lines are flawed, in that there is no loop reported that they are
part of.  I have not determined if it is the loop report or the SDF which is the mistaken report, but I hope you can see they
disagree.   I am happy to share data if desired to debug this further.

Thanks again for the sta::set_debug loop 2 command!

Mark Vancura

maliberty, 2022-01-20:
You could have zero delay for another reason (eg constant propagation).    Unless you see something that you specifically are concerned about I don't feel the need to look further.

Mark-at-Apex, 2022-01-20:
Ok, earlier above I learned how to use the resport_disabled_edges command to identify the constant pins, and I added that to my analysis, and for my design, there are 12725 0.00 delays in SDF which can also be found in the loop report.  And there are 956 0.00 delays in the SDF which can also be found in the constand report, and there are no other 0.00 SDF delays, which don't fall into one of those two categories, so yes, I am satisfied.  Thanks again for the details needed to determine this all made sense.

closed 2022-01-20 by Mark-at-Apex