The-OpenROAD-Project/OpenROAD #1550: USEMINSPACING OBS OFF breaks sky130 hd chameleon_hier

maliberty, 2022-01-13:
re: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/commit/9da2d0659588f37dfea7771b7de6d623b486b050

@antonblanchard this has broken the ORFS test sky130 hd chameleon_hier.  Do you have time to look into it?  The router fails with 

[ERROR DRT-0073] No ap for ibex_core/HADDR[10].

I guess the "USEMINSPACING OBS OFF" makes the pin inaccessible.

maliberty, 2022-01-13:
@vvbandeira FYI

antonblanchard, 2022-01-13:
I can have a go. sky130hd chameleon_hier dies in detailed placement with my setup, so I need to sort that out first:
```
[ERROR DPL-0018] detailed placement failed on RAM.genblk1\[0\].RAM.
```

antonblanchard, 2022-01-14:
@maliberty the issue is we only have minimum clearance around the pins:
[image]

Creating a wide spacing clearance fixes it:

[image]

Which I guess is expected since the blockage around it is very wide. It does mean that pretty much every pin needs wide spacing clearance.

One option is to back this patch out of ORFS, and continue to work on the `write_abstract_lef` patch.

maliberty, 2022-01-14:
I think backing it out is best (would you make a PR to do so?).  I pointed out the same issue was likely with your other PR due to the min spacing assumption.

vijayank88, 2023-06-19:
With latest commit not facing any issue with chameleon_hier and flow completes successfully.

closed 2023-06-19 by vijayank88