The-OpenROAD-Project/OpenROAD #1555: Detailed Placement Failed

donn, 2022-01-15:
```
OpenROAD v2.0-2678-gf8f3d6bb3 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  ./tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: ./in.def
[INFO ODB-0128] Design: spm
[INFO ODB-0130]     Created 38 pins.
[INFO ODB-0131]     Created 1389 components and 7408 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 5352 connections.
[INFO ODB-0133]     Created 425 nets and 2056 connections.
[INFO ODB-0134] Finished DEF file: ./in.def
[ERROR DPL-0019] detailed placement failed on _527_
```

[reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7875887/reproducible.tar.gz)

This is likely something wrong with how we're handling diode strategies in OpenLane.

maliberty, 2022-01-16:
What are you trying to do here?  You seem to have placed antenna cells everywhere and are re-running placement with CELL_PAD='4' which means tons of spacing is required around each antenna cell.  This seems like a methodology issue rather than a placement issue.

maliberty, 2022-01-16:
The antennas cells are highlighted:
[image]

donn, 2022-01-16:
Ah, gotcha. I'm just figuring out why diode insertion strats 2, 5 were failing- didn't write those.

closed 2022-01-16 by donn