The-OpenROAD-Project/OpenROAD #1576: get_nets -of_objects returns a list of valid nets and a NULL

iruhm, 2022-01-23:
I had to catch the NULL in the list of nets to avoid a script error. 
Is this behaviour of get_nets intended or a bug?

This is the script part that I added to flow.tcl to get a robust clock net routing:

# Global routing
create_ndr -name mult -spacing { *2 } -width {met1:met4 *3}
set nets [get_nets -of_objects [get_fanout -from [get_full_name [sta::get_clocks *]]] -quiet]
foreach net $nets {
  if {![string equal $net NULL]} {
    assign_ndr -ndr mult -net [get_full_name $net]
  }
}

maliberty, 2022-01-23:
Please provide a test case

iruhm, 2022-01-24:
[Uploading run_all_aes_sky130hd_base_2022-01-23_21-53.tar.gz…]()

get_nets command is part of run_all_aes_sky130hd_base_2022-01-23_21-53/scripts/pre_global_route.tcl
without the comments the layout is generated with the custom clock tree routing rule

iruhm, 2022-01-24:
[run_all_aes_sky130hd_base_2022-01-23_21-53.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7922886/run_all_aes_sky130hd_base_2022-01-23_21-53.tar.gz)
get_nets command is part of run_all_aes_sky130hd_base_2022-01-23_21-53/scripts/pre_global_route.tcl
without the comments the layout is generated with the custom clock tree routing rule

closed 2022-01-24 by tspyrou