The-OpenROAD-Project/OpenROAD #1577: drt: routing finishes with 0 violations, but creates a short on met1

antonblanchard, 2022-01-23:
While chasing an LVS issue in a design, it looks like detailed routing created a short on met1 without reporting it as a violation:

```
    Completing 100% with 0 violations.
    elapsed time = 00:00:00, memory = 25517.50 (MB).
[INFO DRT-0199]   Number of violations = 0.
```

But there is a short between nets `net718` and `_017753_` on `met1`. This is a screenshot of just met1:

[image]

Coordinates: `1448.994`, `167.121`

antonblanchard, 2022-01-23:
Testcase: [drt-drc-bug.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7920019/drt-drc-bug.tar.gz)

osamahammad21, 2022-01-25:
Working on it

antonblanchard, 2022-02-02:
Thanks @osamahammad21! Any clues here?

maliberty, 2022-02-02:
There is a lot of discussion going on.  The basic issue is that the short falls right on the boundary between two workers and there is some conflict over ownership of the shapes and markers.  It is getting attention.

antonblanchard, 2022-02-03:
Thanks @maliberty, that gives me a possible workaround (running with `ROUTING_CORES=1`)

maliberty, 2022-02-03:
Its not such the number of threads as it is the boundary condition.  I don't think it will help but you can try.

QuantamHD, 2022-02-03:
Do we need to run convolutions over the routing grid,  or is there a more efficient solution here?

maliberty, 2022-02-03:
There's just a bug that needs fixing

osamahammad21, 2022-02-03:
Changing the number of threads won't change a thing. I have a local change that solves this violation. But the real problem I see here is that 0 violations doesn't actually mean 0 violations. This is because of the nature of GC being run on separate workers not on the whole design, so it's affected by what the drWorker initializes it with.

osamahammad21, 2022-02-09:
@antonblanchard The pr is merged into master, so the issue is fixed now.

closed 2022-02-09 by osamahammad21