The-OpenROAD-Project/OpenROAD #1585: Error building odb

GuzTech, 2022-01-25:
After checking out `master` and running `./build_openroad.sh --local`, I get several errors when building odb:

```
/home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp: In member function ‘void LefDefParser::defrData::defInfo(int, const char*)’:
/home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1028:19: error: argument 1 is null but the corresponding size argument 2 value is 64 [-Werror=nonnull]
 1028 |             printf("WARNING(DEFPARS-8500): Unable to open the file defRWarning.log in %s.\n",
      |             ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1029 |             getcwd(NULL, 64));
      |             ~~~~~~~~~~~~~~~~~
In file included from /home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:49:
/usr/include/unistd.h:520:14: note: in a call to function ‘char* getcwd(char*, size_t)’ declared with attribute ‘access (write_only, 1, 2)’
  520 | extern char *getcwd (char *__buf, size_t __size) __THROW __wur
      |              ^~~~~~
/home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1039:19: error: argument 1 is null but the corresponding size argument 2 value is 64 [-Werror=nonnull]
 1039 |             printf("WARNING (DEFPARS-8500): Unable to open the file defRWarning.log in %s.\n",
      |             ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1040 |             getcwd(NULL, 64));
      |             ~~~~~~~~~~~~~~~~~
In file included from /home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:49:
/usr/include/unistd.h:520:14: note: in a call to function ‘char* getcwd(char*, size_t)’ declared with attribute ‘access (write_only, 1, 2)’
  520 | extern char *getcwd (char *__buf, size_t __size) __THROW __wur
      |              ^~~~~~
/home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp: In member function ‘void LefDefParser::defrData::defWarning(int, const char*)’:
/home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1083:19: error: argument 1 is null but the corresponding size argument 2 value is 64 [-Werror=nonnull]
 1083 |             printf("WARNING (DEFPARS-7500): Unable to open the file defRWarning.log in %s.\n",
      |             ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1084 |             getcwd(NULL, 64));
      |             ~~~~~~~~~~~~~~~~~
In file included from /home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:49:
/usr/include/unistd.h:520:14: note: in a call to function ‘char* getcwd(char*, size_t)’ declared with attribute ‘access (write_only, 1, 2)’
  520 | extern char *getcwd (char *__buf, size_t __size) __THROW __wur
      |              ^~~~~~
/home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1094:19: error: argument 1 is null but the corresponding size argument 2 value is 64 [-Werror=nonnull]
 1094 |             printf("WARNING (DEFAPRS-7501): Unable to open the file defRWarning.log in %s.\n",
      |             ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1095 |             getcwd(NULL, 64));
      |             ~~~~~~~~~~~~~~~~~
In file included from /home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:49:
/usr/include/unistd.h:520:14: note: in a call to function ‘char* getcwd(char*, size_t)’ declared with attribute ‘access (write_only, 1, 2)’
  520 | extern char *getcwd (char *__buf, size_t __size) __THROW __wur
      |              ^~~~~~
/home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp: In member function ‘void LefDefParser::defrData::pathIsDone(int, int, int, int*)’:
/home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1347:13: error: ‘void free(void*)’ called on pointer ‘<unknown>’ with nonzero offset 632 [-Werror=free-nonheap-object]
 1347 |         free((char*) &PathObj);
      |         ~~~~^~~~~~~~~~~~~~~~~~
In function ‘void LefDefParser::lefWarning(int, const char*)’,
    inlined from ‘void LefDefParser::lefWarning(int, const char*)’ at /home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lef/lef/lef_keywords.cpp:1158:1:
/home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lef/lef/lef_keywords.cpp:1208:23: error: argument 1 is null but the corresponding size argument 2 value is 64 [-Werror=nonnull]
 1208 |                 printf("WARNING (LEFPARS-2500): Unable to open the file lefRWarning.log in %s.\n",
      |                 ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1209 |                        getcwd(NULL, 64));
      |                        ~~~~~~~~~~~~~~~~~
In file included from /home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lef/lef/lef_keywords.cpp:45:
/usr/include/unistd.h: In function ‘void LefDefParser::lefWarning(int, const char*)’:
/usr/include/unistd.h:520:14: note: in a call to function ‘char* getcwd(char*, size_t)’ declared with attribute ‘access (write_only, 1, 2)’
  520 | extern char *getcwd (char *__buf, size_t __size) __THROW __wur
      |              ^~~~~~
In function ‘void LefDefParser::lefWarning(int, const char*)’,
    inlined from ‘void LefDefParser::lefWarning(int, const char*)’ at /home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lef/lef/lef_keywords.cpp:1158:1:
/home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lef/lef/lef_keywords.cpp:1219:23: error: argument 1 is null but the corresponding size argument 2 value is 64 [-Werror=nonnull]
 1219 |                 printf("WARNING (LEFPARS-2501): Unable to open the file lefRWarning.log in %s.\n",
      |                 ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1220 |                        getcwd(NULL, 64));
      |                        ~~~~~~~~~~~~~~~~~
In file included from /home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lef/lef/lef_keywords.cpp:45:
/usr/include/unistd.h: In function ‘void LefDefParser::lefWarning(int, const char*)’:
/usr/include/unistd.h:520:14: note: in a call to function ‘char* getcwd(char*, size_t)’ declared with attribute ‘access (write_only, 1, 2)’
  520 | extern char *getcwd (char *__buf, size_t __size) __THROW __wur
      |              ^~~~~~
In function ‘void LefDefParser::lefInfo(int, const char*)’,
    inlined from ‘void LefDefParser::lefInfo(int, const char*)’ at /home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lef/lef/lef_keywords.cpp:1087:1:
/home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lef/lef/lef_keywords.cpp:1132:23: error: argument 1 is null but the corresponding size argument 2 value is 64 [-Werror=nonnull]
 1132 |                 printf("WARNING (LEFPARS-3500): Unable to open the file lefRWarning.log in %s.\n",
      |                 ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1133 |                        getcwd(NULL, 64));
      |                        ~~~~~~~~~~~~~~~~~
In file included from /home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lef/lef/lef_keywords.cpp:45:
/usr/include/unistd.h: In function ‘void LefDefParser::lefInfo(int, const char*)’:
/usr/include/unistd.h:520:14: note: in a call to function ‘char* getcwd(char*, size_t)’ declared with attribute ‘access (write_only, 1, 2)’
  520 | extern char *getcwd (char *__buf, size_t __size) __THROW __wur
      |              ^~~~~~
In function ‘void LefDefParser::lefInfo(int, const char*)’,
    inlined from ‘void LefDefParser::lefInfo(int, const char*)’ at /home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lef/lef/lef_keywords.cpp:1087:1:
/home/oguz286/Software/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lef/lef/lef_keywords.cpp:1143:23: error: argument 1 is null but the corresponding size argument 2 value is 64 [-Werror=nonnull]
 1143 |                 printf("WARNING (LEFPARS-3500): Unable to open the file lefRWarning.log in %s.\n",
      |                 ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1144 |                        getcwd(NULL, 64));

```

I am using GCC version 11.1.0 (and clang 13.0.0 if that is used).

dralabeing, 2022-01-26:
@vijayank88 can you test this?

maliberty, 2022-01-26:
Can you try building with ALLOW_WARNINGS=ON

GuzTech, 2022-01-26:
@maliberty I assume like this `ALLOW_WARNINGS=ON ./build_openroad.sh --local`?

maliberty, 2022-01-26:
You would need ./build_openroad.sh --local --openroad-args ALLOW_WARNINGS=ON

GuzTech, 2022-01-26:
That gives me the following:

```
❯ ./build_openroad.sh --local --openroad-args ALLOW_WARNINGS=ON
[INFO FLW-0002] Updating git submodules.
[INFO FLW-0001] Using local build method. This will create binaries at 'tools/install unless overwritten.
[INFO FLW-0017] Compiling Yosys.
make: Entering directory '/home/oguz286/Software/OpenROAD-flow-scripts/tools/yosys'
mkdir -p /home/oguz286/Software/OpenROAD-flow-scripts/tools/install/yosys/bin
cp yosys yosys-config yosys-abc yosys-filterlib yosys-smtbmc /home/oguz286/Software/OpenROAD-flow-scripts/tools/install/yosys/bin
strip -S /home/oguz286/Software/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
strip /home/oguz286/Software/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc
strip /home/oguz286/Software/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-filterlib
mkdir -p /home/oguz286/Software/OpenROAD-flow-scripts/tools/install/yosys/share/yosys
cp -r share/. /home/oguz286/Software/OpenROAD-flow-scripts/tools/install/yosys/share/yosys/.
make: Leaving directory '/home/oguz286/Software/OpenROAD-flow-scripts/tools/yosys'
[INFO FLW-0018] Compiling OpenROAD.
CMake Error: The source directory "/home/oguz286/Software/OpenROAD-flow-scripts/ALLOW_WARNINGS=ON" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
```

maliberty, 2022-01-26:
Perhaps
 ./build_openroad.sh --local --openroad-args "-D ALLOW_WARNINGS=ON"

GuzTech, 2022-01-26:
That did the trick! I can now try the GUI mode that you demonstrated :)

maliberty, 2022-01-26:
Great.  The plan is to swap so that is the default outside of CI but at least you have a solution for now.

closed 2022-01-26 by maliberty