The-OpenROAD-Project/OpenROAD #1597: drt creates met1 wire stubs on sky130hd designs

antonblanchard, 2022-01-29:
I notice quite a lot of met1 wire stubs on sky130hd designs. It appears when we are just connecting through met1:

[image]

So I thought it might be a design rule (like a minimum area rule). But it also appears at the end of longer wires, but not always:

[image]

Is this expected? This makes me wonder if it isn't a design rule, but something in drt. I can package something up, but it does show up on the sky130hd aes test in ORFS.

maliberty, 2022-01-29:
is that a separate shape or part of the via?

antonblanchard, 2022-01-30:
I think it's a separate shape (not sure how to be sure). I see the L1M1 and M1M2 vias and the associated met1 rectangles. The stub is just a rectangle of met1 with a net name.
[image]

LEF and DEF files:
[merged_spacing.lef.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7965157/merged_spacing.lef.gz)
[6_final.def.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7965158/6_final.def.gz)

I'm looking at `423.075, 425.780`

maliberty, 2022-01-30:
That case is a stacked via and it doesn't meet min area without a patch metal on met1 so I don't see a problem at that coordinate.

antonblanchard, 2022-01-30:
@maliberty Thanks. One last question at `423.755, 469.300`, net `us30.a[6]`. Here met1 extends in 3 directions from the stacked via. Shouldn't need a stub there should we?

[image]

maliberty, 2022-01-31:
That one is more interesting.  There is a stack via there as well as met1 segments.  My guess is that it routed the stacked via first and patched it.  Then it added the other segments later and didn't remove the patch.  Could you package a test that produces this DEF?

Its not strictly a problem to have the extra patches though in some case it might lead to suboptimal routing if the patch blocks another wire.

antonblanchard, 2022-01-31:
Here's a packaged up test. I switched to the sky130hd jpeg test because it was smaller. `656.51500, 783.80000` net `_029496_`.

[stub-issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7974230/stub-issue.tar.gz)

maliberty, 2022-01-31:
Thanks, smaller is always better.

Stephanommg, 2022-02-07:
It is not clear to me whether this is a real problem and is expected to be solved or if @antonblanchard is simple curious to know why this happens.

This behavior is expected, due to what @maliberty said: a patch is place to solve a min area viol, then, another wire is connected to the shape, so the patch is not required anymore, but the router doesn't remove it.

antonblanchard, 2022-02-07:
@Stephanommg I've been having routing issues with some of my designs, and looking for all areas that could be improved. Seems like there might be a small degradation in met1 routing capacity due to this issue. Not a high priority, but I would think it is worth fixing.

Stephanommg, 2022-02-07:
@antonblanchard ok, I am currently investigating the best way to proceed with a solution.

maliberty, 2022-02-07:
@tspyrou flagged this as a higher priority but it sounds like it isn't.  Your other two issues seem more important.

rovinski, 2022-02-08:
Seems like there could be a patch cleanup step after routing which removes patches if the minArea constraints are otherwise satisfied. If it comes after the end of DR then it saves wirelength but doesn't improve routing. If it comes between DR iterations, it would free up routing but possibly cause instability; I'm not sure.

Stephanommg, 2022-02-08:
@rovinski  My first idea was to clean the patches after each net is routed, but this way we would have 2 code logic, one for creating and other for removing, while we can just create the necessary patches after the net is routed. So I have chosen this approach.

rovinski, 2022-02-08:
@Stephanommg I recall Banqi saying that they tried adding patches after routing, but there are circumstances where the route doesn't allow any patches without creating a violation. So I think removing the first approach might work better.

rovinski, 2022-02-09:
Consider the following:
[image]
DR doesn't do patch metal until after all routes.
It connects pin 1 with M2 and a via and pin 2 in M1. Pin 2 is fine for metal area, but pin 1 is not. Pin 1 cannot have a patch added either because it is blocked by the route to pin 2.

I don't know how often this comes up, but that's the reason I remember for patch metal being added first.

Stephanommg, 2022-02-09:
> Consider the following: [image] DR doesn't do patch metal until after all routes. It connects pin 1 with M2 and a via and pin 2 in M1. Pin 2 is fine for metal area, but pin 1 is not. Pin 1 cannot [...]

AFAIK, the pin shape is guaranteed to have the minimum area met.

rovinski, 2022-02-09:
Well if you run into DRC problems you might want to look at this / related problems.

Stephanommg, 2022-02-15:
@antonblanchard the solution is on master branch. In the end I tested with other designs and forgot to test with this one, so I await for your confirmation that it really solved the problem for this design. Also, the feature is by default turned off since it increased runtime. You need to activate it using the -clean_patches flag in the tcl with the router parameters.