The-OpenROAD-Project/OpenROAD #1603: `bench_read_spef` crashes

Syndace, 2022-02-01:
I am following [the documentation](https://openroad.readthedocs.io/en/latest/main/src/rcx/doc/calibration.html) regarding the calibration of OpenRCX. I used Cadence Innovus to extract the "golden SPEF". The following step, `generate_rules.tcl`, crashes at `bench_read_spef $golden_spef`, with following stack trace:

```
Stack trace:
 0# 0x0000000000FD1FE9 in openroad
 1# 0x00007F0EE922E400 in /lib64/libc.so.6
 2# odb::dbChip::getBlock() in openroad
 3# rcx::extMain::setBlockFromChip() in openroad
 4# rcx::Ext::read_spef(rcx::Ext::ReadSpefOpts&) in openroad
 5# 0x00000000017009B9 in openroad
 6# 0x00007F0EEC1C1EB2 in /lib64/libtcl8.5.so
 7# 0x00007F0EEC20636C in /lib64/libtcl8.5.so
 8# TclObjInterpProcCore in /lib64/libtcl8.5.so
 9# 0x00007F0EEC1C1EB2 in /lib64/libtcl8.5.so
10# 0x00007F0EEC20636C in /lib64/libtcl8.5.so
11# 0x00007F0EEC20E647 in /lib64/libtcl8.5.so
12# TclEvalObjEx in /lib64/libtcl8.5.so
13# 0x00007F0EEC24927F in /lib64/libtcl8.5.so
14# 0x00007F0EEC1C1EB2 in /lib64/libtcl8.5.so
15# 0x00007F0EEC20636C in /lib64/libtcl8.5.so
16# 0x00007F0EEC20E647 in /lib64/libtcl8.5.so
17# TclEvalObjEx in /lib64/libtcl8.5.so
18# 0x00007F0EEC1C91D0 in /lib64/libtcl8.5.so
19# 0x00007F0EEC1C1EB2 in /lib64/libtcl8.5.so
20# 0x00007F0EEC20636C in /lib64/libtcl8.5.so
21# 0x00007F0EEC20E647 in /lib64/libtcl8.5.so
22# TclEvalObjEx in /lib64/libtcl8.5.so
23# 0x00007F0EEC1CCF00 in /lib64/libtcl8.5.so
24# 0x00007F0EEC1C1EB2 in /lib64/libtcl8.5.so
25# 0x00007F0EEC20636C in /lib64/libtcl8.5.so
26# TclObjInterpProcCore in /lib64/libtcl8.5.so
27# 0x00007F0EEC1C1EB2 in /lib64/libtcl8.5.so
28# 0x00007F0EEC20636C in /lib64/libtcl8.5.so
29# TclObjInterpProcCore in /lib64/libtcl8.5.so
30# 0x00007F0EEC1C1EB2 in /lib64/libtcl8.5.so
31# 0x00007F0EEC1C2F1E in /lib64/libtcl8.5.so
32# Tcl_EvalEx in /lib64/libtcl8.5.so
33# Tcl_Eval in /lib64/libtcl8.5.so
34# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
35# ord::tclAppInit(Tcl_Interp*) in openroad
36# Tcl_Main in /lib64/libtcl8.5.so
37# main in openroad
38# __libc_start_main in /lib64/libc.so.6
39# 0x0000000000FCE8B7 in openroad
```

I've tried passing the golden SPEF using an absolute path too.

In fact, it just crashed on me when I just had `bench_read_spec` in the prompt and pressed tab again LOL:
```
$ openroad 
OpenROAD 3eafa6da2942ec22d27e2208c99c7a449af34fbf 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
readwarning: `//.tclsh-history' is not writable.
openroad> bench_
bench_read_spef  bench_verilog    bench_wires      
openroad> bench_read_spef Stack trace:
 0# 0x0000000000FD1FE9 in openroad
 1# 0x00007FE95A18B400 in /lib64/libc.so.6
 2# odb::dbChip::getBlock() in openroad
 3# rcx::extMain::setBlockFromChip() in openroad
 4# rcx::Ext::read_spef(rcx::Ext::ReadSpefOpts&) in openroad
 5# 0x00000000017009B9 in openroad
 6# 0x00007FE95D11EEB2 in /lib64/libtcl8.5.so
 7# 0x00007FE95D16336C in /lib64/libtcl8.5.so
 8# TclObjInterpProcCore in /lib64/libtcl8.5.so
 9# 0x00007FE95D11EEB2 in /lib64/libtcl8.5.so
10# 0x00007FE95D11FF1E in /lib64/libtcl8.5.so
11# Tcl_EvalEx in /lib64/libtcl8.5.so
12# TclEvalObjEx in /lib64/libtcl8.5.so
13# 0x00007FE95D1269FB in /lib64/libtcl8.5.so
14# 0x00007FE95D11EEB2 in /lib64/libtcl8.5.so
15# 0x00007FE95D16336C in /lib64/libtcl8.5.so
16# TclObjInterpProcCore in /lib64/libtcl8.5.so
17# 0x00007FE95D11EEB2 in /lib64/libtcl8.5.so
18# 0x00007FE95D16336C in /lib64/libtcl8.5.so
19# 0x00007FE95D16B647 in /lib64/libtcl8.5.so
20# TclEvalObjEx in /lib64/libtcl8.5.so
21# 0x00007FE95D129F00 in /lib64/libtcl8.5.so
22# 0x00007FE95D11EEB2 in /lib64/libtcl8.5.so
23# 0x00007FE95D16336C in /lib64/libtcl8.5.so
24# TclObjInterpProcCore in /lib64/libtcl8.5.so
25# 0x00007FE95D11EEB2 in /lib64/libtcl8.5.so
26# 0x00007FE95D16336C in /lib64/libtcl8.5.so
27# TclObjInterpProcCore in /lib64/libtcl8.5.so
28# 0x00007FE95D11EEB2 in /lib64/libtcl8.5.so
29# 0x00007FE95D11FF1E in /lib64/libtcl8.5.so
30# Tcl_EvalEx in /lib64/libtcl8.5.so
31# Tcl_Eval in /lib64/libtcl8.5.so
32# Tcl_VarEvalVA in /lib64/libtcl8.5.so
33# Tcl_VarEval in /lib64/libtcl8.5.so
34# 0x00007FE95CCCB90D in /lib64/libtclreadline-2.1.0.so
35# 0x00007FE95C6826F9 in /lib64/libreadline.so.6
36# rl_complete_internal in /lib64/libreadline.so.6
37# _rl_dispatch_subseq in /lib64/libreadline.so.6
38# readline_internal_char in /lib64/libreadline.so.6
39# rl_callback_read_char in /lib64/libreadline.so.6
40# 0x00007FE95D1CDB97 in /lib64/libtcl8.5.so
41# Tcl_ServiceEvent in /lib64/libtcl8.5.so
42# Tcl_DoOneEvent in /lib64/libtcl8.5.so
43# 0x00007FE95CCCB22A in /lib64/libtclreadline-2.1.0.so
44# 0x00007FE95D11EEB2 in /lib64/libtcl8.5.so
45# 0x00007FE95D16336C in /lib64/libtcl8.5.so
46# 0x00007FE95D16B647 in /lib64/libtcl8.5.so
47# TclEvalObjEx in /lib64/libtcl8.5.so
48# 0x00007FE95D1261D0 in /lib64/libtcl8.5.so
49# 0x00007FE95D11EEB2 in /lib64/libtcl8.5.so
50# 0x00007FE95D16336C in /lib64/libtcl8.5.so
51# 0x00007FE95D16B647 in /lib64/libtcl8.5.so
52# TclEvalObjEx in /lib64/libtcl8.5.so
53# 0x00007FE95D1A627F in /lib64/libtcl8.5.so
54# 0x00007FE95D11EEB2 in /lib64/libtcl8.5.so
55# 0x00007FE95D16336C in /lib64/libtcl8.5.so
56# TclObjInterpProcCore in /lib64/libtcl8.5.so
57# 0x00007FE95D11EEB2 in /lib64/libtcl8.5.so
58# 0x00007FE95D11FF1E in /lib64/libtcl8.5.so
59# Tcl_EvalEx in /lib64/libtcl8.5.so
60# Tcl_Eval in /lib64/libtcl8.5.so
61# ord::tclAppInit(Tcl_Interp*) in openroad
62# Tcl_Main in /lib64/libtcl8.5.so
63# main in openroad
64# __libc_start_main in /lib64/libc.so.6
65# 0x0000000000FCE8B7 in openroad
```

maliberty, 2022-02-01:
I don't see that when I try it:

openroad> bench_read_spef 
[ERROR STA-0407] bench_read_spef requires one positional argument.
STA-0407

it makes suspicious of your build.  In your other crash did you read_def first?  See https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/rcx/calibration/script/generate_rules.tcl

Syndace, 2022-02-01:
What you did (typing `bench_read_spef` and then pressing enter) doesn't cause a crash for me either. Typing `bench_read_spef` and then pressing tab twice for tab-completion crashes for me. Or using `bench_read_spef` to read my golden SPEF.

I've also just reproduced it, on a system that had never run OpenLane before:
```
$ cd /tmp/
$ git clone git@github.com:The-OpenROAD-Project/OpenLane.git
$ cd OpenLane/
$ make mount
OpenLane Container (aac68a0):/openlane$ openroad 
OpenROAD 3eafa6da2942ec22d27e2208c99c7a449af34fbf 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
warning: `//.tclsh-history' is not writable.
openroad> bench_read_spef <tab><tab> Stack trace:
 0# 0x0000000000FD1FE9 in openroad
 1# 0x00007F223EDBF400 in /lib64/libc.so.6
 2# odb::dbChip::getBlock() in openroad
 3# rcx::extMain::setBlockFromChip() in openroad
 4# rcx::Ext::read_spef(rcx::Ext::ReadSpefOpts&) in openroad
 5# 0x00000000017009B9 in openroad
 6# 0x00007F2241D52EB2 in /lib64/libtcl8.5.so
 7# 0x00007F2241D9736C in /lib64/libtcl8.5.so
 8# TclObjInterpProcCore in /lib64/libtcl8.5.so
 9# 0x00007F2241D52EB2 in /lib64/libtcl8.5.so
10# 0x00007F2241D53F1E in /lib64/libtcl8.5.so
11# Tcl_EvalEx in /lib64/libtcl8.5.so
12# TclEvalObjEx in /lib64/libtcl8.5.so
13# 0x00007F2241D5A9FB in /lib64/libtcl8.5.so
14# 0x00007F2241D52EB2 in /lib64/libtcl8.5.so
15# 0x00007F2241D9736C in /lib64/libtcl8.5.so
16# TclObjInterpProcCore in /lib64/libtcl8.5.so
17# 0x00007F2241D52EB2 in /lib64/libtcl8.5.so
18# 0x00007F2241D9736C in /lib64/libtcl8.5.so
19# 0x00007F2241D9F647 in /lib64/libtcl8.5.so
20# TclEvalObjEx in /lib64/libtcl8.5.so
21# 0x00007F2241D5DF00 in /lib64/libtcl8.5.so
22# 0x00007F2241D52EB2 in /lib64/libtcl8.5.so
23# 0x00007F2241D9736C in /lib64/libtcl8.5.so
24# TclObjInterpProcCore in /lib64/libtcl8.5.so
25# 0x00007F2241D52EB2 in /lib64/libtcl8.5.so
26# 0x00007F2241D9736C in /lib64/libtcl8.5.so
27# TclObjInterpProcCore in /lib64/libtcl8.5.so
28# 0x00007F2241D52EB2 in /lib64/libtcl8.5.so
29# 0x00007F2241D53F1E in /lib64/libtcl8.5.so
30# Tcl_EvalEx in /lib64/libtcl8.5.so
31# Tcl_Eval in /lib64/libtcl8.5.so
32# Tcl_VarEvalVA in /lib64/libtcl8.5.so
33# Tcl_VarEval in /lib64/libtcl8.5.so
34# 0x00007F22418FF90D in /lib64/libtclreadline-2.1.0.so
35# 0x00007F22412B66F9 in /lib64/libreadline.so.6
36# rl_complete_internal in /lib64/libreadline.so.6
37# _rl_dispatch_subseq in /lib64/libreadline.so.6
38# readline_internal_char in /lib64/libreadline.so.6
39# rl_callback_read_char in /lib64/libreadline.so.6
40# 0x00007F2241E01B97 in /lib64/libtcl8.5.so
41# Tcl_ServiceEvent in /lib64/libtcl8.5.so
42# Tcl_DoOneEvent in /lib64/libtcl8.5.so
43# 0x00007F22418FF22A in /lib64/libtclreadline-2.1.0.so
44# 0x00007F2241D52EB2 in /lib64/libtcl8.5.so
45# 0x00007F2241D9736C in /lib64/libtcl8.5.so
46# 0x00007F2241D9F647 in /lib64/libtcl8.5.so
47# TclEvalObjEx in /lib64/libtcl8.5.so
48# 0x00007F2241D5A1D0 in /lib64/libtcl8.5.so
49# 0x00007F2241D52EB2 in /lib64/libtcl8.5.so
50# 0x00007F2241D9736C in /lib64/libtcl8.5.so
51# 0x00007F2241D9F647 in /lib64/libtcl8.5.so
52# TclEvalObjEx in /lib64/libtcl8.5.so
53# 0x00007F2241DDA27F in /lib64/libtcl8.5.so
54# 0x00007F2241D52EB2 in /lib64/libtcl8.5.so
55# 0x00007F2241D9736C in /lib64/libtcl8.5.so
56# TclObjInterpProcCore in /lib64/libtcl8.5.so
57# 0x00007F2241D52EB2 in /lib64/libtcl8.5.so
58# 0x00007F2241D53F1E in /lib64/libtcl8.5.so
59# Tcl_EvalEx in /lib64/libtcl8.5.so
60# Tcl_Eval in /lib64/libtcl8.5.so
61# ord::tclAppInit(Tcl_Interp*) in openroad
62# Tcl_Main in /lib64/libtcl8.5.so
63# main in openroad
64# __libc_start_main in /lib64/libc.so.6
65# 0x0000000000FCE8B7 in openroad
```
(this time I've added `<tab>` where I pressed tab for auto-completion)

Also yes, I've tried running the full `generate_rules.tcl` script before.

maliberty, 2022-02-01:
It is a readline issue and I reproduced it and will fix that.  However I suspect the real problem is you haven't loaded a design.

maliberty, 2022-02-01:
Tab completion fixed with https://github.com/The-OpenROAD-Project/OpenROAD/pull/1605

maliberty, 2022-02-01:
The underlying problem is that tcl readline tries the command with '---' which it accepted as the filename.  Since no design is loaded you get the crash.  It now gives an error message instead.

Syndace, 2022-02-01:
Hmm. Do you mean "design loaded" as in "prep -design foo"? That sounds kind of wrong, the OpenRCX calibration is a one-time PDK-level operation that should not require any design to be loaded, if I understand it correctly. Anyway, thanks for being so quick to prepare a fix, I'll try it tomorrow.

maliberty, 2022-02-01:
If you look at the link I provided above you'll see it is reading in the patterns.def.  You have to have something to annotate spef against - it isn't a netlist by itself.

Syndace, 2022-02-01:
So you're not talking about `prep -design foo`, but `load_lef` and friends. Got it. As I said I ran the whole script before, including those commands. I just tried again with a build containing your fix and even though I run the whole script - including `read_lef $TECH_LEF` and `read_def EXT/patterns.def`, I still get the `[ERROR RCX-0497] No design is loaded.` output. Here's the full log:
```
root@cea0f0aded01:~/OpenROAD/src/rcx/calibration/work# ./openroad ../script/generate_rules.tcl 
OpenROAD v2.0-2865-ge714d4a41 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ../data/xxxxx_tech.lef
[INFO ODB-0223]     Created xxx technology layers
[INFO ODB-0224]     Created xxx technology vias
[INFO ODB-0226] Finished LEF file:  ../data/xxxxx_tech.lef
[ERROR RCX-0497] No design is loaded.
Error: generate_rules.tcl, 24 RCX-0497
```

Syndace, 2022-02-01:
Also just in case, here's me entering the commands manually one after another:
```
root@cea0f0aded01:~/OpenROAD/src/rcx/calibration/work# ./openroad
OpenROAD v2.0-2865-ge714d4a41 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad> read_lef ../data/xxxxx_tech.lef 
[INFO ODB-0222] Reading LEF file: ../data/xxxxx_tech.lef
[INFO ODB-0223]     Created xxx technology layers
[INFO ODB-0224]     Created xxx technology vias
[INFO ODB-0226] Finished LEF file:  ../data/xxxxx_tech.lef
openroad> read_def EXT/patterns.def 
openroad> bench_read_spef ../data/golden.spef 
[ERROR RCX-0497] No design is loaded.
RCX-0497
while evaluating bench_read_spef ../data/golden.spef 
openroad>
```

maliberty, 2022-02-02:
I don't see that:

openroad> bench_read_spef foo.spef
[INFO RCX-0001] Reading SPEF file: foo.spef
[INFO RCX-0286] Number of corners in SPEF file = 0.

Syndace, 2022-02-02:
Thanks, that was the hint I needed. If it's neither the command, nor the DEF file, it must be the LEF. It works when I load the phys LEF in addition to the tech LEF. Seems like you expect something to be defined in the tech LEF which isn't there in my case.

Anyway, it works, so I'm happy and I guess the issue can be closed :)

Slightly offtopic: is there any reference for what is considered "good" regarding the `diff_spef` output?

closed 2022-02-02 by Syndace