The-OpenROAD-Project/OpenROAD #1618: installation problem

ashkanr65, 2022-02-04:
Hi,
I am trying to install OpenRoad. However after the command "cmake .." I get this error.
Does anybody have any idea how can I solve this issue?
[image]

maliberty, 2022-02-04:
Have you installed the dependencies with ./etc/DependencyInstaller.sh per https://openroad.readthedocs.io/en/latest/main/README.html

It looks like you are missing spdlog

ashkanr65, 2022-02-04:
> Have you installed the dependencies with ./etc/DependencyInstaller.sh per https://openroad.readthedocs.io/en/latest/main/README.html
> 
> It looks like you are missing spdlog

I installed the dependencies and but I get the same error. I have even installed spdlog separately and I received the same error.

vijayank88, 2022-02-12:
@ashkanr65 
Have you installed both -run and -dev?
      ./etc/DependencyInstaller.sh -run      # installs dependencies to run a pre-compiled binary
       ./etc/DependencyInstaller.sh -dev   # installs dependencies to compile the openroad binary

Try to clone again in new directory and install it

maliberty, 2022-10-05:
Stale

closed 2022-10-05 by maliberty