The-OpenROAD-Project/OpenROAD #1619: Design failed with DRC violations

vijayank88, 2022-02-04:
During detailed routing flow failed with (Metal spacing and short violations) DRC errors.
sky130_fd_sc_hd__o221a_1 cell creating issues.
[run1_droute_packaged.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8005027/run1_droute_packaged.zip)

Attached tar file for reference.

dralabeing, 2022-02-09:
@maliberty This is an MPW design. Should this be re-assigned to OpenLane?

maliberty, 2022-02-09:
If it packaged as a standalone testcase (independent of OL) then it is ok for it to be reported here.

vijayank88, 2022-02-10:
@maliberty @dralabeing 
The above design ran successfully with MPW-4 shuttle without any DRC errors. Used PL_TARGET_DENSITY =0.6.

As a try reduced PL_TARGET_DENSITY=0.55 run successfully without DRC errors.

maliberty, 2022-02-16:
@Stephanommg please take a look at why this isn't able to route cleanly.  It doesn't look especially dense or complex.

Stephanommg, 2022-02-17:
@maliberty  The design seems a heavy one. Very slow since from the beginning and present persisting violations until the end. I ran it with my clipsize increase feature and it worked well, leaving no violation at the end. I still didnt look it in the GUI to see if the problem is really congestion.

Stephanommg, 2022-02-17:
@maliberty  By looking with the GUI the problem really seems congestion
[image]
I think the clipsize increase is the right solution here. It is working well on general, it is just ispd19_test10 that is having some timeout problems, so I am currently searching for the right iteration window to activate the feature.

maliberty, 2022-02-17:
@Stephanommg ok this can wait on your clipsize work.  There is a workaround with lower density.

Stephanommg, 2022-04-12:
Can this issue be closed?

closed 2022-04-27 by Stephanommg