The-OpenROAD-Project/OpenROAD #1620: global routing failing with routing congestion high

vijayank88, 2022-02-04:
Used relative die size calculated by yosys. Used full core utilization still flow failing with routing congestion too high error.

Attached reproducible tar file for reference.
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8005086/openroad_issue_reproducible.zip)

eder-matheus, 2022-03-02:
@vijayank88 I was able to run the global route in this design by reducing the global adjustment (`GLB_RT_ADJUSTMENT`) from `0.3` to `0.25`.  Do you know if the global placement is using the `routability_driven` flag? Perhaps we just need a slightly better placement to make the global route finishes properly with the current configuration.

vijayank88, 2022-03-03:
@eder-matheus 
I've tested with OR commit `OpenROAD 3eafa6da2942ec22d27e2208c99c7a449af34fbf` and faced congestion issue with OpenLane tag: openlane 2022.02.10_02.15.04
in OpenLane `set ::env(PL_ROUTABILITY_DRIVEN) {0}` default set. 
With latest tags OR commits got updated. Let me try again with changes and update you on this.
Thanks...

vijayank88, 2022-03-03:
@eder-matheus 
By reducing the global adjustment (GLB_RT_ADJUSTMENT) from 0.3 to 0.25 resolved the congestion issue.

closed 2022-03-03 by vijayank88