The-OpenROAD-Project/OpenROAD #1640: resizer failing with macro placement

DarylK78, 2022-02-14:
I've created a floorplan with some Skyway RAM macros placed, and am not getting an error in the placement step:

[ERROR DPL-0019] detailed placement failed on _102266_
Error: resizer.tcl, 76 DPL-0019

Is there way to get more information about this error?
[openroad.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8058021/openroad.tar.gz)

vijayank88, 2022-02-14:
@DarylK78 
The attached format is not valid. 
Please compress `openroad_issue_reproducible` directly  with .zip and attach here. Don't compress again with .gz extension

DarylK78, 2022-02-14:
I've uploaded the zipped version here.

[openroad.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8062611/openroad.zip)

vijayank88, 2022-02-14:
@DarylK78 
Is this `openroad_issue_reproducible` generated by openlane flow or do you made any manual edits for zip?
Still I'm see `.gz` extension. See after unzip
```
Archive:  openroad.zip
   creating: openroad_issue_reproducible/
   creating: openroad_issue_reproducible/tmp/
   creating: openroad_issue_reproducible/tmp/synthesis/
  inflating: openroad_issue_reproducible/tmp/synthesis/resizer_sky130_fd_sc_hd__tt_100C_1v80.lib.gz  
   creating: openroad_issue_reproducible/tmp/floorplan/
  inflating: openroad_issue_reproducible/tmp/floorplan/20-initial_fp.sdc.gz  
   creating: openroad_issue_reproducible/tmp/placement/
  inflating: openroad_issue_reproducible/tmp/merged_unpadded.lef.gz  
   creating: openroad_issue_reproducible/openlane/
   creating: openroad_issue_reproducible/openlane/scripts/
   creating: openroad_issue_reproducible/openlane/scripts/openroad/
 extracting: openroad_issue_reproducible/openlane/scripts/openroad/sta.tcl.gz  
 extracting: openroad_issue_reproducible/openlane/scripts/openroad/resizer.tcl.gz  
 extracting: openroad_issue_reproducible/openlane/scripts/openroad/set_rc.tcl.gz  
   creating: openroad_issue_reproducible/results/
   creating: openroad_issue_reproducible/results/synthesis/
  inflating: openroad_issue_reproducible/results/synthesis/LPI_LPD_Tx_DUT.v.gz  
  inflating: openroad_issue_reproducible/in.def  

```

DarylK78, 2022-02-14:
Sorry about that, try this one

[openroad.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8062785/openroad.zip)

vijayank88, 2022-02-15:
@DarylK78 
```
>>> source run.tcl
Error: resizer.tcl, 23 cannot read file sky130_sram_macros/sky130_sram_1r1w_64x18/sky130_sram_1r1w_64x18_TT_1p8V_25C.lib.
```
missing sky130_sram_macros directory.

DarylK78, 2022-02-15:
Sorry again, I added those directories and tested the run.sh script to verify the issue is reproduced.

[openroad.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8072790/openroad.zip)

maliberty, 2022-02-16:
You have a fair amount of placement density coming into this step before repair_design is even run:
[image]
It fails detailed placement even before repair_design.You also have a lot of violations of max slew & cap that exacerbates the problem.    When you design with 60k placeable instances generates:
```
[INFO RSZ-0038] Inserted 1215 buffers in 126 nets.
[INFO RSZ-0039] Resized 52917 instances.
```
I feel the problem lies upstream of this step.  Are you using any non-default settings in this flow?

Also the macros abutted to the edge are likely going to cause you routing/timing problem later on due the pins that will be hard to reach and lack of places to buffer.

vijayank88, 2022-02-16:
@DarylK78 
Update your config.tcl with `set ::env(CELL_PAD) 2` and its resolves DPL error at my end.
Also consider @maliberty response to avoid routing/timing problems.

DarylK78, 2022-02-16:
Thanks for the helpful responses.  A few notes:

1) I am using the default synthesis settings from OpenRoad.
2) When I tried this same design without doing any manual macro placement, it was unable to route due to the macros being placed directly on top of one another.  With the pins being along the top and bottom, this was an issue which is why I tried to move them apart a bit.
3) In my experience it is always recommended to place macros close to the edges of designs to give the placer a maximum amount of unobstructed placeable area.  Given that the pins are along the top and bottom of the RAMs, this would seem to apply here as well.  Is there something I'm missing?

maliberty, 2022-02-16:
1) Yosys does synthesis not OR.  Do you mean default OL settings in general?
2) File an issue if that happens
3) My experience is that a small halo should be left around the macro to leave room for buffering and routing.  If it works for you that's fine.

DarylK78, 2022-02-16:
Yes, I did mean the default OL settings for synthesis with yosys.

maliberty, 2022-02-16:
There are many OL flow settings that affect OR and I'm wondering about those more than yosys.

DarylK78, 2022-02-16:
I can submit my synthesis setup if that would be helpful?

DarylK78, 2022-02-16:
After placement I am running into routing congestion, can you take a look and give some suggestions?

closed 2022-02-16 by DarylK78