The-OpenROAD-Project/OpenROAD #1646: global routing failing

DarylK78, 2022-02-16:
Running a design with 60K placeable gates, seeing routing congestion issues and not sure how to debug.

I am unable to attach my operoad.zip as it say "is not included in the list"

maliberty, 2022-02-16:
Perhaps it dislikes the extension or you have the issue in https://github.com/github/hub/issues/1479

Try changing it to .zip1 and see if it works.

DarylK78, 2022-02-17:
Actually seems to be an issue with the filesize, is there a max on how large files to upload can be in the comments?

maliberty, 2022-02-17:
Probably but idk.  You can always put the file in some cloud storage and provide a link (eg google drive, dropbucket).

vijayank88, 2022-02-17:
@DarylK78 
Yes there is file size limitation to attach here. If you're facing issue, upload file in your personal repo and share link here.

DarylK78, 2022-02-21:
I was able to get the archive to a manageable size, this was run with the PL_TARGET_DENSITY at 0.25 and the FP_CORE_UTIL at 25.  Still seeing too much congestion

[openroad.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8106460/openroad.zip)

vijayank88, 2022-02-21:
@DarylK78 
Core has enough free space. But still I'm seeing high congestion near macro and std cell area as well.
```
set ::env(PL_MACRO_CHANNEL) "10 10"
set ::env(PL_MACRO_HALO) "10 10"
```
Can you try this setting in your OpenLane config.tcl and re-run the flow.

Possible try to place macros manullay based functionality.

eder-matheus, 2022-03-02:
I see this in the congestion map:
[image]
Perhaps the placement density is too large? The global router ends with high congestion even when I don't use the resources adjustments.

mousaq92, 2022-03-03:
How do you get these heat maps in the GUI?

eder-matheus, 2022-03-03:
> How do you get these heat maps in the GUI?

In the `Display Control`, go to `Heat Maps` and select what heat map you want to visualize. In the figure below, I'm showing only the routing congestion heat map:
[image]

Sya0, 2022-03-03:
> I see this in the congestion map: [image] Perhaps the placement density is too large? The global router ends with high congestion even when I don't use the resources adjustments.

@mousaq92, @eder-matheus 
Sorry to interrupt but is this klayout?

vijayank88, 2022-03-03:
@Sya0 its OpenROAD GUI
Command to use: `openroad -gui`

DarylK78, 2022-03-03:
I took the advice to a add macro channels, and now it is able to global route but runs into issues with diodes

See latest run here:

[openroad.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8179731/openroad.zip)
.

luis201420, 2022-03-09:
@DarylK78 Could you upload the tarball containing the stage before the global placement?

DarylK78, 2022-03-09:
This one is the tarball of the placement step I believe

[openroad.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8217858/openroad.zip)

DarylK78, 2022-03-09:
This one runs the init_floorplan

[floorplan.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8217977/floorplan.zip)

eder-matheus, 2022-03-09:
Thanks very much, @DarylK78!

ravi-varadarajan, 2022-03-09:
I examined the congested areas and it is due the dense concentration of combinational cells (4 input nor or ao22/oa22) in the congested area. You should be able to resolve this by using non zero  global and detailed placement padding. We will also examine the init_floorplan tarball -- thanks

eder-matheus, 2022-03-15:
> I took the advice to a add macro channels, and now it is able to global route but runs into issues with diodes
> 
> See latest run here:
> 
> [openroad.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8179731/openroad.zip) .

The latest version of OpenROAD have a fix for the diodes issue. Can you try that?

DarylK78, 2022-03-16:
I was able to get it to route by using DIODE_STRATEGY 4 and turning off fill.  

I'm also seeing a lot of slew violations which might be affecting the diode issue (nets are too long).  I will go back to placement and try some additional things, and also attempt to use the latest version of openroad as well.

vijayank88, 2023-06-14:
Closing as resolved based on user comment.

closed 2023-06-14 by vijayank88